zhen-zen/YogaSMC #4: ThinkSMC

tylernguyen, 2020-07-21:
Hey there, 

Thanks so much for working on this project. I'm looking forward to it.

I own an `x1c6` with functional macOS so if you need anything data or testing for the machine to develop a ThinkPadSMC kext, please let me know and I will gladly help.

zhen-zen, 2020-07-21:
Thanks for your attention! Currently battery conservation feature has read only support. You can try it on with `ioio -s ThinkVPC Battery 0`, maybe I should get a gitter or something?

zhen-zen, 2020-07-21:
Just created a chat room at gitter (see readme). And pre-compiled binary is available in Github Actions.

tylernguyen, 2020-07-21:
@zhen-zen 
GitHub releases is showing only the zipped source code, no binaries. Could you take a look?

zhen-zen, 2020-07-22:
> @zhen-zen
> GitHub releases is showing only the zipped source code, no binaries. Could you take a look?

It's in Github Actions, just next to the PR

benbender, 2020-07-25:
Would add myself to the party with a Xcode and T480 to do testing :)

Sniki, 2020-08-08:
Can provide help as well, i have these ThinkPads:
- Lenovo ThinkPad T440
- Lenovo ThinkPad T440S
- Lenovo ThinkPad X240

zhen-zen, 2020-08-13:
From [`drivers/platform/x86/thinkpad_acpi.c`](https://github.com/torvalds/linux/blob/master/drivers/platform/x86/thinkpad_acpi.c), I got following possible functions:

- hotkey (seems like polling?)
- Fan control / monitor
- Battery Wear control (experimental): BDVT (sys pref), BCLM (3rd party software like https://github.com/davidwernhart/AlDente)

There's also a lot of methods that are not documented clearly (e.g. InhibitCharge / ForceDischarge / PeakShiftState). 

- Thermal
- adaptive keyboard
- tablet mode detection
- nvram store
- keyboard light
- thinklight (not sure of cmos_handle)
- backlight (is it still used?)
- HW mute (?) / LED
- PrivacyGuard (?)
- DYTC (Thermal Control ?)

Since currently I'm out of reach for a ThinkPad, I can't promise the progress and at this stage just follow `thinkpad-acpi`. At the first stage I will focus on HKEY first since it won't require much testing and I'm still unsure how to distinguish SMC for different brands.

benbender, 2020-08-13:
> From [`drivers/platform/x86/thinkpad_acpi.c`](https://github.com/torvalds/linux/blob/master/drivers/platform/x86/thinkpad_acpi.c), I got following possible functions:
>
> * hotkey (seems like polling?) [...]

If my quicklook is correct, it's "only" about warnings? DYTC seems much more interesting.

> * adaptive keyboard

Some early kind of "touch bar", seems to be only used on X1 Gen1 & Gen 2. Needs most likely more software to be controlled and used...

> * tablet mode detection

Could be interesting in conjunction with VoodooInput :)

> * nvram store

???

> * keyboard light

Can be adjusted in 3 levels on T480 (off, low, high). Works already with ThinkpadAssistant.

> * thinklight (not sure of cmos_handle)

Was a LED above the LCD on older Thinkpads. Had one on my R60 in early 2000s :) Don't think it's used anymore bacause of keyboard backlights.

> * backlight (is it still used?)

Could be keyboard or LCD... Seems to be LCD looking at the code linked above and is working nicely already with acpi-patches. (Only culprit: much more delay on change compared to a native mac)

> * HW mute (?) / LED

On the T480 two mute LEDs can be found (mic + speaker). Works already with ThinkpadAssistant. If integrated via EC could be much more stable and simpler.

> * PrivacyGuard (?)

Optional Filter on modern thinkpads to increase privacy by decreasing the point of view of the LCD. 

> * DYTC (Thermal Control ?)

Interesting find! :) Seems to be used, having those fields in my ACPI (maybe only available if DPTF is disabled). 

STMD | QTMD | Mode
-- | -- | --
0x0 | 0x0 | Extreme Performance
0x0 | 0x1 | Battery Saving
0x1 | 0x0 | Intelligent Cooling

Source: https://wiki.archlinux.org/index.php/Lenovo_IdeaPad_5_14are05#System_Performance_Mode

> 
> Since currently I'm out of reach for a ThinkPad, I can't promise the progress and at this stage just follow `thinkpad-acpi`. I will first focus on HKEY since it won't require much testing and I'm still unsure how to distinguish SMC for different brands.
> 
> Btw, I'm quite envy yours for the detailed document and available functions lol. However, they didn't have an ICL one so far and I'm waiting for TGL this year ;)

SukkaW, 2020-08-13:
> Can be adjusted in 3 levels on T480 (off, low, high).

Without ThinkPadAssistant it still works (on ThinkPad E480 with VoodooPS2 directly).

> If integrated via EC could be much more stable and simpler.

ThinkPad E480(E580, R480, L480) has a broken EC while other ThinkPad should be fine.

> Optional Filter on modern thinkpads to increase privacy by decreasing the point of view of the LCD.

I am not sure but the feature is not on my ThinkPad.

Sniki, 2020-08-13:
Everything existing whether ACPI fixed or with ThinkPadAssistant that can be done via SMC is better IMHO.
Less bloated setup and should work more like natively.

Keyboard Backlight works without anything at all it just shows no GUI on macOS unless we create SSDT so it works using ThinkPad Assistant.

Another thing that is interesting is port replicator on Docking Station or whatever is related to docking station.
Everything seems to work nice natively, i added USBPorts and LAN works fine, also audio sort of from the combo jack, the only issue is:

There is a kernel panic when internal display turns off (either reboot,shutdown or sleep) if external display is attached into any of the Docking Station ports (DisplayPort, DVI, VGA)

I provided logs to Acidanthera, opened issue twice but they closed the issue by just saying it's something on the side of Apple drivers.

Not sure what are you referring to hotkey, is it Fn Keyboard Buttons ?

HW Mute/LED would be nice

Fan Reading/Control - yes please (the x240 needs some tweaking as it's loud most of the time.) having this would be awesome.

Battery wear control is nice too - how are we going to handle that on dual battery laptops, regarding the fact on how we implement dual battery support is by merging notifiers into one and adding the BATC device, will that affect this feature ?

All these features would be amazing if implemented in SMC (ThinkSMC or whatever you guys plan to name it), less SSDTs, less ACPI patches, less trouble, less complications.

Thank you for this amazing project !

SukkaW, 2020-08-13:
> There is a kernel panic when internal display turns off (either reboot,shutdown or sleep) if external display is attached into any of the Docking Station ports (DisplayPort, DVI, VGA)

Even MacBook Pro has the same issue and I just assume macOS doesn't work well with external display.

> Not sure what are you referring to hotkey, is it Fn Keyboard Buttons ?

Yes (while ThinkPad E480 has a broken EC, neither Fn nor battery status working after S3 sleep)

zhen-zen, 2020-08-13:
> > * Thermal
> 
> If my quicklook is correct, it's "only" about warnings? DYTC seems much more interesting.

Oh I forgot the warning, that one should belong to hotkey processing. The Thermal here refers to temperature reading.

> > * tablet mode detection
> 
> Could be interesting in conjunction with VoodooInput :)

I have already implemented that on my current Yoga laptop, with keyboard & touchpad disabled automatically. The difference should only be ACPI method.

> 
> > * nvram store
> 

The `thinkpad-acpi` module refers to NVRAM and CMOS for setting persistency, while I'm not sure if the latter will work.

> > * keyboard light
> 
> Can be adjusted in 3 levels on T480 (off, low, high). Works already with ThinkpadAssistant.

Will also work with hotkey processing natively without any DSDT patch. My current yoga laptop won't turn it off when sleeping (Mine only has S0 since S3 can cause kp), so I turn it off automatically for sleep and Yoga mode switch.

> > * thinklight (not sure of cmos_handle)
> 
> Was a LED above the LCD on older Thinkpads. Had one on my R60 in early 2000s :) Don't think it's used anymore bacause of keyboard backlights.

I thought it's the red LED dot on palm rest or back of screen. Or it's just `_SI._SST`?

benbender, 2020-08-13:
> > > * keyboard light
>
> Will also work with hotkey processing natively without any DSDT patch. My current yoga laptop won't turn it off when sleeping (Mine only has S0 since S3 can cause kp), so I turn it off automatically for sleep and Yoga mode switch.

As @Sniki said, it was working, but ThinkAssistant added the GUI. Sorry for the confusion :)

> > > * thinklight (not sure of cmos_handle)
> I thought it's the red LED dot on palm rest or back of screen. Or it's just `_SI._SST`?

Nope, I'm relative sure it's this: https://en.wikipedia.org/wiki/ThinkLight

Models which have it can be found here: (maybe incomplete)

https://thinkwiki.de/Tastaturbeleuchtung

zhen-zen, 2020-08-13:
> Everything existing whether ACPI fixed or with ThinkPadAssistant that can be done via SMC is better IMHO.
> Less bloated setup and should work more like natively.

For hotkey reading, there might be still a userspace application / OSD for that, or those events will stay in the log.

> There is a kernel panic when internal display turns off (either reboot,shutdown or sleep) if external display is attached into any of the Docking Station ports (DisplayPort, DVI, VGA)

Maybe `force-online` could prevent that? I'm not sure of that.

> Battery wear control is nice too - how are we going to handle that on dual battery laptops, regarding the fact on how we implement dual battery support is by merging notifiers into one and adding the BATC device, will that affect this feature ?
> 

There's an arg for that to choose from BAT_ANY, BAT_PRIMARY and BAT_SECONDARY. This happens at EC level so there should be no issue with BATC.

SukkaW, 2020-08-13:
> Oh I forgot the warning, that one should belong to hotkey processing. The Thermal here refers to temperature reading.

Unlike other models from Lenovo, ThinkPad doesn't support thermal control through hotkey.

> I have already implemented that on my current Yoga laptop, with keyboard & touchpad disabled automatically. The difference should only be ACPI method.

Only ThinkPad X1 *Yoga* has tablet mode though.

> I thought it's the red LED dot on palm rest or back of screen. Or it's just `_SI._SST`?

`_SI._SST` should be red dot (ThinkDot, as mentioned). And it could be patched through `_WAK`:

```swift
    Method (_WAK, 1, NotSerialized) //Method (_WAK, 1, Serialized)
    {   
        If (_OSI ("Darwin"))
        {
          If ((3 == Arg0) && CondRefOf (\_SI._SST))
          {
            \_SI._SST (1)
          }
        }

        Local0 = ZWAK(Arg0)
        Return (Local0)
    }
```

benbender, 2020-08-13:
@SukkaW Better method (because it doesn't need the rename of _WAK) imho f.e. here (also implements state-persistance for the Mute-LED with ThinkAssistant): https://github.com/B0hrer/thinkpad-x1c5-hackintosh/blob/master/patches/SSDT-LED.dsl

zhen-zen, 2020-08-13:
> > Oh I forgot the warning, that one should belong to hotkey processing. The Thermal here refers to temperature reading.
> 
> Unlike other models from Lenovo, ThinkPad doesn't support thermal control through hotkey.

Not necessary to trigger from hotkey. If the method exist, we can still call it.

> > I have already implemented that on my current Yoga laptop, with keyboard & touchpad disabled automatically. The difference should only be ACPI method.
> 
> Only ThinkPad X1 _Yoga_ has tablet mode though.

IIRC there's a lot models like X260 Yoga etc.

zhen-zen, 2020-08-13:
> @SukkaW Better method (because it doesn't need the rename of _WAK) imho f.e. here (also implements state-persistance for the Mute-LED with ThinkAssistant): https://github.com/B0hrer/thinkpad-x1c5-hackintosh/blob/master/patches/SSDT-LED.dsl

Could also be implemented with setPowerState XD

Sniki, 2020-08-13:
> > Everything existing whether ACPI fixed or with ThinkPadAssistant that can be done via SMC is better IMHO.
> > Less bloated setup and should work more like natively.
> 
> For hotkey reading, there might be still a userspace application / OSD for that, or those events will stay in the log.
> 
VoodooPS2Controller / VoodooPS2Keyboard does handle them but if we can just at least get rid of the ACPI renames for the methods (around 10 per thinkpad: F4-F12, PrtSc) and have to provide the patched ones into a separate it will be amazing.
Instead having the option to do that from SMC will be awesome, also having the hardware mute LED report correctly (save states) would be amazing too.
> > There is a kernel panic when internal display turns off (either reboot,shutdown or sleep) if external display is attached into any of the Docking Station ports (DisplayPort, DVI, VGA)
> 
> Maybe `force-online` could prevent that? I'm not sure of that. Btw I think on ACPI level `_LID` could be set to always on? I used this function to implement clamshell mode.
> 
Will give this one a try.
> There's an arg for that to choose from BAT_ANY, BAT_PRIMARY and BAT_SECONDARY. This happens at EC level so there should be no issue with BATC.
Excellent to know that.

zhen-zen, 2020-08-24:
Since progress for anticipated functions is even with IdeaSMC, maybe I can close this issue for now? Mentioned functions should have been synced to https://github.com/zhen-zen/YogaSMC/projects/2.

closed 2020-08-25 by tylernguyen

tylernguyen, 2020-08-25:
@zhen-zen Now that you got these things added. Considering renaming the project to `LenovoSMC` instead? I think that title more straight forward.

zhen-zen, 2020-08-25:
> @zhen-zen Now that you got these things added. Considering renaming the project to `LenovoSMC` instead? I think that title more straight forward.

This project started from an idea to exploit available WMI methods that are available on most PCs. I make use of every WMI devices on my laptop and also got it running on my desktop to parse them for basic description. Then it gradually evolved into current one which has a lot of implementation on the vendor's device interface under EC.

While some laptops have similar structure, others have slight differences. That vendor specific device only act as an airplane mode switcher and most of equivalent functions are implemented under WMI devices. So current framework may still compatible with other brands. That's why I leaved a `Yoga` there, and no just for the Yoga brand ;)

benbender, 2020-08-25:
Spinning that thoughts further: wouldn't it be YogaWMI + LenovoSMC (as two projects, where LenovoSMC depends on YogaWMI)?

zhen-zen, 2020-08-25:
> Spinning that thoughts further: wouldn't it be YogaWMI + LenovoSMC (as two projects, where LenovoSMC depends on YogaWMI)?

There are many interactions between WMI and other parts. On my laptop, WMI part recognizes Fn+esc, battery info and Yoga mode switch. And my desktop one also has some sensor info. I don't know what else are on yours other than BIOS management interface, but the connections are even closer with those WMI-centered vendor controllers.

KirisameMarisaa, 2020-10-09:
I'm also willing to provide help. Own one x1 yoga gen3, may be helpful to test yoga modes.

Charlyo, 2020-10-19:
Same here with a T460 and T470. Already running 1.2.0.

Mifaxav, 2022-09-21:
@benbender i need some help with the T480 patches. It used to work on a previous EFI but not on this one. Any tips?

Mifaxav, 2022-09-21:
this is to run yogasmc because yogasmcnc is not detecting the kext