zhen-zen/YogaSMC #11: Lenovo Legion Y540

encoreisone, 2020-09-24:
Hi

Is it possible to make it work on the lenovo legion y540-15irh?

Best regards 
encore

zhen-zen, 2020-09-24:
Typically the driver should just work with common features enabled for Ideapad (all product lines besides Thinkpad) and Thinkpad family. Is there something wrong on yours? Can you provide a DSDT/SSDT dump if the driver is not loading or matching?

encoreisone, 2020-09-24:
I have enable battery conservation mode on but its doesn't work as on my windows with Lenovo Vantage its working fine, its stay 60%.

Here is my DSDT.

[image]

[DSDT.txt](https://github.com/zhen-zen/YogaSMC/files/5279737/DSDT.txt)

zhen-zen, 2020-09-25:
Related method calling is the same with Linux driver,

zhen-zen, 2020-09-26:
There's a verification for `SBMC` in your DSDT. If the prerequisite is not met, the method won't proceed.

```
                    Method (SBMC, 1, NotSerialized)
                    {
                        If (ECAV)
                        {
                            If ((Acquire (LFCM, 0xA000) == Zero))
                            {
                                If ((Arg0 == Zero))
                                {
                                    CDMB = Zero
                                    EDCC = One
                                    Release (LFCM)
                                    Return (Zero)
                                }

                                If ((Arg0 == One))
                                {
                                    CDMB = One
                                    Release (LFCM)
                                    Return (Zero)
                                }

                                If ((Arg0 == 0x03))
                                {
                                    BTSM = One
                                    Release (LFCM)
                                    Return (Zero)
                                }

                                If ((Arg0 == 0x05))
                                {
                                    BTSM = Zero
                                    Release (LFCM)
                                    Return (Zero)
                                }

                                If ((Arg0 == 0x07))
                                {
                                    QCHO = One
                                    BTSM = Zero
                                    Release (LFCM)
                                    Return (Zero)
                                }

                                If ((Arg0 == 0x08))
                                {
                                    QCHO = Zero
                                    Release (LFCM)
                                    Return (Zero)
                                }

                                If ((Arg0 == 0x09))
                                {
                                    ESMC = One
                                    Release (LFCM)
                                    Return (Zero)
                                }

                                If ((Arg0 == 0x10))
                                {
                                    ESMC = Zero
                                    Release (LFCM)
                                    Return (Zero)
                                }

                                Release (LFCM)
                            }
                        }

                        Return (Zero)
                    }
```

Can you try `ioio -s IdeaVPC ReadECName ECAV` and check its value in log?

zhen-zen, 2020-09-26:
Sorry, my bad. I added a lock for conservation mode when testing SMC key interaction. Can you try the latest build in Github Action?

encoreisone, 2020-09-26:
Just tried the latest build, still not working for me. 

_Here are my log_:

[image]

also how do you install pref pane correctly way? I ran the it and it said "Coud not load..."

[image]

encoreisone, 2020-09-26:
Just tried the latest build, its working now. Thanks for helping me :)

_Here are my log_:

[image]

How do you install pref pane correctly way? I ran it and it said "Coud not load..."

[image]

zhen-zen, 2020-09-26:
Yes, since it is not signed. You can use `xattr -c YogaSMCPane.prefPane` to clear the quarantine attribute before open it.

encoreisone, 2020-09-26:
i've tried above command and also tried "sudo xattr -r -d com.apple.quarantine YogaSMCPane.prefPane"

It still said "Could not load.."  ¯\_(ツ)_/¯

encoreisone, 2020-09-26:
i've tried above command and also tried `sudo xattr -r -d com.apple.quarantine YogaSMCPane.prefPane`

It still said "Could not load.."     `¯\_(ツ)_/¯`

zhen-zen, 2020-09-26:
Can you try the latest build in #12 and filter the system log with keyword `YogaSMCPane`?

zhen-zen, 2020-09-27:
Not sure what's going wring with CLI build. 
[YogaSMCPane.prefPane.zip](https://github.com/zhen-zen/YogaSMC/files/5287557/YogaSMCPane.prefPane.zip)

encoreisone, 2020-09-27:
My gatekeeper is disabled. I still got the same message :/

[image]

zhen-zen, 2020-09-28:
I suppose the kext should match all information that the prefpane requested. And it should still load if any property is missing. Could you take a screenshot of IdeaVPC entries in ioreg? Maybe a binary search is necessary to locate the culprit.

encoreisone, 2020-09-28:
Here are my ioreg log

[ioreg.txt](https://github.com/zhen-zen/YogaSMC/files/5291265/ioreg.txt)

encoreisone, 2020-09-28:
Here are my ioreg log

[ioreg.txt](https://github.com/zhen-zen/YogaSMC/files/5291323/ioreg.txt)

[image]
[image]
[image]
[image]
[image]
[image]

zhen-zen, 2020-09-30:
Actually the screenshot of `IdeaVPC` properties is skipped exactly. Can you send a screenshot only for that and expand the `Capability` dictionary?  The latest commit on `master` should have allowed missing value for that. But I still wonder why they are missing.

encoreisone, 2020-09-30:
Is this correct? Am I missing something?

[image]

zhen-zen, 2020-09-30:
Yes... I'm not sure why Capability is not recognized, and there are some invalid values, maybe related method's not patched? Do you have working battery status?

encoreisone, 2020-09-30:
Battery status looks normal to me 

[image]

zhen-zen, 2020-09-30:
It seems that related methods vary from different series? Can you try the latest debug version kext and prefpane?

encoreisone, 2020-10-01:
It seems it is working now =)

 Thank you for your help.  I truly appreciate your hard work.

[image]

zhen-zen, 2020-10-01:
Release version should work now. It seems that the kext is loaded too early that some properties are not available at that time. Maybe reset or equivalent method will be performed in pref pane.

encoreisone, 2020-10-02:
Works as expected, thanks.

closed 2020-10-02 by encoreisone