zhen-zen/YogaSMC #35: Lenovo Ideapad s540

marianopela, 2020-10-20:
Hi, as you know, my device has problems activating Conservation Mode (it works under Linux)
Here are my [ACPI](https://github.com/zhen-zen/YogaSMC/files/5410073/ACPI.zip) tables

I'm currently using https://github.com/zhen-zen/YogaSMC/pull/22, and the result is something like this

```
Filtering the log data using "processIdentifier == 0 AND (senderImagePath CONTAINS "YogaSMC" OR senderImagePath CONTAINS "ACPI")"
Timestamp                       Thread     Type        Activity             PID    TTL  
2020-10-19 13:19:35.975461+0200 0x1b9e     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Battery 0xe0200
2020-10-19 13:19:35.975598+0200 0x1b9e     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 KeyboardMode 0xc650
2020-10-19 13:19:37.999355+0200 0x1b9e     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
2020-10-19 13:19:37.999359+0200 0x1b9e     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 Failed to read VPC1 after 22 attempts
2020-10-19 13:19:37.999484+0200 0x1b9e     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 KeyboardMode 0xc650
2020-10-19 13:19:37.999926+0200 0x1b9e     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 DYTCMode command 0x2 result 0x08013b01
2020-10-19 13:19:37.999936+0200 0x1b9e     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Found DYTC_FUNCTION_STD
2020-10-19 13:19:37.999939+0200 0x1b9e     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Found DYTC_FUNCTION_MMC
2020-10-19 13:19:39.005667+0200 0x1b9e     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Battery 0 cycle count 98
2020-10-19 13:19:39.005677+0200 0x1b9e     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Battery 0 ID 0005 0320 0101 0006
2020-10-19 13:19:42.827094+0200 0x1b9e     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 ConservationMode set to 0x3: on
2020-10-19 13:19:52.217960+0200 0x1e0d     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Battery 0xe0200
2020-10-19 13:19:52.218059+0200 0x1e0d     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 KeyboardMode 0xc650
2020-10-19 13:19:55.227307+0200 0x1e0d     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 read VPC EC result: 0x0 3
2020-10-19 13:19:55.227343+0200 0x1e0d     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 empty EC event
2020-10-19 13:19:55.227647+0200 0x1e0d     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 KeyboardMode 0xc650
2020-10-19 13:19:55.229353+0200 0x1e0d     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 DYTCMode command 0x2 result 0x08013b01
2020-10-19 13:19:55.229392+0200 0x1e0d     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Found DYTC_FUNCTION_STD
2020-10-19 13:19:55.229402+0200 0x1e0d     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Found DYTC_FUNCTION_MMC
2020-10-19 13:19:55.235932+0200 0x1e0d     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Battery 0 cycle count 98
2020-10-19 13:19:55.235958+0200 0x1e0d     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Battery 0 ID 0005 0320 0101 0006
2020-10-19 13:20:05.638697+0200 0x1e0d     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 ConservationMode set to 0x3: on
2020-10-19 13:20:06.858298+0200 0x1e0d     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 ConservationMode already 0
2020-10-19 13:20:13.283204+0200 0x1ef8     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: YogaSMCUserClient::IdeaVPC start
2020-10-19 13:20:13.283580+0200 0x1ef8     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: YogaSMCUserClient::IdeaVPC getTargetAndMethodForIndex (index=0)
2020-10-19 13:20:13.283590+0200 0x1ef8     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: YogaSMCUserClient::IdeaVPC userClientOpen
2020-10-19 13:20:13.284974+0200 0x1ef8     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: YogaSMCUserClient::IdeaVPC registerNotificationPort subscribed
2020-10-19 13:20:17.260134+0200 0x1ef8     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: YogaSMCUserClient::IdeaVPC userClientClose notification unregistered
2020-10-19 13:20:17.260156+0200 0x1ef8     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: YogaSMCUserClient::IdeaVPC userClientClose
2020-10-19 13:20:17.260311+0200 0x1ef8     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: YogaSMCUserClient::IdeaVPC clientClose success
2020-10-19 13:20:17.260348+0200 0x6f       Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: YogaSMCUserClient::IdeaVPC stop
2020-10-19 13:20:17.437439+0200 0x1f38     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: YogaSMCUserClient::IdeaVPC start
2020-10-19 13:20:17.437494+0200 0x1f38     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: YogaSMCUserClient::IdeaVPC getTargetAndMethodForIndex (index=0)
2020-10-19 13:20:17.437501+0200 0x1f38     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: YogaSMCUserClient::IdeaVPC userClientOpen
2020-10-19 13:20:17.456470+0200 0x1f38     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: YogaSMCUserClient::IdeaVPC registerNotificationPort subscribed
``

zhen-zen, 2020-10-20:
Before continue to battery, I want to first confirm whether #20 or #22 works. Are function key events recognized by the app and shown in the log? Such as keyboard backlight, camera and Mic mute.

For conservation mode, please try the latest debug version on master branch. Please install https://github.com/RehabMan/OS-X-ioio first and try following commands.
On: `ioio -s IdeaVPC ConservationMode 3`
Off: `ioio -s IdeaVPC ConservationMode 5`
These commands should do the exact same thing with Linux drivers without additional checks, the behavior should be the same. And please do not enter other values. I'm still not sure of its behavior. You may need to re-plug or execute it when charge level is less than 60%.

marianopela, 2020-10-20:
Ok, will wait for the battery to drain under 60 and report back

> Are function key events recognized by the app and shown in the log? Such as keyboard backlight, camera and Mic mute.

About keyboard backlight, this happens when using Fn + spacebar (but it's working, and it also working from prefpane)
```
2020-10-20 18:41:25.666940+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
2020-10-20 18:41:25.666992+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 Failed to read VPC2 after 57 attempts
2020-10-20 18:41:25.667020+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 read VPC EC result: 0x2 57
2020-10-20 18:41:25.667041+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 Fn+Space keyboard backlight?
2020-10-20 18:41:25.667629+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 KeyboardMode 0xc6f0
2020-10-20 18:41:37.684706+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
2020-10-20 18:41:37.684760+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 Failed to read VPC2 after 69 attempts
2020-10-20 18:41:37.684781+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 read VPC EC result: 0x2 69
2020-10-20 18:41:37.684808+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 Fn+Space keyboard backlight?
2020-10-20 18:41:37.685362+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 KeyboardMode 0xc6f0
2020-10-20 18:41:43.196474+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
2020-10-20 18:41:43.196508+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 Failed to read VPC2 after 138 attempts
2020-10-20 18:41:43.196521+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 read VPC EC result: 0x2 138
2020-10-20 18:41:43.196553+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 Fn+Space keyboard backlight?
2020-10-20 18:41:43.196909+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 KeyboardMode 0xc6d0
```

Here I pressed Mic mute four times (F4)
```
2020-10-20 18:47:34.865249+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 read VPC EC result: 0x100 40
2020-10-20 18:47:34.865288+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 Fn+F4 mic
2020-10-20 18:47:54.164426+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 read VPC EC result: 0x0 3
2020-10-20 18:47:54.164463+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 empty EC event
2020-10-20 18:48:03.070155+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
2020-10-20 18:48:03.070189+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 Failed to read VPC2 after 49 attempts
2020-10-20 18:48:03.070201+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 read VPC EC result: 0x0 49
2020-10-20 18:48:03.070219+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 empty EC event
2020-10-20 18:48:13.398452+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 read VPC EC result: 0x100 7
2020-10-20 18:48:13.398506+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 Fn+F4 mic
```
What about camera? I don't have a relative Fn key, but for example I use F6 to enable/disable trackpad (here I disable it and re-enable it
```
2020-10-20 20:54:43.054279+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
2020-10-20 20:54:43.054316+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 Failed to read VPC2 after 52 attempts
2020-10-20 20:54:43.054328+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 read VPC EC result: 0x20 52
2020-10-20 20:54:43.057129+0200 0x27af     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 Fn+F6 touchpad 0x0 off
2020-10-20 20:54:52.191545+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 read VPC EC result: 0x0 43
2020-10-20 20:54:52.191585+0200 0x27ae     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 empty EC event
```

zhen-zen, 2020-10-21:
By the way, do you have the `_OSI` to `XOSI` patch? During initialization, some EC fields seems to be written according to system version.

marianopela, 2020-10-21:
> By the way, do you have the `_OSI` to `XOSI` patch? During initialization, some EC fields like `OSTY` seem to be written according to system version.

Nope, I did not. I just tried it and it seems to have little or no impact on EC reading errors
How should I test it so that it's useful for you?

zhen-zen, 2020-11-01:
Can you try #41 again?

marianopela, 2020-11-01:
I charged from 50% to 64%

```mariano@MacBook-Pro ~ % log stream --predicate 'processID=0 && (senderImagePath contains "YogaSMC" OR senderImagePath contains "ACPI")'
Filtering the log data using "processIdentifier == 0 AND (senderImagePath CONTAINS "YogaSMC" OR senderImagePath CONTAINS "ACPI")"
Timestamp                       Thread     Type        Activity             PID    TTL  
2020-11-01 10:57:19.388331+0100 0x1e23     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 ConservationMode set to 0x3: see ioreg
2020-11-01 10:57:27.381809+0100 0x1e5f     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 ConservationMode set to 0x5: see ioreg
2020-11-01 10:57:31.382108+0100 0x1e90     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 ConservationMode set to 0x3: see ioreg
2020-11-01 10:57:34.406079+0100 0x1edb     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 ConservationMode set to 0x5: see ioreg
2020-11-01 10:57:43.278882+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Battery 0xe0200
2020-11-01 10:57:46.292163+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 read VPC EC result: 0x0 10
2020-11-01 10:57:46.292210+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 empty EC event
2020-11-01 10:57:46.293582+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 DYTCMode command 0x2 result 0x08013b01
2020-11-01 10:57:46.293603+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Found DYTC_FUNCTION_STD
2020-11-01 10:57:46.293610+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Found DYTC_FUNCTION_MMC
2020-11-01 10:57:54.135206+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 DYTCMode command 0x2 result 0x08013b01
2020-11-01 10:57:54.159189+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 DYTCMode command 0x13b001 result 0x08013b01
2020-11-01 10:57:54.159219+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Found DYTC_FUNCTION_STD
2020-11-01 10:57:54.159226+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Found DYTC_FUNCTION_MMC
2020-11-01 10:57:54.159243+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 DYTCMode set to 0x3: see ioreg
2020-11-01 10:57:58.647980+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 DYTCMode command 0x1ff result 0x0001f001
2020-11-01 10:57:58.648003+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Found DYTC_FUNCTION_STD
2020-11-01 10:57:58.648015+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 DYTCMode set to 0xf: see ioreg
2020-11-01 10:57:59.832597+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 DYTCMode command 0x2 result 0x0001f001
2020-11-01 10:57:59.856025+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 DYTCMode command 0x12b001 result 0x08012b01
2020-11-01 10:57:59.856095+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Found DYTC_FUNCTION_STD
2020-11-01 10:57:59.856113+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Found DYTC_FUNCTION_MMC
2020-11-01 10:57:59.856144+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 DYTCMode set to 0x2: see ioreg
2020-11-01 10:58:02.286185+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 BacklightLevel already 0
2020-11-01 10:58:06.638779+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 BacklightLevel set to 0x9: off
2020-11-01 10:58:16.437368+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 ConservationMode already 0
2020-11-01 10:58:18.021907+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 ConservationMode set to 0x3: on
2020-11-01 10:58:19.470343+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 ConservationMode already 0
2020-11-01 10:58:20.551000+0100 0x1f75     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 ConservationMode set to 0x3: on
2020-11-01 10:58:24.969333+0100 0x21e7     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Battery 0xe0200
2020-11-01 10:58:30.619902+0100 0x21e7     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
[... cut ...]
2020-11-01 11:09:05.209681+0100 0x2f68     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 DYTCMode command 0x2 result 0x08012b01
2020-11-01 11:09:05.209717+0100 0x2f68     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Found DYTC_FUNCTION_STD
2020-11-01 11:09:05.209727+0100 0x2f68     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 Found DYTC_FUNCTION_MMC
2020-11-01 11:09:17.462732+0100 0x2f68     Default     0x0                  0      0    kernel: (YogaSMC) YSMC - Debug: IdeaVPC::VPC0 BacklightLevel set to 0x8: on
```

marianopela, 2020-11-01:
Conservation mode's tick seems to be sticking though, but I lost the other infos
[image]

[image]

Here is also the [bootlog](https://github.com/zhen-zen/YogaSMC/files/5470555/bootlog.txt) from YogaSMC

Please tell me if you need anything else

marianopela, 2020-11-01:
> Conservation mode's tick seems to be sticking though

Nevermind, it remains selected even if I unselect it and reload the pane, so it's just pane problems

zhen-zen, 2020-11-01:
Please update the prefpane as well. And the hotkey is still not working? Do you have every ec hotkey press recognized in Linux?

marianopela, 2020-11-09:
I'm testing from Big Sur now and gave https://github.com/zhen-zen/YogaSMC/pull/52 a try
These ACPI errors rose

```Filtering the log data using "processIdentifier == 0 AND (senderImagePath CONTAINS "YogaSMC" OR senderImagePath CONTAINS "ACPI")"
Timestamp                       Thread     Type        Activity             PID    TTL  
2020-11-09 09:52:43.817888+0100 0x25ec     Default     0x0                  0      0    kernel: (AppleACPIPlatform) AcpiOsIdleQueue exhausted
2020-11-09 09:52:43.818108+0100 0x25ec     Default     0x0                  0      0    kernel: (AppleACPIPlatform) 
2020-11-09 09:52:43.818122+0100 0x25ec     Default     0x0                  0      0    kernel: (AppleACPIPlatform) ACPI Error:
2020-11-09 09:52:43.818124+0100 0x25ec     Default     0x0                  0      0    kernel: (AppleACPIPlatform) ACPI Error:
2020-11-09 09:52:43.818131+0100 0x25ec     Default     0x0                  0      0    kernel: (AppleACPIPlatform) Method parse/execution failed
2020-11-09 09:52:43.818134+0100 0x25ec     Default     0x0                  0      0    kernel: (AppleACPIPlatform) Method parse/execution failed
2020-11-09 09:52:43.818142+0100 0x25ec     Default     0x0                  0      0    kernel: (AppleACPIPlatform) [\_SB.PCI0.LPCB.EC0._Q11] (Node ffffff937e639600)
2020-11-09 09:52:43.818146+0100 0x25ec     Default     0x0                  0      0    kernel: (AppleACPIPlatform) [\_SB.PCI0.LPCB.EC0._Q11] (Node ffffff937e639600)
2020-11-09 09:52:43.818153+0100 0x25ec     Default     0x0                  0      0    kernel: (AppleACPIPlatform) , AE_NO_MEMORY
2020-11-09 09:52:43.818157+0100 0x25ec     Default     0x0                  0      0    kernel: (AppleACPIPlatform) , AE_NO_MEMORY
2020-11-09 09:52:43.818166+0100 0x25ec     Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/psparse-632)
2020-11-09 09:52:43.818171+0100 0x25ec     Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/psparse-632)
2020-11-09 09:52:43.818638+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) AcpiOsIdleQueue exhausted
2020-11-09 09:52:43.818654+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) [_Q04] @00008 #0086:
2020-11-09 09:52:43.818657+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) [_Q04] @00008 #0086:
2020-11-09 09:52:43.818665+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) Notify
2020-11-09 09:52:43.818668+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) Notify
2020-11-09 09:52:43.818671+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (
2020-11-09 09:52:43.818673+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (
2020-11-09 09:52:43.818677+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) V
2020-11-09 09:52:43.818681+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) V
2020-11-09 09:52:43.818683+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) P
2020-11-09 09:52:43.818685+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) P
2020-11-09 09:52:43.818687+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) C
2020-11-09 09:52:43.818690+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) C
2020-11-09 09:52:43.818692+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) 0
2020-11-09 09:52:43.818694+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) 0
2020-11-09 09:52:43.818698+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) ,
2020-11-09 09:52:43.818700+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) ,
2020-11-09 09:52:43.818704+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) 0x80
2020-11-09 09:52:43.818707+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) 0x80
2020-11-09 09:52:43.818710+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) )
2020-11-09 09:52:43.818712+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) )
2020-11-09 09:52:43.818716+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform)  // Status Change
2020-11-09 09:52:43.818719+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform)  // Status Change
2020-11-09 09:52:43.818724+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) 
2020-11-09 09:52:43.818726+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) 
2020-11-09 09:52:43.818746+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) 
2020-11-09 09:52:43.818751+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) 
2020-11-09 09:52:43.818758+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) No Local Variables are initialized for method [_Q04]
[... cut ...]
2020-11-09 09:52:43.818839+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) , AE_NO_MEMORY
2020-11-09 09:52:43.818842+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform) , AE_NO_MEMORY
2020-11-09 09:52:43.818849+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/psparse-632)
2020-11-09 09:52:43.818853+0100 0x25f0     Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/psparse-632)
```

zhen-zen, 2020-11-09:
Have no idea why `AE_NO_MEMORY`/`AcpiOsIdleQueue exhausted` happened, maybe the EC is slow to respond?

marianopela, 2020-11-09:
> Have no idea why `AE_NO_MEMORY`/`AcpiOsIdleQueue exhausted` happened, maybe the EC is slow to respond?

Could be. The logs appeared after bit in fact. Also, When updating YogaSMC, NC and pane I'm often experiencing slowdowns at login, boot failures (last was a hang) and KPs. I'm trying to get KPs logged on file for you to have a look at

zhen-zen, 2020-11-09:
If you have enabled `ApplePanic` in OC, then related log files will be available at the root of your EFI partition.

marianopela, 2020-11-09:
Unfortunately I lost those and didn't get those on Big Sur (just hangs), but I'll update to last commit and see what happens
Meanwhile:

>  Do you have each press of EC hotkey recognized in Linux?

I just tested those and they are all working, apart from F9 and F10 (Lock Screen and Display mode, but I guess it has to do with Ubuntu Live USB)

Now, after rebooting from Ubuntu (I was booting to enable ApplePanic) it is hanging here 
[image]

zhen-zen, 2020-11-09:
How about booting without this kext? Sounds like MT2 driver issues.

marianopela, 2020-11-09:
> How about booting without this kext? Sounds like MT2 driver issues.

Do you mean without YogaSMC? MT2 meaning VoodooI2C?

zhen-zen, 2020-11-09:
Yes. And the latter one sounds like Apple's driver.

marianopela, 2020-11-09:
I had it disabled in the last few days and had no problems, but I guess this is not very helpful

I'm retrying https://github.com/zhen-zen/YogaSMC/pull/41 as latest commits don't give me any info in the log
Do you recommend to always keep `XOSI` patch? I don't have it atm

zhen-zen, 2020-11-09:
Yes. Since Linux also report itself as latest Windows.

marianopela, 2020-11-09:
Ok, just added XOSI and rebooted
[image]

zhen-zen, 2020-11-09:
Quite weird. Can you try removing the `YogaWMI` entry in `Info.plist`?

marianopela, 2020-11-09:
It happened exactly as before. Now it's stuck on apple logo 
[IMG_0472.MOV.zip](https://github.com/zhen-zen/YogaSMC/files/5513718/IMG_0472.MOV.zip)

marianopela, 2020-11-09:
> Quite weird. Can you try removing the `YogaWMI` entry in `Info.plist`?

Sure, I'll force reboot and try

marianopela, 2020-11-09:
> Quite weird. Can you try removing the `YogaWMI` entry in `Info.plist`?

KP finally! Then stuck at Apple Logo again

marianopela, 2020-11-09:
```panic(cpu 4 caller 0xffffff8004ff0a76): Kernel trap at 0xffffff8005616e98, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x000000001adf7000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0xffffff86a7dfbbe0, RCX: 0x0000000000000000, RDX: 0x0000000000000000
RSP: 0xffffffa06aa53cb0, RBP: 0xffffffa06aa53cd0, RSI: 0x0000000000000000, RDI: 0xffffff9367341870
R8:  0x0000000000000000, R9:  0x0000000000000000, R10: 0x0000000000000000, R11: 0x0000000000000000
R12: 0x0000000000000000, R13: 0x0000000000000000, R14: 0xffffff9367341870, R15: 0x00000000e00002c7
RFL: 0x0000000000010246, RIP: 0xffffff8005616e98, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0, VF: 1

Backtrace (CPU 4), Frame : Return Address
0xffffffa06aa536d0 : 0xffffff8004ebd66d 
0xffffffa06aa53720 : 0xffffff8005000073 
0xffffffa06aa53760 : 0xffffff8004ff06aa 
0xffffffa06aa537b0 : 0xffffff8004e62a2f 
0xffffffa06aa537d0 : 0xffffff8004ebcf0d 
0xffffffa06aa538f0 : 0xffffff8004ebd1f8 
0xffffffa06aa53960 : 0xffffff80056bfe1a 
0xffffffa06aa539d0 : 0xffffff8004ff0a76 
0xffffffa06aa53b50 : 0xffffff8004ff075d 
0xffffffa06aa53ba0 : 0xffffff8004e62a2f 
0xffffffa06aa53bc0 : 0xffffff8005616e98 
0xffffffa06aa53cd0 : 0xffffff800561a88f 
0xffffffa06aa53d30 : 0xffffff80091a116d 
0xffffffa06aa53df0 : 0xffffff80055fdfe1 
0xffffffa06aa53e60 : 0xffffff80055fdd7a 
0xffffffa06aa53e80 : 0xffffff8004f02345 
0xffffffa06aa53ef0 : 0xffffff8004f03254 
0xffffffa06aa53fa0 : 0xffffff8004e6213e 
      Kernel Extensions in backtrace:
         org.zhen.YogaSMC(1.2.1)[4183E33D-2EFF-39EE-858F-844CFE1EDB23]@0xffffff800917d000->0xffffff80091c5fff
            dependency: as.vit9696.Lilu(1.5.0)[D44FCC5F-CB88-32FF-9C2B-59CE9969D4C0]@0xffffff8008b7d000->0xffffff8008ba6fff
            dependency: as.vit9696.VirtualSMC(1.1.9)[A05D183D-9D34-31C8-8D55-8FE54DCE6C44]@0xffffff8008ba9000->0xffffff8008bbffff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[B9AEA347-5086-364F-932A-211B8CF8C661]@0xffffff80072af000->0xffffff80072b0fff

Process name corresponding to current thread: Unknown
Boot args: brcmfx-country=#a brcmfx-aspm alcid=15 -v chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 20.1.0: Thu Oct 29 05:35:40 PDT 2020; root:xnu-7195.50.5~4/RELEASE_X86_64
Kernel UUID: 0FC1EF69-BB5B-336F-B8FB-C1121939CE17
KernelCache slide: 0x0000000004c00000
KernelCache base:  0xffffff8004e00000
Kernel slide:      0x0000000004c10000
Kernel text base:  0xffffff8004e10000
__HIB  text base: 0xffffff8004d00000
System model name: MacBookPro16,2 (Mac-5F9802EFE386AA28)
System shutdown begun: NO
Panic diags file unavailable, panic occurred prior to initialization
Hibernation exit count: 0

System uptime in nanoseconds: 3834453408
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x00000000e58d0400
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x00000008314bfcea 0x0000000000000000
```

zhen-zen, 2020-11-09:
Can you add `keepsyms=1` to `boot-arg`? It will print some more details.

marianopela, 2020-11-09:
> Can you add `keepsyms=1` to `boot-arg`? It will print some more details.

Yes, It's always too late when I realise, sorry

marianopela, 2020-11-09:
After a few hangs, here it is

```panic(cpu 4 caller 0xffffff8001bf0a76): Kernel trap at 0xffffff8002216e98, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x00000000179f7000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0xffffff86a4980d70, RCX: 0x0000000000000000, RDX: 0x0000000000000000
RSP: 0xffffffa0675dbcb0, RBP: 0xffffffa0675dbcd0, RSI: 0x0000000000000000, RDI: 0xffffff9363ec66e0
R8:  0x0000000000000000, R9:  0x0000000000000000, R10: 0x0000000000000000, R11: 0x0000000000000000
R12: 0x0000000000000000, R13: 0x0000000000000000, R14: 0xffffff9363ec66e0, R15: 0x00000000e00002c7
RFL: 0x0000000000010246, RIP: 0xffffff8002216e98, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0, VF: 1

Backtrace (CPU 4), Frame : Return Address
0xffffffa0675db6d0 : 0xffffff8001abd66d mach_kernel : _handle_debugger_trap + 0x3dd
0xffffffa0675db720 : 0xffffff8001c00073 mach_kernel : _kdp_i386_trap + 0x143
0xffffffa0675db760 : 0xffffff8001bf06aa mach_kernel : _kernel_trap + 0x55a
0xffffffa0675db7b0 : 0xffffff8001a62a2f mach_kernel : _return_from_trap + 0xff
0xffffffa0675db7d0 : 0xffffff8001abcf0d mach_kernel : _DebuggerTrapWithState + 0xad
0xffffffa0675db8f0 : 0xffffff8001abd1f8 mach_kernel : _panic_trap_to_debugger + 0x268
0xffffffa0675db960 : 0xffffff80022bfe1a mach_kernel : _panic + 0x54
0xffffffa0675db9d0 : 0xffffff8001bf0a76 mach_kernel : _sync_iss_to_iks + 0x2c6
0xffffffa0675dbb50 : 0xffffff8001bf075d mach_kernel : _kernel_trap + 0x60d
0xffffffa0675dbba0 : 0xffffff8001a62a2f mach_kernel : _return_from_trap + 0xff
0xffffffa0675dbbc0 : 0xffffff8002216e98 mach_kernel : __ZN10IOWorkLoop13_maintRequestEPvS0_S0_S0_ + 0x28
0xffffffa0675dbcd0 : 0xffffff800221a88f mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xaf
0xffffffa0675dbd30 : 0xffffff8005da116d org.zhen.YogaSMC : __ZN7YogaSMC13setPowerStateEmP9IOService + 0x1fd
0xffffffa0675dbdf0 : 0xffffff80021fdfe1 mach_kernel : __ZN9IOService19driverSetPowerStateEv + 0x1f1
0xffffffa0675dbe60 : 0xffffff80021fdd7a mach_kernel : __ZN9IOService15pmDriverCalloutEPS_ + 0x2a
0xffffffa0675dbe80 : 0xffffff8001b02345 mach_kernel : _thread_call_delayed_timer + 0x4a5
0xffffffa0675dbef0 : 0xffffff8001b03254 mach_kernel : _thread_call_delayed_timer + 0x13b4
0xffffffa0675dbfa0 : 0xffffff8001a6213e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         org.zhen.YogaSMC(1.2.1)[4183E33D-2EFF-39EE-858F-844CFE1EDB23]@0xffffff8005d7d000->0xffffff8005dc5fff
            dependency: as.vit9696.Lilu(1.5.0)[D44FCC5F-CB88-32FF-9C2B-59CE9969D4C0]@0xffffff800577d000->0xffffff80057a6fff
            dependency: as.vit9696.VirtualSMC(1.1.9)[A05D183D-9D34-31C8-8D55-8FE54DCE6C44]@0xffffff80057a9000->0xffffff80057bffff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[B9AEA347-5086-364F-932A-211B8CF8C661]@0xffffff8003eaf000->0xffffff8003eb0fff

Process name corresponding to current thread: Unknown
Boot args: brcmfx-country=#a brcmfx-aspm alcid=15 -v keepsyms=1 chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 20.1.0: Thu Oct 29 05:35:40 PDT 2020; root:xnu-7195.50.5~4/RELEASE_X86_64
Kernel UUID: 0FC1EF69-BB5B-336F-B8FB-C1121939CE17
KernelCache slide: 0x0000000001800000
KernelCache base:  0xffffff8001a00000
Kernel slide:      0x0000000001810000
Kernel text base:  0xffffff8001a10000
__HIB  text base: 0xffffff8001900000
System model name: MacBookPro16,2 (Mac-5F9802EFE386AA28)
System shutdown begun: NO
Panic diags file unavailable, panic occurred prior to initialization
Hibernation exit count: 0

System uptime in nanoseconds: 3960043791
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x00000000ed3fc308
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x000000075306ffae 0x0000000000000000
```

zhen-zen, 2020-11-10:
It's weird that the kext failed at this stage, since the first setPowerState will be skipped. This part is not related to specific models, I'm also on Big Sur but have't got such issues. Which version do you have? It changed a lot since last release.

marianopela, 2020-11-10:
I’m on last beta. 11.0.1
Had similar problems on Catalina, too

zhen-zen, 2020-11-10:
Hmmm, I mean the kext version

marianopela, 2020-11-10:
Lol sorry, I suspected that. I’m on https://github.com/zhen-zen/YogaSMC/pull/41

Never tried alter. What’s changed there?

zhen-zen, 2020-11-10:
The alter one don't support SMC integration. Since all readings are from EC, maybe that part breaks.

marianopela, 2020-11-10:
I’ll test that one. Should I try last commits on master and forget about the PR?

marianopela, 2020-11-10:
Latest YogaSMCAlter is booting fine. I now noticed a line in the preverbose of OC
`Dependency com.apple.iokit.IOHIDSystem fallback to com.apple.iokit.IOHIDFamily succeeded. Please fix your kext!`

marianopela, 2020-11-10:
> `Dependency com.apple.iokit.IOHIDSystem fallback to com.apple.iokit.IOHIDFamily succeeded. Please fix your kext!`

This one I solved, and I realized my hang problems could be related to GPRW (or just my ACPI patches in general) so I need to rework the fix for POWS, which I previously solved through DSDT, into an SSDT. 
Here are the very last original [ACPIs](https://github.com/zhen-zen/YogaSMC/files/5516594/ACPI.zip). The error is POWS is not found, and to solved I declared it like so
[image]

Is it right? How could I do it with an SSDT?

zhen-zen, 2020-11-11:
> Latest YogaSMCAlter is booting fine. I now noticed a line in the preverbose of OC
> `Dependency com.apple.iokit.IOHIDSystem fallback to com.apple.iokit.IOHIDFamily succeeded. Please fix your kext!`

Oh, I forgot to remove it from alter one. 

> This one I solved, and I realized my hang problems could be related to GPRW (or just my ACPI patches in general) so I need to rework the fix for POWS, which I previously solved through DSDT, into an [...]

No. DON'T insert anything into field declaration as the address are fixed for each variable. Just create an empty `Name (POWS, Zero)` and leave it along.

marianopela, 2020-11-11:
> No. DON'T insert anything into field declaration as the address are fixed for each variable. Just create an empty `Name (POWS, Zero)` and leave it along.

Seems to be solved with this
[image]

I tried updating to latest release. Hotkeys are recognised fine, so this is the situation
- random hangs like this
[image]
- Pane not working 
[image]
- No logging at all with `log stream --predicate 'processID=0 && (senderImagePath contains "YogaSMC" OR senderImagePath contains "ACPI")'`

marianopela, 2020-11-13:
Good news! Figured out some things

> * random hangs like this (usually when I update YogaSMC.kext and NC)

Not always a problem. Only happens with debug versions
> * Pane not working

Solved by disabling SIP
> * No logging at all with `log stream --predicate 'processID=0 && (senderImagePath contains "YogaSMC" OR senderImagePath contains "ACPI")'`

This one I could not figure out

Apart from this, conservation mode is working at last! I'm on release from last commit

[image]

Now two points remain unclear

- I'm using BrightnessKeys, and it's working greatly by itself, but it seems not to like YogaSMC very much. In fact, when loading both of them in the config, brightness keys are vary unresponsive, and they block after a few keystrokes. If I could read the log it would probably show a timeout I guess.

- I cannot enable RapidChargeMode, even though the option is present in the laptop advanced BIOS (and I enabled it today). This is not really a problem, just wanted to let you know

Thanks again for your really great work

zhen-zen, 2020-11-14:
> > * random hangs like this (usually when I update YogaSMC.kext and NC)
> 
> ~Not always a problem. Only happens with debug versions~ Nevermind. It just happened on release version

It's quite weird that there would be difference between debug / release version.

> > * Pane not working
> 
> Solved by disabling SIP

I disabled SIP by default for kext development so I didn't recognized it could be an issue.However, it should still work. Later I will enable that and do some experiments.

> > * No logging at all with `log stream --predicate 'processID=0 && (senderImagePath contains "YogaSMC" OR senderImagePath contains "ACPI")'`
> 
> This one I could not figure out

This seems to be a common issue for kexts injected by bootloader since Big Sur. However, it's still available from `dmesg`. You can increase its buffer size with https://github.com/acidanthera/DebugEnhancer.

> Apart from this, conservation mode is working at last! I'm on release from last commit
> 
> [image]

Maybe that's SIP issue?

> Now two points remain unclear
>
> * I'm using BrightnessKeys, and it's working greatly by itself, but it seems not to like YogaSMC very much. In fact, when loading both of them in the config, [...]

YogaSMC will actively poll EC regions for sensor readings. And notification to brightness keys are also received from EC. Maybe there could be potential conflict. Can you try the Alter version? Also, how about other hotkeys?

> * I cannot enable RapidChargeMode, even though the option is present in the laptop advanced BIOS (and I enabled it today). This is not really a problem, just wanted to let you know

I left that checkbox disabled since it's not possible to tell whether this function is supported. And this method is not well documented in Linux. I can enabled it in debug version of prefpane.

marianopela, 2020-11-15:
> It's quite weird that there would be difference between debug / release version.

There isn't. I was mislead by my laptop behaving strange

> This seems to be a common issue for kexts injected by bootloader since Big Sur. However, it's still available from `dmesg`. You can increase its buffer size with https://github.com/acidanthera/DebugEnhancer.

I will look into it, thanks 
> > Apart from this, conservation mode is working at last! I'm on release from last commit
> > [image]
> 
> Maybe that's SIP issue?

Could be. I will look into that as well, after we figure out those hang problems
> YogaSMC will actively poll EC regions for sensor readings. And notification to brightness keys are also received from EC. Maybe there could be potential conflict. Can you try the Alter version? Also, how about other hotkeys?

There seems to be a problem with brightness on my machine, and it's showing even if I'm not using YogaSMC (as a matter or fact, other hotkeys have no problem at all). There is a strange correlation I can't see between the way I configure the trackpad, XOSI, and the way I configure brightness. I've been using @lietxia 's configuration for brightness (EFI [here](https://github.com/lietxia/XiaoXinAir14IML_2019_hackintosh)), and only recently switched to BrightnessKeys, but there have always been problems

> I left that checkbox disabled since it's not possible to tell whether this function is supported. And this method is not well documented in Linux. I can enabled it in debug version of prefpane.

Thanks, I tried that but the setting seems to not be sticking (as it happened with ConservationMode before, even though that one is working now), but I don't really want to focus on this because of my unstable situation and what discussed above

marianopela, 2020-11-15:
Ok, I restarted fresh and moved one step at a time. Boot problems arise when adding XOSI patch (ACPI errors and hangs)

marianopela, 2020-11-15:
Here is what happens with BrightnessKeys. I have no YogaSMC at the moment, to isolate brightness issues
Basically after a few keystrokes they stop working, then keyboard stops working and the whole system stops responding. 
[BrightnessKeys.mov.zip](https://github.com/zhen-zen/YogaSMC/files/5542948/BrightnessKeys.mov.zip)

Here is the [EFI.zip](https://github.com/zhen-zen/YogaSMC/files/5542949/EFI.zip)

zhen-zen, 2020-11-16:
How was that with the old `XQXX` approach? I will later check the efi and the video later when I got my laptop.

marianopela, 2020-11-16:
> How was that with the old `XQXX` approach? I will later check the efi and the video later when I have my laptop.

Just the same. The only way to not to run into this problem was [this](https://github.com/lietxia/XiaoXinAir14IML_2019_hackintosh/issues/2#issuecomment-612587939) by Lietxia, but it's not working with XOSI

zhen-zen, 2020-11-16:
You don't need that superfluous I2C SSDT with latest VoodooI2C and `XOSI`.

marianopela, 2020-11-16:
Right, in fact I got it to work with interrupts a few days ago by setting GPHD to Zero

Wow, now that I reapplied GPI0 patch and rebooted a few times, brightness seems to keep working and system doesn't freeze anymore. I'll try disabling `XQXX` renames and give BrightnessKeys another try. I'll let you know in a few minutes

marianopela, 2020-11-16:
Ok, all seems fine. BrightnessKeys is working properly, trackpad too. I will now add YogaSMC and see what happens

marianopela, 2020-11-16:
Inserted YogaSMC, updated NC and pane. At first reboot it was stuck at boot. At second reboot it started, and now brightness keys are "lagging", kinda unresponsive, so we're back at this
> YogaSMC will actively poll EC regions for sensor readings. And notification to brightness keys are also received from EC. Maybe there could be potential conflict. Can you try the Alter version? [...]

I will retry Alter version. Also, other hotkeys are fine and responsive as usual

marianopela, 2020-11-16:
Something strange is now happening. If I put device to sleep (while on charge with Conservation Mode on), when I come back and open the lid I have black screen, and nothing seems to work. So I force shutdown and next boot keyboard is not working (not even in picker or bios) and I need to unplug and plug the battery back in. (This also happened sometimes in the past on another Ideapad device). 

I analysed pmset log and found this [sleep.txt](https://github.com/zhen-zen/YogaSMC/files/5547075/sleep.txt)

zhen-zen, 2020-11-16:
You can remove `msgbuf=1048576` if you have [DebugEnhancer](https://github.com/acidanthera/DebugEnhancer), but they should be the same and apply for both kexts.

And unfortunately, the `dmesg` from macOS don't support arguments like `-w`. And iirc `watch` didn't act as expected when the content is filtered.

There could be some issues with EC when it's polled actively for sensor readings (per 2sec by default), which is the case for non-Alter one. Currently I'm unable to find possible workarounds.

marianopela, 2020-11-16:
> Btw, can you provide related logs when the hotkey is not responsive? I wonder if there are more retries or just late notifications.

```[  190.081771]: YSMC - Debug: IdeaVPC::VPC0 read VPC EC result: 0x1010 1
[  192.138175]: YSMC - Debug: IdeaVPC::VPC0 Brightness up? 0x0 -> 0x10
[  192.138199]: YSMC - Debug: IdeaVPC::VPC0 Fn+Space keyboard backlight?
[  192.626557]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
[  192.626582]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  192.790837]: YSMC - Debug: IdeaVPC::VPC0 read VPC EC result: 0x1010 1
[  192.792851]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[  192.792865]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  193.007309]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[  193.007325]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  193.302187]: AcpiOsIdleQueue exhausted
[  193.302210]: [_Q12] @0002B #0086:  [_Q12] @0002B #0086:  NotifyYSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
[  193.302246]: Notify ( (^^^^^^YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  193.302274]: GGFFXX00..DDDD11FF, , 0x860x86)) // Device-Specific // Device-Specific
[  193.302341]: 
[  193.302346]: 
[  193.302348]: 
[  193.302353]: No Local Variables are initialized for method [_Q12]
[  193.302358]: No Local Variables are initialized for method [_Q12]
[  193.302366]: 
[  193.302368]: 
[  193.302372]: No Arguments are initialized for method [_Q12]
[  193.302377]: No Arguments are initialized for method [_Q12]
[  193.302384]: 
[  193.302387]: 
[  193.302400]: ACPI Error: ACPI Error: Method parse/execution failed Method parse/execution failed [\134_SB.PCI0.LPCB.EC0._Q12] (Node ffffff9377cf8b40)[\134_SB.PCI0.LPCB.EC0._Q12] (Node ffffff9377cf8b40), AE_NO_MEMORY, AE_NO_MEMORY (20160930/psparse-632)
[  193.302443]:  (20160930/psparse-632)
[  193.516736]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
[  193.516761]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  193.517685]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[  193.517701]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  193.722006]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[  193.722031]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  194.237874]: AcpiOsIdleQueue exhausted
[  194.237895]: [_Q12] @0003C #0086:  [_Q12] @0003C #0086:  NotifyNotify ( (VVPPCC00, , 0x800x80)) // Status Change // Status Change
[  194.237964]: 
[  194.237968]: 
[  194.237971]: 
[  194.237976]: No Local Variables are initialized for method [_Q12]
[  194.237981]: No Local Variables are initialized for method [_Q12]
[  194.237989]: 
[  194.237992]: 
[  194.237996]: No Arguments are initialized for method [_Q12]
[  194.238000]: No Arguments are initialized for method [_Q12]
[  194.238008]: 
[  194.238010]: 
[  194.238022]: ACPI Error: ACPI Error: Method parse/execution failed Method parse/execution failed [\134_SB.PCI0.LPCB.EC0._Q12] (Node ffffff9377cf8b40)[\134_SB.PCI0.LPCB.EC0._Q12] (Node ffffff9377cf8b40), AE_NO_MEMORY, AE_NO_MEMORY (20160930/psparse-632)
[  194.238066]:  (20160930/psparse-632)
[  194.242276]: AcpiOsIdleQueue exhausted
[  194.242297]: [_Q12] @0003C #0086:  [_Q12] @0003C #0086:  NotifyNotify ( (VVPPCC00, , 0x800x80)) // Status Change // Status Change
[  194.242380]: 
[  194.242383]: 
[  194.242385]: 
[  194.242388]: No Local Variables are initialized for method [_Q12]
[  194.242391]: No Local Variables are initialized for method [_Q12]
[  194.242410]: 
[  194.242412]: 
[  194.242415]: No Arguments are initialized for method [_Q12]
[  194.242418]: No Arguments are initialized for method [_Q12]
[  194.242423]: 
[  194.242425]: 
[  194.242430]: ACPI Error: ACPI Error: Method parse/execution failed Method parse/execution failed [\134_SB.PCI0.LPCB.EC0._Q12] (Node ffffff9377cf8b40)[\134_SB.PCI0.LPCB.EC0._Q12] (Node ffffff9377cf8b40), AE_NO_MEMORY, AE_NO_MEMORY (20160930/psparse-632)
[  194.242458]:  (20160930/psparse-632)
[  194.246040]: YSMC - Debug: IdeaVPC::VPC0 empty EC event: 1
[  194.439153]: YSMC - Debug: IdeaVPC::VPC0 read VPC EC result: 0x10 1
[  194.444329]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[  194.444351]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  194.558586]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
[... 114 lines cut ...]
[  198.864310]: Restarting poll type 1
[  199.098633]: [IGFB][LOG  ][DISPLAY   ] [Set_attribute] (attribute='bksv', value=9240)
[  199.124944]: Failed to read nominal voltage rating key     . voltage:0 rc:0x84
[  199.224982]: SmartBattery: finished polling type 1
[  199.579103]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[  199.579130]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  199.581543]: YSMC - Debug: IdeaVPC::VPC0 read VPC EC result: 0x1000 0
[  199.581576]: YSMC - Debug: IdeaVPC::VPC0 Fn+Space keyboard backlight?
[  199.587098]: YSMC - Debug: IdeaVPC::VPC0 read VPC EC result: 0x1000 4
[  199.587113]: YSMC - Debug: IdeaVPC::VPC0 Fn+Space keyboard backlight?
[  199.590306]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
[  199.590320]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  199.596044]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[  199.596057]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  199.596579]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[  199.596593]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  199.597006]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[  199.597039]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  199.597516]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[  199.597575]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
```

Here you are. So there are some new ACPI errors I see, along with the expected timeouts

I attach the current [EFI.zip](https://github.com/zhen-zen/YogaSMC/files/5550183/EFI.zip) as a reference

zhen-zen, 2020-11-16:
Can you try #62? The brightness check is skipped by default now.

marianopela, 2020-11-16:
Man you're fast. I sure can

marianopela, 2020-11-16:
I'm installing Xcode atm. Is it normal for the building process to be queued?

marianopela, 2020-11-16:
```A: 2YSMC - Debug: IdeaVPC::VPC0 read VPC EC result: 0x10 1
[  101.319435]: Sandbox: 8 duplicate reports for ContainerMetadat deny(1) user-preference-read kCFPreferencesAnyApplicationSandbox: routined(350) deny(1) mach-lookup com.apple.Maps.MapsSync.store[IGFB][LOG  ][DISPLAY   ] [Set_attribute] (attribute='bksv', value=12713)
[  106.614165]: AcpiOsIdleQueue exhausted
[  106.614173]: [_Q11] @00036 #0086:  [_Q11] @00036 #0086:  NotifyNotify ( (VVPPCC00, , 0x800x80)) // Status Change // Status Change
[  106.614197]: 
[  106.614198]: 
[  106.614199]: 
[  106.614200]: No Local Variables are initialized for method [_Q11]
[  106.614201]: No Local Variables are initialized for method [_Q11]
[  106.614203]: 
[  106.614204]: 
[  106.614205]: No Arguments are initialized for method [_Q11]
[  106.614206]: No Arguments are initialized for method [_Q11]
[  106.614207]: 
[  106.614208]: 
[  106.614210]: ACPI Error: ACPI Error: Method parse/execution failed Method parse/execution failed [\134_SB.PCI0.LPCB.EC0._Q11] (Node ffffff93785f0ae0)[\134_SB.PCI0.LPCB.EC0._Q11] (Node ffffff93785f0ae0), AE_NO_MEMORY, AE_NO_MEMORY (20160930/psparse-632)
[  106.614221]:  (20160930/psparse-632)
[  107.481856]: AcpiOsIdleQueue exhausted
[  107.481862]: [_Q04] @00008 #0086:  [_Q04] @00008 #0086:  NotifyNotify ( (VVPPCC00, , 0x800x80)) // Status Change // Status Change
[  107.481879]: 
[  107.481880]: 
[  107.481881]: 
[  107.481882]: No Local Variables are initialized for method [_Q04]
[  107.481883]: No Local Variables are initialized for method [_Q04]
[  107.481885]: 
[  107.481885]: 
[  107.481886]: No Arguments are initialized for method [_Q04]
[  107.481887]: No Arguments are initialized for method [_Q04]
[  107.481889]: 
[  107.481890]: 
[  107.481891]: ACPI Error: ACPI Error: Method parse/execution failed Method parse/execution failed [\134_SB.PCI0.LPCB.EC0._Q04] (Node ffffff93785eeed0)[\134_SB.PCI0.LPCB.EC0._Q04] (Node ffffff93785eeed0), AE_NO_MEMORY, AE_NO_MEMORY (20160930/psparse-632)
[  107.481907]:  (20160930/psparse-632)
[  107.982224]: AcpiOsIdleQueue exhausted
[  107.982237]: [_Q11] @00025 #0086:  [_Q11] @00025 #0086:  NotifyNotify ( (^^^^^^GGFFXX00..DDDD11FF, , 0x870x87)) // Device-Specific // Device-Specific
[  107.982266]: 
[  107.982272]: 
[  107.982273]: 
[  107.982274]: No Local Variables are initialized for method [_Q11]
[  107.982275]: No Local Variables are initialized for method [_Q11]
[  107.982277]: 
[  107.982278]: 
[  107.982279]: No Arguments are initialized for method [_Q11]
[  107.982280]: No Arguments are initialized for method [_Q11]
[  107.982281]: 
[  107.982282]: 
[  107.982284]: ACPI Error: ACPI Error: Method parse/execution failed Method parse/execution failed [\134_SB.PCI0.LPCB.EC0._Q11] (Node ffffff93785f0ae0)[\134_SB.PCI0.LPCB.EC0._Q11] (Node ffffff93785f0ae0), AE_NO_MEMORY, AE_NO_MEMORY (20160930/psparse-632)
[  107.982294]:  (20160930/psparse-632)
[  108.624626]: [IGFB][LOG  ][DISPLAY   ] [Set_attribute] (attribute='bksv', value=6750)
[  108.962736]: AcpiOsIdleQueue exhausted
[  108.962745]: [_Q11] @00025 #0086:  [_Q11] @00025 #0086:  NotifyNotify ( (^^^^^^GGFFXX00..DDDD11FF, , 0x870x87)) // Device-Specific // Device-Specific
[  108.962774]: 
[  108.962775]: 
[  108.962776]: 
[  108.962777]: No Local Variables are initialized for method [_Q11]
[  108.962778]: No Local Variables are initialized for method [_Q11]
[  108.962780]: 
[  108.962780]: 
[  108.962781]: No Arguments are initialized for method [_Q11]
[  108.962782]: No Arguments are initialized for method [_Q11]
[  108.962784]: 
[  108.962784]: 
[  108.962787]: ACPI Error: ACPI Error: Method parse/execution failed Method parse/execution failed [\134_SB.PCI0.LPCB.EC0._Q11] (Node ffffff93785f0ae0)[\134_SB.PCI0.LPCB.EC0._Q11] (Node ffffff93785f0ae0), AE_NO_MEMORY, AE_NO_MEMORY (20160930/psparse-632)
[  108.962797]:  (20160930/psparse-632)
[  109.957166]: AcpiOsIdleQueue exhausted
[  109.957173]: [_Q11] @00025 #0086:  [_Q11] @00025 #0086:  NotifyNotify ( (^^^^^^GGFFXX00..DDDD11FF, , 0x870x87)) // Device-Specific // Device-Specific
[  109.957202]: 
[  109.957203]: 
[  109.957204]: 
[  109.957205]: No Local Variables are initialized for method [_Q11]
[  109.957206]: No Local Variables are initialized for method [_Q11]
[... 441 lines cut ...]
[  215.876993]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[  215.877017]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  215.883339]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
[  215.883353]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 5
[  216.603492]: ARPT: 216.603491: txpkt (MPDU) Complete
[  216.603504]: ARPT: 216.603503: FrameID: 0x0a13   Seq: 0x0334   TxStatus: 0x010b
[  216.603518]: ARPT: 216.603517: ACK 0 IM 0 PM 1 Suppr 0 (None)
[  216.603523]: ARPT: 216.603523: CNT(rts_tx)=0 CNT(frag_tx_cnt)=15 CNT(cts_rx_cnt)=0
[  216.603529]: ARPT: 216.603528: DequeueTime: 0xaaaaaaaa LastTxTime: 0x09ac0483 PHYTxErr:   0x0000 RxAckRSSI: 0x00aa RxAckSQ: 0x00aa
[  216.603548]: ARPT: 216.603548: Raw[0]    1 Valid
[  216.603555]: ARPT: 216.603555: [2]    0 IM
[  216.603559]: ARPT: 216.603559: [3]    1 PM
[  216.603563]: ARPT: 216.603563: [7-4]  0 Suppr
[  216.603567]: ARPT: 216.603567: [14:8] 1 Ncons
[  216.603571]: ARPT: 216.603571: [15]   0 Acked
[  216.603576]: ARPT: 216.603576: txpktpend AC_BK 0 AC_BE 0 AC_VI 0 AC_VO 1 BCMC 0 ATIM 0
[  217.889010]: YSMC - Debug: IdeaVPC::VPC0 empty EC event: 2
[  217.891532]: YSMC - Debug: IdeaVPC::VPC0 empty EC event: 3
[  217.892169]: YSMC - Debug: IdeaVPC::VPC0 empty EC event: 2
[  217.895093]: YSMC - Debug: IdeaVPC::VPC0 empty EC event: 2
[  218.929066]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
[  218.929091]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC 26
[  218.930601]: YSMC - Debug: IdeaVPC::VPC0 empty EC event: 27
[  218.931152]: YSMC - Debug: IdeaVPC::VPC0 empty EC event: 26
```

zhen-zen, 2020-11-17:
Hmmm, let me rebase some past patches there.

zhen-zen, 2020-11-17:
Queue in Github Actions seems to be delayed a bit today.

marianopela, 2020-11-17:
> Hmmm, let me rebase some past patches there.

That's good. Still lacking behind when "spamming" brightness keys, but no ACPI errors whatsoever

```[   93.760604]: Sandbox: swcd(390) deny(1) system-fsctl _IO('h', 47)YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
[   94.451587]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC2 after 2 attempts
[   94.451595]: YSMC - Debug: IdeaVPC::VPC0 read VPC EC result: 0x10 2
[   94.611944]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
[   94.611952]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC2 after 1 attempts
[   94.611954]: YSMC - Debug: IdeaVPC::VPC0 read VPC EC result: 0x10 1
[   94.736020]: Sandbox: routined(340) deny(1) mach-lookup com.apple.Maps.MapsSync.storeYSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[   95.015217]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC1 after 1 attempts
[   95.023380]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[   95.023387]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC1 after 1 attempts
[   95.435962]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[   95.435969]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC1 after 1 attempts
[   95.437050]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[   95.437056]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC1 after 1 attempts
[   95.439124]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x1a
[   95.439130]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC2 after 1 attempts
[   95.439133]: YSMC - Debug: IdeaVPC::VPC0 read VPC EC result: 0x10 1
[   96.447153]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[   96.447160]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC1 after 2 attempts
[   98.130896]: tx_flush:1074: disk1 xid 64965 tx stats: # 120 finish 125 enter 958 wait 1 2158us close 5us flush 885us
[   98.785791]: Sandbox: 1 duplicate report for routined deny(1) mach-lookup com.apple.Maps.MapsSync.storeSandbox: ContainerMetadat(472) deny(1) user-preference-read kCFPreferencesAnyApplicationYSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[   99.322979]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC1 after 2 attempts
[   99.326643]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[   99.326650]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC1 after 2 attempts
[   99.957203]: ARPT: 99.957202: txpkt (MPDU) Complete
[   99.957217]: ARPT: 99.957217: FrameID: 0x4803   Seq: 0x01e5   TxStatus: 0x010b
[   99.957227]: ARPT: 99.957227: ACK 0 IM 0 PM 1 Suppr 0 (None)
[   99.957231]: ARPT: 99.957230: CNT(rts_tx)=0 CNT(frag_tx_cnt)=15 CNT(cts_rx_cnt)=0
[   99.957234]: ARPT: 99.957234: DequeueTime: 0xaaaaaaaa LastTxTime: 0x02cb2600 PHYTxErr:   0x0000 RxAckRSSI: 0x00aa RxAckSQ: 0x00aa
[   99.957247]: ARPT: 99.957247: Raw[0]    1 Valid
[   99.957252]: ARPT: 99.957251: [2]    0 IM
[   99.957254]: ARPT: 99.957254: [3]    1 PM
[   99.957257]: ARPT: 99.957257: [7-4]  0 Suppr
[   99.957260]: ARPT: 99.957259: [14:8] 1 Ncons
[   99.957262]: ARPT: 99.957262: [15]   0 Acked
[   99.957265]: ARPT: 99.957265: txpktpend AC_BK 0 AC_BE 0 AC_VI 0 AC_VO 1 BCMC 0 ATIM 0
[  100.030316]: Sandbox: 8 duplicate reports for ContainerMetadat deny(1) user-preference-read kCFPreferencesAnyApplicationSandbox: mobileassetd(240) deny(1) file-write-create /private/var/root/Library/Caches/mobileassetdYSMC - Debug: IdeaVPC::VPC0 read VPC EC result: 0x1010 0
[  100.379497]: YSMC - Debug: IdeaVPC::VPC0 Fn+Space keyboard backlight?
[  100.379946]: YSMC - Debug: IdeaVPC::VPC0 read VPC EC result: 0x1010 1
[  100.379951]: YSMC - Debug: IdeaVPC::VPC0 Fn+Space keyboard backlight?
[  100.909306]: Sandbox: routined(340) deny(1) mach-lookup com.apple.Maps.MapsSync.storeYSMC - Debug: IdeaVPC::VPC0 read VPC EC result: 0x1010 1
[  102.843179]: YSMC - Debug: IdeaVPC::VPC0 Fn+Space keyboard backlight?
[  103.334150]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[  103.334157]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC1 after 1 attempts
[  104.853452]: [IGFB][LOG  ][DISPLAY   ] [Set_attribute] (attribute='bksv', value=4980)
[  105.295131]: YSMC - Info: IdeaVPC::VPC0 ReadEC timeout 0x10
[  105.295138]: YSMC - Info: IdeaVPC::VPC0 Failed to read VPC1 after 1 attempts
```

zhen-zen, 2020-11-18:
The `Failed to read VPC2` one could be omitted since only `VPC1` register is critical. 
When brightness +/- is pressed, your laptop's EC will send both event to `BrightnessKeys` and `IdeaVPC`. However, it could not be distinguished whether it's a brightness event (`result: 0x10`) or other hotkeys until reading those registers. I wonder if each brightness key and hotkey is recognized by either drivers. Also, can you try debug version of `BrightnessKeys` and add `-brkeysdbg` to `boot-arg` and check which event fired first?

[27 more replies]

commit 6d6de9bf (2020-11-17): Another attempt to fix #35 (#63)
commit be3caaff (2020-12-22): IdeaVPC: only check brightness with poller (#62)

* IdeaVPC: only check brightness with poller

* Another attempt to fix #35 (#63)

* IdeaVPC: Alternative patch for VPC2 read timeout (#64)

* IdeaVPC: skip duplicate touchpad status report

* IdeaVPC: report brightness level for both keyboard backlight events

* IdeaVPC: add IODelay back

* NC: Mute 0x0C: "Keyboard Backlight" for now
closed 2020-12-22 by zhen-zen