zhen-zen/YogaSMC #47: Incorrect graphic notification - F8 Wireless Toggle

H15teve, 2020-11-05:
Device: X1 Carbon Gen 6
When turning on/off F8 wireless toggle, the icon indicates "Wireless Off" all the time.

zhen-zen, 2020-11-05:
Both wifi and bluetooth are OS level implementations. I just tried that action and nothing's going wrong.
@tylernguyen do you have the same issue?

tylernguyen, 2020-11-05:
@zhen-zen 

No, the toggle works perfectly for me, switching between On and Off

zhen-zen, 2020-11-05:
Many thanks. Btw I wonder which wireless card are you using? I don't know about the api compatibility of itlwm.

tylernguyen, 2020-11-05:
> Many thanks. Btw I wonder which wireless card are you using? I don't know about the api compatibility of itlwm.

I'm using the native Broadcom card.

H15teve, 2020-11-05:
> Many thanks. Btw I wonder which wireless card are you using? I don't know about the api compatibility of itlwm.

hmmm，my wireless card is DW1830

zhen-zen, 2020-11-05:
And did the wifi turn on / off actually work despite the OSD prompt?

H15teve, 2020-11-06:
> And did the wifi toggle actually work despite the OSD prompt?

Yes I confirm. Though sometimes through a frequent switching my wireless card appears to be failed, OSD does indicate the failure.

zhen-zen, 2020-11-06:
Can you try #49 ? The prebuilt binaries are in GitHub Actions.

H15teve, 2020-11-06:
> Can you try #49 ? The prebuilt binaries are in GitHub Actions.

This time it works properly. Thanks for keep tracking^_^

closed 2020-11-06 by H15teve