zhen-zen/YogaSMC #51: Mute button on after sleep on X1C6

buyddy, 2020-11-06:
From wakeup, LED of Mute is on. The volume status is reversed, and there is no sound. Use Volume up and down will fix the issue. Confirmed by multiple users on X1C6. Turn off LED Automation for Mute seems to solve the problem.

SukkaW, 2020-11-06:
Same on E480. But unlike X1C6, due to a bug in E480's EC, there is no way to fix the sound.

The issue could be reproduced by entering S3 sleep and then wake from it.

zhen-zen, 2020-11-06:
Without LED Automation for mute one, will that LED turned off during sleep?

buyddy, 2020-11-07:
> Without LED Automation for mute one, will that LED be turned off during sleep?

Yes, it will turn off during sleep and automatically turn back on after resume. So I don't think the automation is needed.

zhen-zen, 2020-11-07:
OK, I will comment out that part. Is micMute one also not required?

buyddy, 2020-11-07:
> OK, I will comment out that part. Is micMute one also not required?

I can further confirm that the FnLock/ESC LED does not need it too. The MicMute does not work either way, however (won't light up with/without automation). I am using the most recent released version. So I am not sure if there is anything else that changed.

buyddy, 2020-11-07:
> OK, I will comment out that part. Is micMute one also not required?

I can confirm that when MicMute is on, and go to sleep, if Automation is checked for MicMute LED, the Mute button will turn on instead after resume. I think there is a bug for that.

zhen-zen, 2020-11-07:
I didn't get why MicMute will affect Mute one. It's a bit confusing. What's their behavior without the kext?

buyddy, 2020-11-07:
> I didn't get why MicMute will affect Mute one. It's a bit confusing. What's their behavior without the kext?

I don't think it is related to the kext, could be just a bug to turn on the "Mute" LED instead of the "MuteMic" LED when restoring. Again I am using the released version. So not sure if you had made changes already.

zhen-zen, 2020-11-07:
Released version... Alright the UI one is fixed in #36. Maybe I need weekly releases?

buyddy, 2020-11-07:
@zhen-zen I have download and compile the latest version. I think the UI issue is gone. I did discover a small issue. When the Mute is on at boot. The sound is muted but the Volume icon on top is not shown as muted. Then if I press the Mute button again, the Volume icon will show as muted, but still no sound. Basically Volume icon and the Mute button will be reversed of each other. I will have to use Volume up or Volume down to resolve it.

junaedahmed, 2020-11-07:
> @zhen-zen I have download and compile the latest version. I think the UI issue is gone. I did discover a small issue. When the Mute is on at boot. The sound is muted but the Volume icon on top is [...]

I can confirm this is a issue with think variants. I think for think variants Vol Mute meddling should altogether be left out as it  works fine without any modification.

zhen-zen, 2020-11-08:
Just updated the kext, please try it out.

buyddy, 2020-11-08:
> Just updated the kext, please try it out.

The issue with boot on mute seems to still be there with the new kext. The sleep issue is fixed though.

SukkaW, 2020-11-08:
> The issue with boot on mute seems to still be there with the new kext.

Try boot into Windows, unmute, then perform a cold reboot back to macOS (a cold reboot means shutdown then start).

zhen-zen, 2020-11-08:
I wonder whether the issue presents without the kext, since now it should not change mute status on boot.

buyddy, 2020-11-08:
> > The issue with boot on mute seems to still be there with the new kext.
> 
> Try boot into Windows, unmute, then perform a cold reboot back to macOS (a cold reboot means shutdown then start).
I think if I boot unmute everything is fine.

buyddy, 2020-11-08:
Looks like the kext is not the problem. It also has the same issue without the kext. Maybe you can read the Mute status and set it just on boot?

> I wonder whether the issue presents without the kext, since now it should not change mute status on boot.

zhen-zen, 2020-11-09:
Feel free to open a PR since I don't own any Thinkpads. It should be an easy userland fix like the [mic mute led](https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMCUtils/ScriptHelper.swift#L58) one.

zhen-zen, 2020-12-08:
Please correct me if I have misunderstood some thing.

On X1C6 @buyddy and T460 @junaedahmed, Mute LED is always on after wakeup while it was correctly turned off during sleep. So If the volume is unmuted, it's necessary to press the button twice (Unmute: on -> Mute: on -> Unmute: off). Currently it's the only issue regarding mute button with this model. This issue seems to affect all Think variants.

Tentative solution: check volume status and set it on every wakeup in NC.

On E480, although @SukkaW claims the problem could not be fixed for EC issue, in #82 the button works on @notiflux ones after corresponding `XOSI` fixes. With `Linux` in that list, `LNUX` will be set to 0x01 and execute `SAUM (0x02)` and `\UCMS (0x1C)` when initializing. The first command is known to set mute button mode to `TP_EC_MUTE_BTN_NONE` while the usage for the second one is still unclear. The first operation is executed in linux but was disabled in #53 . Maybe it's better to set it like `SSDT-DYTC-Tx80.dsl` and I would like to merge those XOSI related names as default into `SSDT-THINK.dsl`. CC @benbender

For LED synchronization, E480 have no control over the LED. Without the driver, Mute LED will never be turned on.

Tentative solution: as an option for NC, listen to all volume change events and set the Mute LED correspondingly.

zhen-zen, 2020-12-09:
Just remembered there's a helper in Linux: https://github.com/torvalds/linux/blob/master/sound/pci/hda/thinkpad_helper.c
I'm no sure when it will be triggered. It's applied for all Conexant ones but use a whitelist for Realtek. Can you report your codec id (HDEF/.../IOHDACodecDevice) and subsystem id (HDEF/.../IOHDACodecFunction)?

tylernguyen, 2020-12-09:
@zhen-zen 

For my x1c6:
[image]
[image]

notiflux, 2020-12-09:
mine look like this:
[image]
[image]

zhen-zen, 2020-12-09:
For X1C6, the workaround is unnecessary since the subsystem id is not included for Realtek codec. Please try the linked PR for the first issue.

However E480 has Conexant codec, where the workaround seems necessary. I'm still working on it.

junaedahmed, 2020-12-09:
> Please correct me if I have misunderstood some thing.
>
> On X1C6 @buyddy and T460 @junaedahmed, Mute LED is always on after wakeup while it was correctly turned off during sleep. So If the volume is [...]

@zhen-zen , The problem was for T460 was not switching VolMute on when vol is decreased to zero by Vol- button. So if I vol- to zero then push the Mute button it will demute the vol but also switch the Led on. As a result the state will be reversed. And there is no Led state surviving problem on T460. If led is on it will stay on during sleep and reboot. That works. By the way, I think your tantative solution if implemented will work for all mute led issues if handled by NC then by firmware.

zhen-zen, 2020-12-10:
T460 is in the whitelist that requires that helper. You can try the commit there as well and the function an be enabled in prefpane.

zhen-zen, 2020-12-15:
Any updates? Or I will leave this PR to the next release.

buyddy, 2020-12-15:
@zhen-zen  I just tried the fix by adding the LNUX = 0x01 into ssdt. Also compile the latest driver and app. If I disable sound before booting (mute LED is on), when entering the system, the sound will be unmuted (and LED will go off). So this seems to be working. However, now the mute LED is completely off regardless what state the mute is.

zhen-zen, 2020-12-16:
Have you enabled LED fixup in prefpane?

zhen-zen, 2020-12-16:
It seems that the volume change event is deprecated in newer OS and it will only receive notifications in "Sound" prefpane. New workaround it to monitor volume+-/mute key press for that. Other ways to change volume, such as the volume slider in menubar won't trigger the notification.

buyddy, 2020-12-16:
> Have you enabled LED fixup in prefpane?

Do you mean LED automation? I think for Thinkpad the Mute LED is disabled.

zhen-zen, 2020-12-16:
> > Have you enabled LED fixup in prefpane?
> 
> Do you mean LED automation? I think for Thinkpad the Mute LED is disabled.

In this PR, there's an option in prefpane to sync mute LED.

buyddy, 2020-12-18:
I compiled it and it seems to work fine. However I do have a question. In this branch and the newest pull from master, you have:

                if ECCap != "RW" {
                    if !getBoolean("Dual fan", conf.service),
                       defaults.bool(forKey: "SecondThinkFan") {
                        fanHelper2 = ThinkFanHelper(appMenu, conf.connect, false, false)
                        fanHelper2?.update(true)
                    }
                    fanHelper = ThinkFanHelper(appMenu, conf.connect, true, fanHelper2 == nil)
                    fanHelper?.update(true)
                } else {
                    showOSD("ECAccessUnavailable")
                }

When ECCap=='RW', the OSD will show EC Access Unavailable. This is exactly the opposite behavior as in the source code of 1.3.0 release (when ECCap!=3 then we will have the warning), and it causing the error every time I open the app. Is there a reason? Or it is just a typo.

zhen-zen, 2020-12-19:
Thanks for catching that bug!

buyddy, 2020-12-19:
My pleasure! I think the audio PR can be merged now. The mute button is working as expected with the fix.

PR #88 (2020-12-20): NC: improvements for audio status handling
Close #51
closed 2020-12-20 by zhen-zen