zhen-zen/YogaSMC #59: Improvements on fan ctrl

KirisameMarisaa, 2020-11-12:
Hi. I’m now using ThinkPad X1 Carbon 2018. in ver 1.30, we are able to surpress the fan from running, and monitor the running speed of it. However, we still cannot adjust the running speed of it. Will this feature be further implemented in the future? thanks

zhen-zen, 2020-11-13:
What do you mean by "surpass"? And current fan control mechanism follows thinkpad_acpi and should have the same result. Can you try if that works under linux?
Ref:https://github.com/torvalds/linux/blob/master/Documentation/admin-guide/laptops/thinkpad-acpi.rst

KirisameMarisaa, 2020-11-13:
> What do you mean by "surpass"? And current fan control mechanism follows thinkpad_acpi and should have the same result. Can you try if that works under linux?
> Ref:https://github.com/torvalds/linux/blob/master/Documentation/admin-guide/laptops/thinkpad-acpi.rst

It was a typo. I was meant to say “suppress”. (;;;^^)

I just cannot find any options for me to manually set fan speed from 0 to higher values...
[image]

By the way, does the panel support dark mode?

zhen-zen, 2020-11-14:
Fan speed control is supported by NC app. The prefpane will stop update once it's running. You will also need to add `WE1B` method for EC write access, see `SSDT-ECRW.dsl`.

Yes, the prefpane support dark mode.

KirisameMarisaa, 2020-11-14:
The NC app is quite confusing. After uninstalled the prefpane, I moved it to /applications, try to run it, but it just doesn't display anything, and I coundn't find any settings or icons to show its interface (if it exists). 

Also, although I enabled the Dark Mode, prefpane still display as light mode.

zhen-zen, 2020-11-14:
Because you have checked the "Hide menubar icon" option in the prefpane. It's on the left part of `General` tab.

I suppose the dark mode switch could be some SDK issues. It should be handled by the system without any modification. Does this one work? https://github.com/zhen-zen/YogaSMC/actions/runs/357071122

KirisameMarisaa, 2020-11-15:
Fan control works after a few attempts to merge the battery patch & SSDT-ECRW, Battery & FanCtrl both works properly. Thank you!
(I replaced RE1B and RECB method in the battery patch with the one in SSDT-ECRW)

[image]

However, after that, the laptop's sleep & wake became abnormal, one assertion called "RTC/UserActivity" kept on waking the machine. One workaround seems to be preventing macOS from writing/reading EC, but it will break fan ctrl functionality. 

[image]

Then by how can we fix RTC sleep issue while keep fan ctrl usable? thank you!

zhen-zen, 2020-11-16:
The app will only poll fan speed actively when the menu is open. Also those EC R/W methods is called passively. Please check if you have any ACPI errors.

KirisameMarisaa, 2020-11-17:
Moved to @tylernguyen ‘s latest EFI patch. The problem never occours, fan speed reading even can be displayed correctly in iStatsMenu, except for the built-in fan-speed controlling slider (not work). I think it‘s ok to close this issue now. Thank you!

[image]

closed 2020-11-17 by KirisameMarisaa

VinylNerd, 2021-03-16:
You will also need to add `WE1B` method for EC write access, see `SSDT-ECRW.dsl`.
>

im very confused where i can do this, i am a bit noob, could you please explain a or link to where i can learn to do this, thanks!