zhen-zen/YogaSMC #61: PrefPane not working under macOS Big Sur

kasti0, 2020-11-16:
error message translated from german:
"Error in System preferences: the system preference "YogaSMCPane" could not be loaded".

the prefpane was trusted under System> Security

marianopela, 2020-11-16:
Try disabling SIP. I have SIP enabled by OC and I disabled it via Recovery (`csrutil disable`)

kasti0, 2020-11-16:
do you need to keep it disabled?

marianopela, 2020-11-16:
> do you need to keep it disabled?
> and is it displaying anything for you?

I did nothing else and now SIP is showing as enabled (strange but also cool, as I liked it on) but pane is still working

Yes, it is displaying all the infos currently implemented and working for Ideapad devices

kasti0, 2020-11-16:
That worked!

closed 2020-11-16 by kasti0