zhen-zen/YogaSMC #65: anyway one could help with Fan read/ control on IdeaPads?

kasti0, 2020-11-17:
i tried to do it myself on windows with NBFanControl with probing the EC and examining my DSDT.
however I still couldn´t find the EC registers or methods that are related to the laptop fans.
my laptop is an IdeaPad 330s 15ik 81F5

zhen-zen, 2020-11-18:
It's not implemented yet since mine don't support that.

marianopela, 2020-11-18:
Is there anything I can do to help? I think mine supports it, as in the advanced bios there are many fan control settings

zhen-zen, 2020-11-18:
Some models have it exposed directly to EC, such as NBFC ones. And there are also `VPCCMD` commands. https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMC/YogaVPC/IdeaVPC.hpp#L50

marianopela, 2020-11-18:
Would this be something useful?

```
[    5.097431]: Clearing out battery data
[    5.097518]: SMCGetKeyInfo cmd 0x9(key     ) size 0x2
[    5.097568]: SMCGetKeyInfo cmd 0xa(key     ) size 0x2
[    5.097605]: SMC key      for cmd 0xb is not supported
[    5.097653]: SMC key      for cmd 0x76 is not supported
[    5.097692]: SMC key      for cmd 0x2 is not supported
[    5.097727]: SMC key      for cmd 0x1 is not supported
[    5.097774]: SMCGetKeyInfo cmd 0x16(key     ) size 0x2
[    5.097824]: SMC key      for cmd 0x54 is not supported
[    5.097870]: SMC key      for cmd 0x8 is not supported
[    5.097913]: SMCGetKeyInfo cmd 0x17(key     ) size 0x2
[    5.097956]: SMCGetKeyInfo cmd 0x13(key     ) size 0x2
[    5.097999]: SMCGetKeyInfo cmd 0x53(key     ) size 0x2
[    5.098043]: SMC key      for cmd 0x9c is not supported
[    5.098087]: SMC key      for cmd 0x12 is not supported
[    5.098131]: SMC key      for cmd 0x8b is not supported
[    5.098178]: SMC key      for cmd 0x21 is not supported
[    5.098222]: SMCGetKeyInfo cmd 0xf(key     ) size 0x2
[    5.098268]: SMCGetKeyInfo cmd 0x10(key     ) size 0x2
[    5.098312]: SMC key      for cmd 0x18 is not supported
```

zhen-zen, 2020-11-18:
No, they are not published as SMC keys. There are few documents on it, so I haven't implement that.

zhen-zen, 2020-12-10:
Please continue the thread at #91

closed 2020-12-10 by zhen-zen