zhen-zen/YogaSMC #67: Pane: Indicator "LED Automation" resets to "on" on reboot

benbender, 2020-11-17:
I'm playing with ACPI-S0 on my X1C6. 
Therefor I want to disable the LED Automation for "Indicator". 
It works but gets reset to "enabled" on reboot.

zhen-zen, 2020-11-18:
The NC app should save current led settings when exiting and re-apply then at start-up. Maybe you check the value in ioreg since the prefpane will not be updated until system preferences is reopened.

benbender, 2020-11-18:
Hmm, so it's expected to not apply until the prefpane is opened? That would kill it for my (test-)usecase as it will put the leds in the wrong state on sleep until the prefpane is opened after a restart - which seems unrealistic/uncomfortable.
Any chance that could change?

zhen-zen, 2020-11-18:
No. That option should be turned off once. And after quitting the NC that time, the app will save the value without this option.

benbender, 2020-11-19:
Hmm, it doesn't seem to save anything related in `defaults read org.zhen.YogaSMC`. Am I missing smth?

zhen-zen, 2020-11-19:
It should be saved along with other events here https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMCNC/AppDelegate.swift#L331

zhen-zen, 2020-12-08:
Are there any updates?

buyddy, 2020-12-15:
I was trying to manually edit org.zhen.YogaSMC.plist. I want to change the "Custom Hot Key". However, the file is overwritten every time I start the NC. It won't be changed at all.

zhen-zen, 2020-12-16:
You will need to stop it first before making any changes. The app will save current events when exiting thus overwriting yours.

buyddy, 2020-12-16:
Does the app register with the kext? I felt I have met ghost. I actually change the code to replace prefpaneAS to open another app and recompile, but my re-compiled version still open the Yoga prefpane.

zhen-zen, 2020-12-16:
No, kext will only send the event and it's the app that decide what to do next in userspace. I suppose you will need to clean current events if hardcoded new ones.