zhen-zen/YogaSMC #68: Thinkpad L390 Yoga: Trackpad and Trackpoint aren't disabled if screen is rotated in tablet mode + strange message + disabling wifi + fan reading error

antoniomcr96, 2020-11-18:
Good evening and thanks for all your work. I switched from Thinkpad assistant with all the ACPI renames and this kext works almost out of the box so you did a great job.

I wanted to report some errors:
1. If I rotate my screen in tablet mode trackpad and trackpoint aren't disabled. I use a personal keyboard remap for VoodooPS2Controller:
```
    Name (_SB.PCI0.LPCB.KBD.RMCF, Package (0x02)
    {
        "Keyboard", 
        Package (0x02)
        {
            "Custom PS2 Map", 
            Package (0x04)
            {
                Package (0x00){}, 
                "e037=64", 
```
This avoids that clicking the Print key on the keyboard disables the trackpad. I don't know if it has something to do with this problem.

2. If I rotate my screen in tablet mode I see: _event:0x60f0:0_ . Can I remove it?

3. Disabling an re-enabling wifi is buggy. Disabling WIFI works fast and well. if I re-enable it, I have to wait 5-10 sec until the message "Wireless On" appears and I'm unable to connect to Wifi network (wifi doesn't work at all). If I try to connect to a known network it asks me the password, and anyway it doesn't allow to connect. Disabling and re-enabling wifi from system preferences doesn't give this kind of problem and it's fast. This is very annoying.

4. I see "fan reading error" on boot. I installed all the SSDT files according to my EC path, but on this laptop it was impossible to read Fan also on windows or linux (I think it's impossible in any case). How can I remove that message?

Thanks in advance

zhen-zen, 2020-11-18:
1. Tablet mode detection is not implemented for think variant.
2. I'm not sure how rotate works on your laptop. But that event code should be `TP_HKEY_EV_THM_TRANSFM_CHANGED` and feel free to open a PR like #40.
3. Since the fan speed & control should all go through SMI interface, the only known approach is the `thinkpad-acpi` one. If you laptop have some kind of dust clean mode in Windows, there could be some workaround.
4. Maybe just hide the menu bar icon?
5. They should be able to control all the LEDs. Maybe yours is also an exception.

antoniomcr96, 2020-11-18:
> * Tablet mode detection is not implemented for think variant.

ok, sorry for the question.

> * I'm not sure how rotate works on your laptop. But that event code should be `TP_HKEY_EV_THM_TRANSFM_CHANGED` and feel free to open a PR like #40.

It doesn't seem difficult, I'll try. I'm not an expert unfortunately

> * Since the fan speed & control should all go through SMI interface, the only known approach is the `thinkpad-acpi` one. If you laptop have some kind of dust clean mode in Windows, there could be some workaround.
> * Maybe just hide the menu bar icon?

I like the menu bar icon. Maybe the idea could be to hide the fan-control slider if there is a fan reading problem (and **hide also the fan reading problem message on login**). For the moment I'll hide the icon.
Anyway I have never been able to control / run the fans neither on linux. I'll try with some software on windows and if it doesn't work it's surely an hardware/bios problem. Maybe I will report to Lenovo.

> * They should be able to control all the LEDs. Maybe yours is also an exception.

I don't know how my LED work except for micmute led.  If the method `_SB.PCI0.LPCB.H_EC.HKEY.MMTS` with 0x02 arg is called it should light up. Anyway at the moment unfortunately that panel for me doesn't work at all. I don't know if others have the same problem

Thanks for the fast reply!

zhen-zen, 2020-11-18:
Dust clean mode is usually provided with OEM software, i.e. Vantage.

i18n is also scheduled for next release. I have started separating those strings, but got busier recently.

antoniomcr96, 2020-11-18:
No dust clean mode on vantage for this laptop. Speed Fan doesn't recognize any fan controller but only the temperature. I will report to Lenovo, but maybe SMI interface isn't available on this laptop.

Ok, let me know if I can help. Thanks for your work. 

Pull request opened: https://github.com/zhen-zen/YogaSMC/pull/71

zhen-zen, 2020-11-21:
> No dust clean mode on vantage for this laptop. Speed Fan doesn't recognize any fan controller but only the temperature. I will report to Lenovo, but maybe SMI interface isn't available on this laptop.

What's the temperature?

> EDIT: Sorry, should this kext make the brightness keys work? Because for me they won't, so I re-enabled the renames and the SSDT for the two methods

You will need https://github.com/acidanthera/BrightnessKeys for that and check the readme.

zhen-zen, 2020-11-21:
I have just implemented i18n and yoga mode support. And in DSDT, there's no method called `LED` and `BEEP` under EC. Instead, your `_SI.SST` calls `SYSC` to set LED and system states, which its not documented. So your model might not support separate LED control and you can refer to `CSSI` method in `SSDT-THINK.dsl` to sync them during sleep / wake.

antoniomcr96, 2020-11-21:
> I have just implemented i18n and yoga mode support. And in DSDT, there's no method called `LED` and `BEEP` under EC. Instead, your `_SI.SST` calls `SYSC` to set LED and system states, which its not [...]

Okay, good to know. I'll eventually have to figure out how CSSI Method works. At the moment there seems to be no problem with the synchronization of the LEDs

Another question: at the moment `SSDT-RCSM` doesn't seem documented. It has something to do with clamshell mode, right?
Anyway, the question is: I don't have a`VPC0` device in my DSDT. Is it possible to use the SSDT somehow?
Thanks as always

zhen-zen, 2020-11-21:
Just rename it to `HKEY`.

zhen-zen, 2020-11-22:
> Okay, good to know. I'll eventually have to figure out how CSSI Method works. At the moment there seems to be no problem with the synchronization of the LEDs

I forgot to comment on this. `CSSI` is actually a proxy method to `_SI._SST` since it's not exposed to the driver. You don't need to modify that part.

zhen-zen, 2020-11-22:
Please try whether ca8ed08 works for you.

antoniomcr96, 2020-11-22:
> Please try whether [ca8ed08](https://github.com/zhen-zen/YogaSMC/commit/ca8ed089ece845d5ebf848b065fafdacf68c2100) works for you.

It works, thanks.

In Big Sur adding prefpane files often doesn't work. Even if I allow the execution of the prefpane from the "Security and Privacy" menu, it often shows a message "Unable to open YogaSMCPane.prefpane ...". The only way to fix the problem is:
1. Add the prefpane for all users;
2. Execute from terminal
`sudo xattr -d com.apple.quarantine /Library/PreferencePanes/YogaSMCPane.prefPane `
3. Prefpane correctly opens

It could be useful for other users

zhen-zen, 2020-11-22:
I suppose it's possible press alt + click the app / prefpane for quarantine.

Actually current customization is only limited to AppleScript (although it can do a lot of things). After quit the app, you can change its name and the script to execute (default one is open prefpane).

antoniomcr96, 2020-11-22:
> I suppose it's possible press alt + click the app / prefpane for quarantine.
For prefpane this doesn't seem to work. You can add the prefpane while pressing alt, but then when you try to open it from preferences it asks for permission. If you allow the execution from Security and Privacy it says "Unable to open" (11.0.1). Maybe I'm the only one but I found that command useful.

[image]
What should Autosleep do? It has something to do with keyboard backlight?

I would propose, if possible, to add an option to activate keyboard backlight when typing and deactivating it after a custom interval. In the old linux days I used this script: https://github.com/saibotd/tp-auto-kbbl and it worked very well

zhen-zen, 2020-11-22:
For prefpane this doesn't seem to work. You can add the prefpane while pressing alt, but then when you try to open it from preferences it asks for permission. If you allow the execution from Security and Privacy it says "Unable to open" (11.0.1). Maybe I'm the only one but I found that command useful.

Do you have SIP on?

> [image]
> What should Autosleep do? It has something to do with keyboard backlight?

It will turn off the keyboard backlight when entering sleep state. Or is that handled by firmware on Think variant?

> I would propose, if possible, to add an option to activate keyboard backlight when typing and deactivating it after a custom interval. In the old linux days I used this script: https://github.com/saibotd/tp-auto-kbbl and it worked very well
> I don't honestly know if macOS allows key logging

There could be kind of native emulation with virtual HID device or something like https://www.logcg.com/en/archives/2902.html. You can still open a new issue for that but recently I don't plan to implemented new user space features.

antoniomcr96, 2020-11-22:
> Do you have SIP on?

Yes.
 
> It will turn off the keyboard backlight when entering sleep state. Or is that handled by firmware on Think variant?
> 
I did some tests:
If I deselect that checkbox the keyboard backlight turns off on sleep, but doesn't turn on on wake.
If I select that checkbox the keyboard backlight turns off on sleep and then turns on on wake. So it definitely does something useful

> There could be kind of native emulation with virtual HID device or something like https://www.logcg.com/en/archives/2902.html. You can still open a new issue for that but recently I don't plan to implemented new user space features.
Don't worry, just an idea. :)

zhen-zen, 2020-11-23:
Can you try it with SIP off? I have `com.apple.quarantine` at `~/Library/PreferencePanes/YogaSMCPane.prefPane` or `/Library/PreferencePanes/YogaSMCPane.prefPane` and they both work.

zhen-zen, 2020-11-23:
Just enabled SIP and `com.apple.quarantine` is blocking now. However, in `Security & Privacy`, there's an `Open Anyway` button. Not sure when did this restriction began. There used to be a one-time prompt with Ctrl (Command) + Click.

[image]

antoniomcr96, 2020-11-23:
> Just enabled SIP and `com.apple.quarantine` is blocking now. However, in `Security & Privacy`, there's an `Open Anyway` button. Not sure when did this restriction began. There used to be a one-time prompt with Ctrl (Command) + Click.
> 
> [image]
> 
> I'm still investigating if SIP would break other functionalities.

Apparently no problems.

However if you click that “Open Anyway” for me often it doesn’t work. After allowing to open the prefpane it says “unable to load the prefpane” or something similar. 
The only way to fix it is to add the prefpane and then, without going to security and privacy, executing the command

zhen-zen, 2020-11-23:
The driver will try to parse available WMI description, which is not critical since it's just human readable entires. However, there's no indication which guid contains that information. So it will try to evaluate entries with instance_count 0x1 and flags 0x0, which is usually a large blob buffer so no argument are required. You can ignore this warning since required type shouldn't take an argument. Current approach will only try the first matched guid. I just opened #78 which would stop after found matching ones.

antoniomcr96, 2020-11-24:
Tried the build, the ACPI error disappeared so I think that it worked 👍 Thanks

Unfortunately on my Thinkpad setting **battery charge threshold** doesn't seem to work. It's a minor problem because my battery seems to have a default threshold of 95% (for MacOS it doesn't charge over this limit). Maybe a SSDT for BAT is needed, I haven't found any info on the readme though.

zhen-zen, 2020-11-24:
Please try if it works under linux. https://github.com/torvalds/linux/blob/master/Documentation/admin-guide/laptops/thinkpad-acpi.rst

zhen-zen, 2020-11-24:
And that ACPI warning should still present. Can you try the last commit?

antoniomcr96, 2020-11-27:
> And that ACPI warning should still present. Can you try the last commit there?

```
2020-11-27 21:28:54.667411+0100 0xc7       Default     0x0                  0      0    kernel: (AppleACPIPlatform) ACPI Warning: \_SB.AMW0.WQAA:
2020-11-27 21:28:54.667412+0100 0xc7       Default     0x0                  0      0    kernel: (AppleACPIPlatform) ACPI Warning: \_SB.AMW0.WQAA:
2020-11-27 21:28:54.667414+0100 0xc7       Default     0x0                  0      0    kernel: (AppleACPIPlatform) Insufficient arguments - Caller passed 0, method requires 1
2020-11-27 21:28:54.667415+0100 0xc7       Default     0x0                  0      0    kernel: (AppleACPIPlatform) Insufficient arguments - Caller passed 0, method requires 1
2020-11-27 21:28:54.667417+0100 0xc7       Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/nsarguments-333)
2020-11-27 21:28:54.667418+0100 0xc7       Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/nsarguments-333)
```
The problem is here again with https://github.com/zhen-zen/YogaSMC/actions/runs/386348029 . With https://github.com/zhen-zen/YogaSMC/actions/runs/379956818 it disappeared.

In linux I can set the battery threshold with [TLP](https://linrunner.de/tlp/)
Also in Windows it works
With YogaSMC, if I set a threshold it disappears if I reopen the app (and it doesn't actually work if I plug the AC adaptor, it always charges until 95%)

zhen-zen, 2020-11-28:
> The problem is here again with https://github.com/zhen-zen/YogaSMC/actions/runs/386348029 . With https://github.com/zhen-zen/YogaSMC/actions/runs/379956818 it was disappeared.

That's intended since it's not specified which method contains the information. And the previous commit breaks and actually it does nothing there.

> In linux I can set the battery threshold with [TLP](https://linrunner.de/tlp/)
> Also in Windows it works
> With YogaSMC, if I set a threshold it disappears if I reopen the app (and it doesn't actually work if I plug the AC adaptor, it always charges until 95%)

Please attach a screenshot for ThinkVPC and expand those battery sections

antoniomcr96, 2020-11-28:
Do you mean this?
[image]

Regarding specifying the battery, I can't understand what name/value to send with ioio. Sorry. If you could make an example, I could try.

zhen-zen, 2020-11-28:
ThinkVPC: Battery Wear control
Select battery first
`ioio -s ThinkVPC Battery 0/1/2` (will trigger an update)
`ioio -s ThinkVPC setCMstart 60`
`ioio -s ThinkVPC setCMstop 60`

zhen-zen, 2020-11-28:
Can you try the pane in https://github.com/zhen-zen/YogaSMC/pull/81 again?

antoniomcr96, 2020-11-30:
> ThinkVPC: Battery Wear control
> Select battery first
> `ioio -s ThinkVPC Battery 0/1/2` (will trigger an update)
> `ioio -s ThinkVPC setCMstart 60`
> `ioio -s ThinkVPC setCMstop 60`

Manually selecting the battery in this way works.

> Can you try the pane in #81 again?

In fact it works also in this way, thanks!!!

These days I have done some testing. There are some problems, although not very important:
- if I select the threshold, close the prefpane and reopen it, I lose the previously saved value (the app shows me 100, but the value is still set correctly);
- for some strange reason if I select, for example, 75 as the threshold the battery stops charging at 77; if I select 80 it stops charging at 83; if I select 85 it stops charging at 89. The value does not match exactly what I set

zhen-zen, 2020-11-30:
> * if I select the threshold, close the prefpane and reopen it, I lose the previously saved value (the app shows me 100, but the value is still set correctly);

Is the value in ioreg accurate?

> * for some strange reason if I select, for example, 75 as the threshold the battery stops charging at 77; if I select 80 it stops charging at 83; if I select 85 it stops charging at 89. The value does not match exactly what I set

That's inevitable since the battery firmware will calculate the stop point on its own. And this process involves wear calculation. Is it accurate under Windows?

antoniomcr96, 2020-12-01:
> > * if I select the threshold, close the prefpane and reopen it, I lose the previously saved value (the app shows me 100, but the value is still set correctly);
> 
> Is the value in ioreg accurate?

In IOReg I have:
[image]
and
[image]

or should I check something else? (I set 50% threshold in this case)

> That's inevitable since the battery firmware will calculate the stop point on its own. And this process involves wear calculation. Is it accurate under Windows?

Yes, it's not a big problem though. At least there's an explanation :)

zhen-zen, 2020-12-08:
> In IOReg I have:
> [image]
> and
> [image]
> 
> or should I check something else? (I set 50% threshold in this case)
> 
> EDIT: moreover threshold isn't automatically set on startup

Can you get current threshold setting in linux and is that persisted after reboot?

antoniomcr96, 2020-12-18:
sorry for the late reply.
I did some testing and:
- on linux the charge threshold is maintained;
- even on macos it is manteined on reboot, but it is lost if I start windows or if the battery runs out completely (so it isn't a problem of the kext)

anyway the section on the prefpane always says 0-100 and not the selected value. But it works.

zhen-zen, 2020-12-19:
Sounds like firmware issues. I'm going to add separate battery settings in prefpane after this release, maybe that could help?

antoniomcr96, 2021-01-26:
Really sorry for the delay, I had some problems.
I tried https://github.com/zhen-zen/YogaSMC/pull/101

I have lost the possibility to change thresholds:
[image]

Mic Mute (F4) doesn't work anymore

zhen-zen, 2021-01-27:
Please just use the Pane only. The Mic Mute issue is resolved later in master branch and you can use the driver from latest release. Can you attach screenshots of BAT_ANY etc in ioreg?

antoniomcr96, 2021-02-01:
Thanks. Installed latest driver and application from the release and microphone is working again.
From IOReg:
[image]
[image]

zhen-zen, 2021-02-04:
I will duplicate the dictionaries on my laptop and find out why it didn't work.

antoniomcr96, 2021-02-04:
Thanks as always. Previously it worked, then unfortunately I have not followed the project anymore, so I don’t know which update caused the problem

zhen-zen, 2021-02-09:
Please check the latest commit in #101 , now it should work for all scenarios.

antoniomcr96, 2021-02-17:
Hi zhen-zen, I tried latest release https://github.com/zhen-zen/YogaSMC/releases/tag/1.4.3 and it works. 
[image]
I'm very happy about it, thank you very much

I have to report another problem:
- on Mac OS my laptop goes in tablet mode only if rotated to the maximum (360 degrees), and it returns on laptop mode even if it's moved by a few degrees.
- on Windows it works well and the system goes in tablet mode starting from a rotation of 180 degrees and it returns normal if rotated below 180 degrees.

I can make a video if it can help you to understand better. 

Moreover I was wondering if you had an address for a small donation for all your work

zhen-zen, 2021-02-18:
Can you paste the log from debug binary when the hinge is at different degree? Usually there are 0/360 (tablet), 0-180 (laptop), 180-360 (may be separate tent / stand depending on keyboard direction).

Also thank you for the praise, but I don't have plan for that.

antoniomcr96, 2021-02-25:
I tried to install the debug  version(app + kernel extension) from the latest release. 
_log stream --predicate 'senderImagePath contains "YogaSMC"'_ doesn't show anything. 
_log stream|grep YogaSMC_ shows for example if I open preferences using the hotkey but nothing else

zhen-zen, 2021-03-01:
For Big Sur, it's only possible to grab the log from side loaded kext with [DebugEnhancer](https://github.com/acidanthera/DebugEnhancer/) and `dmesg|grep YSMC`.

antoniomcr96, 2021-03-03:
> For Big Sur, it's only possible to grab the log from side loaded kext with [DebugEnhancer](https://github.com/acidanthera/DebugEnhancer/) and `dmesg|grep YSMC`.

I'm really sorry. Now lots of things and lots of time spent to get logs of other stuff make a sense.

If I rotate the screen of the laptop this happens:
```
[  112.135163]: YSMC - Debug: ThinkVPC::HKEY tablet mode: 40003
[  112.135173]: YSMC - Debug: ThinkVPC::HKEY TopCase Input disabled
[  112.145420]: YSMC - Info: ThinkVPC::HKEY Thermal Transformation changed (GMTS)
[  114.916031]: YSMC - Debug: ThinkVPC::HKEY tablet mode: 40001
[  114.916063]: YSMC - Debug: ThinkVPC::HKEY TopCase Input enabled
[  114.927428]: YSMC - Info: ThinkVPC::HKEY Thermal Transformation changed (GMTS)
```
On Windows, somehow, keyboard and trackpad are disabled with a 180degrees rotation. With YogaSMC a complete (360°) rotation is needed

zhen-zen, 2021-03-05:
Only laptop and tablet mode (last 2 bytes) is recognized according to the log. Can you get other modes with different angles?

antoniomcr96, 2021-03-06:
Hi @zhen-zen ,
unfortunately other angles dont show anything in the logs.
I tried with Linux and, link WIndows, everything seems to be working correctly (it goes in tablet mode and disables keyboard and touchpad after a rotation 180 degrees)
Can I provide other informations useful for you?

zhen-zen, 2021-03-06:
Can you try https://github.com/zhen-zen/YogaSMC/pull/121 and find if there might be more modes in the log?

antoniomcr96, 2021-03-07:
```
[  118.594493]: YSMC - Info: ThinkVPC::HKEY failed to evaluate raw tablet mode
[  118.594509]: YSMC - Debug: ThinkVPC::HKEY TopCase Input disabled
[  118.606122]: YSMC - Info: ThinkVPC::HKEY Thermal Transformation changed (GMTS): 0x10003
[  154.397897]: YSMC - Info: ThinkVPC::HKEY failed to evaluate raw tablet mode
[  154.397928]: YSMC - Debug: ThinkVPC::HKEY TopCase Input enabled
[  154.409524]: YSMC - Info: ThinkVPC::HKEY Thermal Transformation changed (GMTS): 0x10001
```
If the laptop is rotated 360degrees. Nothing happens at 180 degrees

zhen-zen, 2021-03-09:
I just fixed a bug there and now the evaluation should not fail. Also, I wonder if you can get tablet mode recognized with HID Sensor disabled in Device Manager (Win).

antoniomcr96, 2021-03-10:
[image]
In Windows tablet mode still works when rotated at 180°

With your new kext:
```
[  153.693549]: YSMC - Debug: ThinkVPC::HKEY tablet mode: 40003, raw: 4
[  153.693593]: YSMC - Debug: ThinkVPC::HKEY TopCase Input disabled
[  153.703307]: YSMC - Info: ThinkVPC::HKEY Thermal Transformation changed (GMTS): 0x10003
[  156.970955]: YSMC - Debug: ThinkVPC::HKEY tablet mode: 40001, raw: 0
[  156.970994]: YSMC - Debug: ThinkVPC::HKEY TopCase Input enabled
[  156.982133]: YSMC - Info: ThinkVPC::HKEY Thermal Transformation changed (GMTS): 0x10001
```
No way to show other raw modes

zhen-zen, 2021-03-11:
Hmmm, then the driver in Windows should just rely on the input from `GMMS` to identify current mode.

It's weird that other modes are skipped. Can you check the value of `CMMD` (raw) at different positions? https://github.com/zhen-zen/YogaSMC#ec-reading

By the way, do you have ACPI warnings from `IOACPIPlatform` when flipping the screen? You can see it live using `log stream --predicate 'processID=0 && senderImagePath contains "ACPI"'`.

antoniomcr96, 2021-03-11:
A few updates, there are some ACPI warnings when the screen is rotated 360° and your kext detects it, nothing when it's rotated 180 degrees or other values.
```
2021-03-11 10:15:21.767637+0100 0x7de      Default     0x0                  0      0    kernel: (AppleACPIPlatform) ACPI Warning: \_SB.PCI0.LPCB.H_EC.HKEY.GMMS:
2021-03-11 10:15:21.767675+0100 0x7de      Default     0x0                  0      0    kernel: (AppleACPIPlatform) Insufficient arguments - Caller passed 0, method requires 1
2021-03-11 10:15:21.767692+0100 0x7de      Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/nsarguments-333)
2021-03-11 10:15:21.779265+0100 0x7dd      Default     0x0                  0      0    kernel: (AppleACPIPlatform) ACPI Warning: \_SB.PCI0.LPCB.H_EC.HKEY.GTMS:
2021-03-11 10:15:21.779306+0100 0x7dd      Default     0x0                  0      0    kernel: (AppleACPIPlatform) Insufficient arguments - Caller passed 0, method requires 1
2021-03-11 10:15:21.779322+0100 0x7dd      Default     0x0                  0      0    kernel: (AppleACPIPlatform)  (20160930/nsarguments-333)
```
I don't know what these two methods are, but in my dsdt they are defined as:
```
                Method (GMMS, 1, Serialized)
                {
                    Local0 = 0x00040000
                    Switch (ToInteger (\PMMD))
                    {
                        Case (0x00)
                        {
                            Local0 |= 0x01
                        }
                        Case (0x01)
                        {
                            Local0 |= 0x02
                        }
                        Case (0x02)
                        {
                            Local0 |= 0x04
                        }
                        Case (0x03)
                        {
                            Local0 |= 0x05
                        }
                        Case (0x04)
                        {
                            Local0 |= 0x03
                        }
                        Default
                        {
                            Local0 |= 0x01
                        }

                    }

                    Return (Local0)
                }

                Method (GTMS, 1, NotSerialized)
                {
                    Local0 = 0x00010000
                    Local0 |= \LVMD
                    Return (Local0)
                }
```
The argument is never required inside the method, I don't know if patching the DSDT to require no args can help.

I also tried to read EC fields, but maybe I'm doing something wrong: (EC access is R/W)
```
antonio@MacBook-Pro Release % ./ioio -s YogaVPC ReadECName CMMD
ioio: setting property 'YogaVPC:ReadECName' as string to 'CMMD'
antonio@MacBook-Pro Release % ./ioio -s YogaVPC ReadECOffSet 0xA4
ioio: setting property 'YogaVPC:ReadECOffSet' as number to 164 (0xa4)
```

zhen-zen, 2021-03-14:
Can you try `cat /sys/devices/platform/thinkpad_acpi/hotkey_tablet_mode` for different states in Linux? I don't known what else could make the mode detection other than the ACPI implementation.

antoniomcr96, 2021-04-19:
I'm really sorry for the delay. That command returns _0_ if the laptop isn't in tablet mode, instead it returns _1_ if the laptop is in tablet mode. The laptop goes in tablet mode after a rotation of 180 degrees (like in windows). If the laptop is rotated 360 degrees the result is still _1_
Thanks (as always)

zhen-zen, 2021-05-15:
Currently waiting for other ThinkPad Yoga users for feedback. The implementation should follow the linux one unless different sensor is being used.

zhen-zen, 2021-06-11:
DYTC implements several sets of functions and currently we are unable to tell which one is available. The driver follows the logic from Linux and that's the only documented source.

According to your DSDT, DYTC doesn't support DYTC_MODE_QUIET for DYTC_FUNCTION_MMC. Can you try debug kext + DebugEnhancer and record the DYTC mode in dmesg log after changing it in Windows?

antoniomcr96, 2021-06-13:
It seems that the log is the same, if I boot after selecting "performance" in windows:
```
antonio@MacBook-Pro ~ % sudo dmesg|grep DYTC
Password:
[    0.778780]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x0 result 0x40020101
[    0.778787]: YSMC - Debug: HKEY::ThinkVPC DYTCMode 0x1
```

After "Quiet":
```
[    0.747492]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x0 result 0x40020101
[    0.747498]: YSMC - Debug: HKEY::ThinkVPC DYTCMode 0x1
```

Maybe a firmware issue, who knows. Thanks as always

zhen-zen, 2021-06-13:
Sorry, can you try the latest master build again? Command 0x0 is just for query DYTC version, 0x2 should be get current DYTC status.

antoniomcr96, 2021-06-14:
Done. Quiet:
```
[    0.739701]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x0 result 0x40020101
[    0.741257]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x3 result 0x09110001
[    0.758818]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x2 result 0x0001f001
```

Performance:
```
[    0.766466]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x0 result 0x40020101
[    0.769241]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x3 result 0x09190001
[    0.783999]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x2 result 0x0001f001
```

zhen-zen, 2021-06-15:
It's weird. Both modes has the same standard (intelligent cooling) DYTC setting. I guess your case is the same as @1Revenger1 , but PSC mode is not available on yours.

[13 more replies]

PR #101 (2021-02-11): Pane: display threshold for all three batteries
Close #68