zhen-zen/YogaSMC #72: Kext and application for Big Sur?

htrung4, 2020-11-18:
Hello,
Thank you for the very nice kext and application. It's working perfectly with Catalina the latest, but broken on Big Sur (the icon is broken, for example when press micmute). 
[image]

zhen-zen, 2020-11-18:
You will need to move the app into `/Applications` as prompted at startup for proper resources handling.

htrung4, 2020-11-19:
I already copy it into applications. It's strange: if I copy the app first into applications and open it, then install the setting (yogasmcpane), it will not work. But if I install the setting first, then copy the app into application and open it, then it works. And, it's fixed now. Thank you for your amazing app.

closed 2020-11-19 by htrung4