zhen-zen/YogaSMC #75: [Features Request] app icon, palm detection, app icon...

Ab2774, 2020-11-21:
Hi, these are some things that I would like to see in your amazing projects.
- Add an icon for `YogaSMCNC` and `YogaSMCPane`
- Palm Detection, just like `VoodooPS2Keyboard` and `BrightnessKeys`
- Customizable Conversation Mode, in some Lenovo models, you can customize when the battery stop charging like at %70, %80, not just %60
- Disable Trackpad, in some Lenovo laptops, F6 disables the trackpad, there's a workaround which is remapping F6 to PrtSc, hope you can add to just like F4 mic mute and F7 Airplane Mode
- Key shortcut, a key shortcut to switch between FnKeys and F1-F12
- Key hold, when I hold a key like brightness keys and volume keys, they keep repeating until I stop holding it, but in YogaSMCNC, holding a key doesn't work, it fades even if I'm still holding it
- Key text, when pressing a key like F4 mic mute, it says under it "Mute" and "Unmute", can please make it like macOS with no text under it?
------
These are all the features, there are some issues and bugs to fix.
- Camera disable, F8 doesn't disable the camera, it just says "Camera"
- Airplane Mode, when pressing Airplane Mode key multiple times it still repeating even after I stop
- F14 and F15, F11 and F12 are remapped with `BrightnessKeys` as F14 and F15, however, using them with YogaSMCNC it says "Event 0x0004" every time I press them, let me know if you need any logs for that.

Thanks

zhen-zen, 2020-11-21:
> * Add an icon for `YogaSMCNC` and `YogaSMCPane`

Hmmm can you suggest some?

> * Palm Detection, just like `VoodooPS2Keyboard` and `BrightnessKeys`
> * Key hold, when I hold a key like brightness keys and volume keys, they keep repeating until I stop holding it, but in YogaSMCNC, holding a key doesn't work, it fades even if I'm still holding it

Maybe, however compared to PS2, EC keys don't have separate key_down and key_up events. And for some models, those keys are not responsive, making it useless. I will ask them for feedback though.

> * Customizable Conversation Mode, in some Lenovo models, you can customize when the battery stop charging like at %70, %80, not just %60

It's not documented in `ideapad_laptop`, so only Think variant support that.

> * Disable Trackpad, in some Lenovo laptops, F6 disables the trackpad, there's a workaround which is remapping F6 to PrtSc, hope you can add to just like F4 mic mute and F7 Airplane Mode

I don't know which workaround do you mean. However, this key usually disable the trackpad at firmware level and the driver will only send a notification.

> * Key shortcut, a key shortcut to switch between FnKeys and F1-F12

On my laptop, there's separate WMI event for fn+esc and I used that to switch Fnkeys. But that one is missing on some laptops.

> * Key text, when pressing a key like F4 mic mute, it says under it "Mute" and "Unmute", can please make it like macOS with no text under it?

Sure.

> * Camera disable, F8 doesn't disable the camera, it just says "Camera"

Currently I have no idea how to disable a USB device at runtime. Maybe some customized HID report? But still there's lack of document.

> * Airplane Mode, when pressing Airplane Mode key multiple times it still repeating even after I stop

Switch Wifi on and off will take a long time to respond, which is the case of https://github.com/zhen-zen/YogaSMC/issues/47.

> * F14 and F15, F11 and F12 are remapped with `BrightnessKeys` as F14 and F15, however, using them with YogaSMCNC it says "Event 0x0004" every time I press them, let me know if you need any logs for that.

That's associated with brightness change. I will just silence that event.

Ab2774, 2020-11-21:
Hi, thanks for responding, using "40=e037" remapping in `VoodooPS2` allows me to disable the trackpad with Fn+F6, maybe you can adapt the code from `VoodooPS2` to do it like PrtSc, the official logo for Yoga laptops is this, maybe you can use it.

[image]
Another feature I forgot to mention is Fn key recognition, macOS recognize Alt and Command but doesn't recognize the Fn key itself, it can be really useful for shortcuts to press it twice, etc.
Otherwise, your projects are pretty good.
I think [ThinkpadAssistant](https://github.com/MSzturc/ThinkpadAssistant) can help with Wi-Fi.
Cheers

zhen-zen, 2020-11-22:
> Hi, thanks for responding, using "40=e037" remapping in `VoodooPS2` allows me to disable the trackpad with Fn+F6, maybe you can adapt the code from `VoodooPS2` to do it like PrtSc, the official logo for Yoga laptops is this, maybe you can use it.
> 
I'm avoiding using official logos since they might cause issues.

> Another feature I forgot to mention is Fn key recognition, macOS recognize Alt and Command but doesn't recognize the Fn key itself, it can be really useful for shortcuts to press it twice, etc.

Fn key on OEM laptops usually don't act as a separate key. It will only modify another pressed key and sent the result as EC event or PS2 key code.

> I think [ThinkpadAssistant](https://github.com/MSzturc/ThinkpadAssistant) can help with Wi-Fi.

It should have the same behavior since the approach is similar. Let me check if https://github.com/MSzturc/ThinkpadAssistant/commit/e83ed82b6c25d2cdf5bf97fa32daa3dea957d0a3 could help.

zhen-zen, 2020-11-22:
Just tried similar approach for Wi-Fi but it's still not satisfying. Maybe just won't fix it. Do you have specific use case for that?

Ab2774, 2020-11-22:
Hi, thanks for replying, I suggest that you use a logo that presents Clamshell Mode instead of just letters, I hope you can find a solution soon because it's working pretty good but the only issue is that it keeps repeating even after I stop pressing, hope you can fix it soon.
Thanks

zhen-zen, 2020-11-22:
For prompt text, you can disable them if corresponding name in config is left empty. I just pushed that commit.

Clamshell mode is just experimental and send nothing to the app.

The airplane mode key might be an EC bug. It should only send one notification per press, no longer when you release that.

Ab2774, 2020-11-22:
> For prompt text, you can disable them if corresponding name in config is left empty. I just pushed that commit.
> 
> Clamshell mode is just experimental and send nothing to the app.
> 
> The airplane mode key issue might be an EC bug. It should only send one notification per press, no longer when you release that.

Hi, another thing I forget to mention is icons design, the icons are bold while Apple ones are less bolder, here's an example:
[image] [image]

I know these features may be hard to adapt but it would be really cool to have them.

Ab2774, 2020-11-22:
As I can see in https://github.com/zhen-zen/YogaSMC/commit/50ac2b898561a7fd9a3975974e2dc9f3c7da72c9 that you have added localization support, that would be a great opportunity for multi-languages support, I can help with that.
Thanks

Ab2774, 2020-11-22:
Hi again, I suggest you change the `FnKeys` icon from [this](https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMCNC/Resources/FunctionKey.pdf) to this one, that's what Apple officially uses in the latest M1 Macbooks.
[image]
Thanks for your patience

zhen-zen, 2020-11-22:
The globe icon is just meant for language switching and Fn does not have an icon. You can still customize the menubar icon in prefpane.

Current icons are based on SF Symbols. Feel free to open a PR to if you have better resources since I'm not much a designer.

Ab2774, 2020-11-22:
Hi, I understand, I'm not a designer too, maybe you can ask @antoniomcr96 since he designed this one https://github.com/zhen-zen/YogaSMC/pull/74#issuecomment-731572634.
BTW, I build it from the latest commit and https://github.com/zhen-zen/YogaSMC/commit/015a31b4f59d3d5770bf119ffb1503e50384291c doesn't seem to work, is there's something I'm missing?
Thanks

zhen-zen, 2020-11-22:
Just simplified empty name support. You can edit the preference (~/Library/Preferences/org.zhen.YogaSMC.plist) and replace corresponding names to empty string. Names in current preset is still non-empty and it's unlikely to update them for current users. I will have some kind of poll in Gitter and decide whether to leave them empty in preset for the next release.

Ab2774, 2020-11-25:
> Just simplified empty name support. After quitting the app, you can edit the preference (~/Library/Preferences/org.zhen.YogaSMC.plist) and replace corresponding names to empty string. Names in current preset is still non-empty and it's unlikely to update them for current users. I will have some kind of poll in Gitter and decide whether to leave them empty in preset for the next release.

Hi, if you're planning on removing the text in the next release, please adjust the icons like Airplane Mode and mic mute in the middle since there's no indicator or anything else under them.
Thanks

zhen-zen, 2020-11-25:
The OSD functionality borrows the system one. It's not possible to modify the layout.

Ab2774, 2020-11-27:
Hi, thanks for adopting `timestamp` functionality into `YogaSMC`, hope the key hold and Airplane Mode issues are fixed soon too.
Thanks for the hard work!

zhen-zen, 2020-11-27:
As previously mentioned, EC will only send one notification per press instead of separate `KEY_DOWN` & `KEY_UP` for standard PS2 keys. So that's not going to be implemented.

And for Airplane Mode, ThinkAssistant's approach doesn't work well. I'm not sure if that works for them.

zhen-zen, 2020-12-08:
Seems like all the entries in the list have been answered. Did I miss something?

Ab2774, 2020-12-08:
Hi @zhen-zen, no you haven't, your projects are much better now with these features, but I think it may get way cooler if you implement those features, btw, can you release v1.3.1 since a lot of things has changed?
Thanks

closed 2020-12-08 by Ab2774

zhen-zen, 2020-12-08:
I will publish next release after receiving responses from other issues. The logo and other resources are still open for PR. Feel free to join the Gitter chat.

zhen-zen, 2020-12-10:
#86