zhen-zen/YogaSMC #82: Mute button unresponsive on Thinkpad E480

notiflux, 2020-11-28:
On my laptop, the mute button does not work natively. I can remap it through DSDT to send a PS/2 scan code to toggle mute, but that still doesn't make the mute LED work. It seems for this laptop you actually need to make use of the MMTS method to automate the LED / check for the button event in the first place. DSDT is attached, let me know if you need any other files too.
[DSDT.dsl.zip](https://github.com/zhen-zen/YogaSMC/files/5611132/DSDT.dsl.zip)

zhen-zen, 2020-11-29:
With debug version of the driver, are there anything in the log when pressing that button? Also, does the button work in Linux?

notiflux, 2020-12-01:
[ysmc.dmesg.log](https://github.com/zhen-zen/YogaSMC/files/5621640/ysmc.dmesg.log)
nothing in the log file as far as I can tell. Nothing in Console either and `sudo log stream --predicate 'senderImagePath contains "YogaSMC"'` doesn't show anything either. The button does work in linux

zhen-zen, 2020-12-01:
Only `dmesg` can dump the log in Big Sur.

It sounds like your mute status is `TP_EC_MUTE_BTN_TOGGLE` at startup. Can you try `ioio -s ThinkVPC Mute 2`?

notiflux, 2020-12-02:
that command yields `YSMC - Info: ThinkVPC::HKEY setMutestatus toggle failed` in the log

zhen-zen, 2020-12-03:
Can you add `Linux` entry along with Windows ones in your `XOSI` hotpatch?

notiflux, 2020-12-04:
that actually worked! Mute button works now, HKEY Mute is set to 2. MuteLED automation still does not work though

zhen-zen, 2020-12-04:
I suppose the LED will just follow the key press. Is it turned off with and without the automation during sleep?

notiflux, 2020-12-05:
the LED does not follow the key press. when I put it to sleep, I was able to enable the mute LED by pressing the button once, but I wasn't able to turn it back off after putting it to sleep another time (pressing the key woke the machine up)

zhen-zen, 2020-12-08:
It sounds like a firmware issue. If the issue also presents in Linux or in macOS without this driver, I suppose it could not be fixed by driver.

notiflux, 2020-12-08:
the issue is not present in linux, everything works fine there. it seems the only thing left to do is to manually toggle the LED from within the driver based on the current mute status of the system

zhen-zen, 2020-12-08:
The driver itself could not access to volume status in userspace. Maybe the NC app could do something and I will continue the topic in #51 .

closed 2020-12-08 by zhen-zen