zhen-zen/YogaSMC #83: Thinkpad T450s - Fan speed control isn't working

i3p9, 2020-12-07:
Currently running `REL-130-2020-11-11` version of YogaSMC, on Catalina 10.15.7. Opencore 0.6.4, VirtualSMC 1.1.9, Lilu 1.5.0

Issue: Fan speed control is not working.

After installing the Kext, I can see the fan speed properly in third party apps like iStat Menu/MacFanControl but there's no way to control them. Also strangely, YogaSMCNC panel can't read fan speed. I think maybe because it says DYTCFunction: unsupported.

Screenshots of YogaSMCNC:
[image]

[image]

And another thing is when opening YogaSMCNC, there's this message about EC not loading: 
[image]

For now, all I want is the ability to control Fan Speed. Please let me know if there's any information I can provide, I've switched to the debug kext to provide any info if necessary.

zhen-zen, 2020-12-08:
Please merge `SSDT-RCSM.dsl` in `SSDTSample` into your battery hotpatch or as a standalone one after checking for conflict.

i3p9, 2020-12-08:
I just tried loading `SSDT-RCSM` patch but it didn't work unfortunately. My battery patches are here: https://github.com/i3p9/Hackintosh-BigSur-Catalina-OpenCore-Lenovo-T450s-efi/tree/master/EFI/OC/ACPI

Can you let me know how to check for conflicts? I'm very new to patching and willing to learn. Do I check for conflicting definitions in the dsl files?

zhen-zen, 2020-12-08:
Add `WE1B` and `WECB` methods to your `SSDT-BAT` at the same scope of `RE1B` and `RECB`.

zhen-zen, 2020-12-08:
I'm closing this issue since it's not related to the driver. If the it still won't work, please continue the discussion in https://github.com/zhen-zen/YogaSMC/discussions?discussions_q=category%3A%22ACPI+Patching%22

closed 2020-12-08 by zhen-zen

VinylNerd, 2021-03-16:
> Add `WE1B` and `WECB` methods to your `SSDT-BAT` at the same scope of `RE1B` and `RECB`.

im confused how to do this, could you please link to somewhere which explains as id love to get this working and dont have ec is RO because havnt done this.

zhen-zen, 2021-03-19:
Please see https://github.com/zhen-zen/YogaSMC#ec-reading and the sample SSDT hotpatch at https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMC/SSDTSample/SSDT-ECRW.dsl.

VinylNerd, 2021-03-19:
> Please see https://github.com/zhen-zen/YogaSMC#ec-reading and the sample SSDT hotpatch at https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMC/SSDTSample/SSDT-ECRW.dsl.

sorry, this doesnt make any sense to me, if you could explain in a more simple way that would help alot, thanks!

zhen-zen, 2021-03-19:
Unfortunately there's no easy way to do that. Most people have to read Rehabman's manual in the readme. Current hotpatch approach is already simplified a lot. Dortania's ACPI guide is also a good staring point: https://dortania.github.io/Getting-Started-With-ACPI/.

Zkingtut, 2022-08-28:
Hello @zhen-zen 
Please I will be so grateful if you explain me why is that ACPI Error , Everything is working fine but I've to investigate why Windows restart sometimes BSOD ( blue screen with ACPI Bios error ) .
if I delete that RE1B - there is no fans and I got Yoga error .. something like ECRW.dsl 
really I don't know and need your advice . Laptop Lenovo Thinkpad P50
thank you
[image]

zhen-zen, 2022-08-29:
> Hello @zhen-zen
>
> Please I will be so grateful if you explain me why is that ACPI Error , Everything is working fine but I've to investigate why Windows restart sometimes BSOD ( blue screen with [...]

Sorry I clicked edit instead of quote reply earlier.

Usually it means you already have RE1B in another SSDT, such as the battery hotpatch one. Just removed the duplicated method in the sample SSDT. Also I don't think it's related to BSOD since this method is only used for fan control.