zhen-zen/YogaSMC #94: Bluetooth Enable/Disable Hotkey

xuandung38, 2020-12-11:
Can i have bluetooth Enable/Disable Hotkey like thinkpad assistian ?

zhen-zen, 2020-12-12:
That function is already assigned to key `TP_HKEY_EV_BLUETOOTH`: `0x1314` for action `.bluetooth` https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMCNC/Configuration.swift#L100

It should work ootb if yours have that key. And you can also customize it in the configuration.

xuandung38, 2020-12-12:
How to get keycode combo ? Like FN + Shift + f8

zhen-zen, 2020-12-12:
Fn keys are different from PS2 ones and there are no modifiers like `Shift`.

zhen-zen, 2020-12-12:
However, it should be possible to listen to modifiers and add extra states to a key. Since EC key press is a one time event without separate up/down, the modifier should be held during that.

PR #96 (2020-12-13): NC: support modifier
Close #94
closed 2020-12-13 by zhen-zen