zhen-zen/YogaSMC #100: VoodooPS2Controller causes KP (Thinkpad X250)

r33int, 2020-12-22:
Hello,

I am facing an issue with this kext. When VoodooPS2Controller and YogaSMC are present, the system would not boot and kernel panic. When I remove VoodooPS2Controller, the system boots as intended.

I am using RehabMan's VoodooPS2 2018-1008. acidanthera's version 2.1.9 works though. However, this version contains magic trackpad emulation, which does not work properly on my machine, so I want to stick to a version with only basic gestures.

I wanted to know if it was possible to get YogaSMC to work while using RehabMan's VoodooPS2 driver.
Any help is appreciated, thanks in advance!

[image]

zhen-zen, 2020-12-22:
Maybe do a `git bisect` in https://github.com/acidanthera/VoodooPS2/ to locate which commit fix that? And have you tried https://github.com/VoodooSMBus/VoodooRMI over SMBus?

r33int, 2020-12-22:
I just tried with VoodooRMI, and now my trackpad works better than ever, and YogaSMC works with it. Thanks a lot!

closed 2020-12-22 by r33int