zhen-zen/YogaSMC #102: New Audio Helper crash

buyddy, 2020-12-23:
Application Specific Information:
Fatal error: 'try!' expression unexpectedly raised an error: YogaSMCNC.AudioPropertyError.emptyResult: file YogaSMCNC/AudioHelper.swift, line 112
 

I think the most recent change in audio helper fails to initiate on these 

let inputDevice = try! AudioProperty<AudioDeviceID>(device: nil, emptyValue: nil, prop: defaultInputDeviceProp).get().get()
let outputDevice = try! AudioProperty<AudioDeviceID>(device: nil, emptyValue: nil, prop: defaultOutputDeviceProp).get().get()

zhen-zen, 2020-12-24:
Actually that's why I put them in a separate class before. This part should be model independent but I'm not sure what's going wrong.

zhen-zen, 2020-12-25:
Will the product from last commit in #88 work?

buyddy, 2020-12-26:
Looks like the most recent one fix this issue!

closed 2020-12-26 by buyddy