zhen-zen/YogaSMC #106: conservation mode doesn´t seem to work

kasti0, 2020-12-29:
in conservation mode the laptop should usually charge only up to 60% to preserve battery health.
when I enable it it still charges all the way up to 100%.

I´m using YogaSMC 1.4.0
my Laptop: Ideapad 330s 15ikb (i5-8250u, UHD 620)

zhen-zen, 2021-01-07:
Conservation mode toggle may not notify the battery device. As a result, SMCBatteryManager won't poll and update battery status at a steady state until you reconnect the AC adapter. Also, please try to set it in Linux as current ACPI call replicate the one in `ideated_laptop`.

kasti0, 2021-01-08:
Actually I'm not sure what you're trying to tell me with the last line about Linux (I don't use it at all).

But it seems like it's working now, unfortunately I can't tell you what I changed. I just recently updated all other kext that usually get released at the beginning of the month by acidanthera.
It stops charging at 62% when in conservation mode. Not exactly 60%, may this bis an issue with my battery patching or the kext?

zhen-zen, 2021-01-09:
> Actually I'm not sure what you're trying to tell me with the last line about Linux (I don't use it at all).

See https://wiki.archlinux.org/index.php/Laptop/Lenovo#Battery_Conservation_Mode_on_IdeaPad_laptops

The conservation mode switch is pure ACPI implementation and won't be affected by other drivers. There could be some delay from firmware but that's out of hand.

The percentage might stays in a range around 55%-60% and its calculation is different when wear level is involved. Please reboot to Windows and check the percentage there.

zhen-zen, 2021-02-09:
Closed for lack of information. Feel free to reopen if you have any updates.

closed 2021-02-09 by zhen-zen