zhen-zen/YogaSMC #109: Setting hotkey to open System Preferences

simprecicchiani, 2021-01-08:
Hi, thanks for the amazing package you developed.
Would it be possible to map the setting hotkey to open system preferences (and close it if already open) instead of YogaSMC pref panel?
I don't know if others share my opinion but, coming from ThinkpadAssistant, I really miss this feature.
Thanks again for the effort.

zhen-zen, 2021-01-08:
You can customize application to open with `launchapp` action:
https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMCUtils/Configuration.swift#L96
Personally I think it's anti-instinctive to close an application on second key press.

zhen-zen, 2021-02-10:
Please check option + Star combo with app from #117.

closed 2021-02-11 by zhen-zen

soeryobadja, 2021-03-11:
Hello.

My understanding is that the `launchbundle` in version 1.4.2 is to enable the F9 key to open SystemPreferences instead of opening SMCPane like the previous version. 
However, I tested version 1.4.2 but seems the F9 is still opening the SMCPane and not the SystemPreferences. Do I make the correct understanding? 
Also, is the installation of SMCPane compulsory? Meaning, if I dont install SMCPane, can I still use YogaSMC without any warning to install the SMCPane? 

Thanks.

zhen-zen, 2021-03-11:
`launchbundle` is not the default action. You can press option + Star to try it out and swap it on later. See https://github.com/zhen-zen/YogaSMC/blob/2ed60c00e9807902b0d3396ba71ba51a10aa46e0/YogaSMCUtils/Configuration.swift#L137 for current preset.

The prefpane is not mandatory. It's just an ugly GUI to access and modify driver settings through IORegistry. [ioio](https://github.com/RehabMan/OS-X-ioio) should be able to do the same.

soeryobadja, 2021-03-13:
Hello. 
Thanks for your guide. 

> You can press option + Star to try it out and swap it on later.
I tried the combination, but does not change anything. It still ask for YogaSMCPane when I press the F9. 

> The prefpane is not mandatory. It's just an ugly GUI to access and modify driver settings through IORegistry. ioio should be able to do the same.
Wondering why the F9 function button is linked to the Prefpane (YogaSMCPane) if it is not mandatory? I prefer to have F9 open the System Preferences though. 

I would love to edit the apps to achieve this function, unfortunately I have limited knowledge in this. It will be very much appreciated if in the next release, such the link to F9 is back to SystemPreferences, or at least have some feature to change the button assignment from the current to opening YogaSMCPane to opening the SystemPreferences. 

Thanks.

zhen-zen, 2021-03-14:
> I tried the combination, but does not change anything. It still ask for YogaSMCPane when I press the F9.

The key combo relies on separate key press report via PS2(normal keys including modifier) and ACPI (hotkey) and typically they should report simultaneously. You can check it with Karabiner-EventViewer bundled in karabiner-elements. Some people might also have swapped option / command keys.

> Wondering why the F9 function button is linked to the Prefpane (YogaSMCPane) if it is not mandatory? I prefer to have F9 open the System Preferences though.

That action is just a showcase of AppleScript usage. You can edit the configuration plist per https://github.com/zhen-zen/YogaSMC#yogasmcnc .

> I would love to edit the apps to achieve this function, unfortunately I have limited knowledge in this. It will be very much appreciated if in the next release, such the link to F9 is back to [...]

Actually I'm tired of building a GUI app and digging further with few examples. As I mentioned before, the prefpane is not satisfying and I haven't got some ideas on refactoring. Same for assigning function to a key, I want to build something like `Notification` pane, but don't know how to build that dynamically. So if someone found such open-source apps with similar function, please notify me. I opened a post at https://github.com/zhen-zen/YogaSMC/discussions/123.

PR #117 (2021-02-11): NC: support launchbundle
Close #109