zhen-zen/YogaSMC #110: Lenovo Thinkpad E590 DYTC Issues

Someone52, 2021-01-18:
Lenovo Thinkpad E590
When trying to set the modes, I get the following:
 
`YSMC - Info: ThinkVPC::KHEY DYTCMode command 0x12b001 result: 0x0000000a (invalid argument)
YSMC - Info: ThinkVPC::HKEY DYTCMode toggle failed`

After a bit of investigation, i've decompiled the dytc method on my laptop, which is as follows:
`          Local0 = Arg0
            Local1 = Zero
            ADBG (Concatenate ("DYTC STT=", ToHexString (Local0)))
            If ((WNTF && TATC))
            {
                Switch (ToInteger ((Local0 & 0x01FF)))
                {
                    Case (Zero)
                    {
                        Local1 = 0x0100
                        Local1 |= 0x50000000
                        Local1 |= Zero
                        Local1 |= One
                    }
                    Case (One)
                    {
                        Local2 = ((Local0 >> 0x0C) & 0x0F)
                        Local3 = ((Local0 >> 0x10) & 0x0F)
                        Local4 = ((Local0 >> 0x14) & One)
                        ADBG ("DYTC_CMD_SET")
                        ADBG (Concatenate ("ICFunc=", ToHexString (Local2)))
                        ADBG (Concatenate ("ICMode=", ToHexString (Local3)))
                        ADBG (Concatenate ("ValidF=", ToHexString (Local4)))
                        Local5 = (One << Local2)
                        If ((FCAP & Local5)){}
                        Else
                        {
                            Local1 = 0x06
                            ADBG (Concatenate ("DYTC END=", ToHexString (Local1)))
                            Return (Local1)
                        }

                        Switch (Local2)
                        {
                            Case (One)
                            {
                                If ((Local3 != 0x0F))
                                {
                                    Local1 = 0x0A
                                    ADBG (Concatenate ("DYTC END=", ToHexString (Local1)))
                                    Return (Local1)
                                }

                                If ((Local4 == Zero))
                                {
                                    VCQL = Zero
                                }
                                Else
                                {
                                    VCQL = One
                                }
                            }
                            Case (0x04)
                            {
                                If ((Local3 != 0x0F))
                                {
                                    Local1 = 0x0A
                                    ADBG (Concatenate ("DYTC END=", ToHexString (Local1)))
                                    Return (Local1)
                                }

                                If ((Local4 == Zero))
                                {
[... 533 lines cut ...]
                                    If ((Local4 != Zero))
                                    {
                                        Local1 = 0x0A
                                        ADBG (Concatenate ("DYTC END=", ToHexString (Local1)))
                                        Return (Local1)
                                    }
                                }
                                Else
                                {
                                    Local1 = 0x0A
                                    ADBG (Concatenate ("DYTC END=", ToHexString (Local1)))
                                    Return (Local1)
                                }

                                If ((Local4 == Zero))
                                {
                                    VPSC = Zero
                                    SPSC = Zero
                                }
                                Else
                                {
                                    VPSC = One
                                    SPSC = Local3
                                }
                            }`

zhen-zen, 2021-01-19:
Available DYTC modes may vary in different models. However, only documented modes in https://patchwork.kernel.org/project/linux-acpi/patch/20201211020630.305905-3-markpearson@lenovo.com/ is added. Other modes in DSDT are just made into unused macro.

If you are interested in further investigation, you can try following steps and maybe can find something useful.

1. Reboot into windows / BIOS and change performance modes.
2. Back into macOS and check raw DYTC status in the log (updated once during boot).
3. Document the specific mode / function combinations. (Mine are just the same as the ones in the patch above)
4. Try sending raw DYTC command, i.e. `ioio -s ThinkVPC DYTCMode 0x12b001`

Someone52, 2021-01-19:
After some more investigation, there seem to be 8 performance modes for the 0xD (13) command.

ioio -s ThinkVPC DYTCMode 0x11D001
ioio -s ThinkVPC DYTCMode 0x11D001
ioio -s ThinkVPC DYTCMode 0x11D001

Someone52, 2021-01-19:
I have made a pull request containing the implementation of my findings.
https://github.com/zhen-zen/YogaSMC/pull/111

zhen-zen, 2021-01-20:
Thank you for implementing support for this function. The PR looks good and maybe we can carry out preciser control based on DYTC version.

PSC function is also spotted on [X1 Extreme](https://github.com/Errrneist/Hackintosh-Thinkpad-X1-Extreme/blob/master/10.15/10.15.2/3.7/CLOVER/ACPI/DUMP/DSDT.aml) and [X1C6](https://github.com/tylernguyen/x1c6-hackintosh/blob/master/ACPI/Stock%20ACPI/BIOS-v1.45/DSDT.aml). Both machine has DYTC version 4.2, but the latter one has the following restriction:

```
If (((Local3 != 0x02) && ((Local3 != 0x07) && (Local3 != 0x08))))
{
    ADBG ("PSC InValid Mode, Clear the PSC State.")
    Local4 = 0x00
}
```

So I'm not sure if this function is ready on 4.2 version.

@1Revenger1 @benbender Does current mode control work on your laptop? And how about the linked PR by @Someone52 ?

And prior to version 4.2, it seems that PSC mode is not supported. The 4.1 one from [E580](https://github.com/dandepeched/Hackintosh-ThinkPad-E580/blob/master/EFI-Clover/CLOVER/ACPI/origin/DSDT.aml) even has few modes than mine (4.0). Maybe we can judge the availability of PSC mode by DYTC version?

The most complete DYTC method I have seen is obtained from [Yoga C940 (15) ](https://www.tonymacx86.com/threads/yoga-c940-installation-help-request.286947/)

```
STD: 0x0, 0xf, - // standard
CQL: 0x1, 0xf, 0/1 // lap
TIO: 0x2, 0xf, 0/1
MYH: 0x3,
	TBL: 0x0, 1
	TNT: 0x1, 1
	LFT: 0x2, 1
	MASK: 0xf, 0
STP: 0x4, 0xf, 0/1
CQH: 0x5, 0xf, 0/1
DCC: 0x6, 0xf, 0/1
SFN: 0x7, 0xf, 0/1
DMC: 0x8, 0xf, 0/1
FHP: 0x9, 0xf, 0/1
IFC: 0xa, 0xf, 0/1
MMC: 0xb, // desk
	PFM: 0x2, 1 // performance
	QUT: 0x3, 1 // quiet
	MASK: 0xf, 1 // balance
MSC: 0xc, 0xf, 0/1
PSC: 0xd, 
	0x1-0x8, 1
	0xf, 0
CSC: 0xe, 0xf, 0/1
```

Currently I don't have many samples on DYTC 5.0 and newer. But at least yours is covered by the list above.

1Revenger1, 2021-01-20:
Just tested on my device and it appears to function correctly. The changes are reflected in IORegistryExplorer and no errors are created when looking through `log show --last boot | grep -i ysmc`. There doesn't seem to really be any change on how my device uses the fans, though that was the case with the old implementation of DYTC as well. Additionally, I am able to select Quiet values (before if I set quiet, it would revert back to Normal). Here is a picture from IOReg:
[image]

zhen-zen, 2021-01-20:
The normal mode is called MMC AUTO mode in Yoga C940 (15) example above. It seems that on X1E and X1C6, supported mode is 1/2/f instead of 3/2/f for quiet/performance/balance (aka AUTO). I don't know how to distinguish them, maybe some different command below may help that.

For version 5.0, only normal mode is supported for E590 and [X1C7](https://github.com/acoutts/x1c7-hackintosh/blob/master/ACPI/Stock%20ACPI/BIOS-v1.12/DSDT.aml) don't support MMC at all.

And it's quiet confusing that they still stick to the MMC function in https://patchwork.kernel.org/project/platform-driver-x86/patch/20210111162237.3469-1-markpearson@lenovo.com/ but require DYTC version 5 and newer.

tylernguyen, 2021-01-24:
@zhen-zen 

The current mode control doesn't work on my x1c6. It's only limited to Balance or Performance. The same thing goes for the new PR.

zhen-zen, 2021-01-27:
> @zhen-zen
> 
> The current mode control doesn't work on my x1c6. It's limited to only Balance or Performance. The same thing goes for the new PR.

I suppose at least 2/7/8 should work for your model. So it's a bit strange here. When you are available, can you check the initial DYTC result in debug log after switching to different thermal mode in Vantage / BIOS?

zhen-zen, 2021-02-07:
Finally got some time for #116 . Please try that and I will merge both PR.