zhen-zen/YogaSMC #114: Help with implementing EC reading

ldan93, 2021-02-01:
Hello,
This issue is not directly related to YogaSMC. I'm kindly asking for some advice, your help would be very appreciated. 

With @nekr0z and @profzei, we are trying to implement a battery conservation feature (charging thresholds) on the Huawei Matebook X Pro :  nekr0z/matebook-applet#26 . @nekr0z wrote an app that works on Catalina. It relies on `ioio` and `ACPI_Debug.kext` to call ACPI methods to set the thresholds, and to read the related EC fields.

Reading the thresholds is currently done by outputting EC values to system logs, and then we get them by parsing the result of the `log stream --predicate 'senderImagePath contains "ACPIDebug"'` command.
The problems are : 
- Getting values this way doesn't sound very clean
- `ACPI_Debug.kext` seems to be partially broken on Big Sur : ACPI methods can still be called, but [outputting text to system logs doesn't work anymore](https://github.com/nekr0z/matebook-applet/issues/26#issuecomment-770353763). So the app is broken on Big Sur since we can't read the current thresholds...

I see that YogaSMC implement EC field reading. We would like to have your insight on our issue : is there a simple and cleaner way than parsing the system logs to get the EC values ? What would you advise us to do ?

zhen-zen, 2021-02-04:
I also used to dig variables using ACPIDebug and customized ACPI methods before I come up with this kext.

Try if #115 will attach to HWMI interface. (Use the Alter one.)

ldan93, 2021-02-08:
Thank you !
I've successfully built the `mate-attach` branch and I've loaded the Alter kext with `kextload`. But then, I'm not sure what I should do / be looking for about the `HWMI` interface you mentioned...

zhen-zen, 2021-02-09:
There's already prebuilt one at https://github.com/zhen-zen/YogaSMC/actions/runs/536369821 . And please check your ioreg if `YogaVPC` is attached. For usage, see https://github.com/zhen-zen/YogaSMC#ec-reading

ldan93, 2021-02-09:
YogaVPC is not attached in [my ioreg](https://github.com/zhen-zen/YogaSMC/files/5949970/ioreg.txt.zip). But there are 2 mentions of `YogaWMI`.
Trying to read EC fields with `ioio` returns `No service matching "YogaVPC"`

zhen-zen, 2021-02-10:
It seems that the driver couldn't match the interface by `_UID`. So I will just use `AMW0` for now. The same name is also used by WMI reference code so it won't be merged into master in this way.

The driver also supports EC RW functionality through `IOUserClient`, which can be accessed directly from user-space and won't mess up with log format. Here is an example: https://github.com/zhen-zen/YogaSMC/commit/87074b45d964ba4db82c1feee72703677ecb8c9a

zhen-zen, 2021-02-10:
You can also inject a fake device for YogaVPC to attach. Check the latest commit of that PR.

ldan93, 2021-02-14:
I've added `SSDT-YVPC.dsl` to my OpenCore. I don't understand what I'm supposed to do then to make the driver work on the matebook...

zhen-zen, 2021-02-14:
Compile it first and add it to `config.plist`.

zhen-zen, 2021-05-10:
Close it for now. I will continue working on #135 after receiving further response.

closed 2021-05-10 by zhen-zen