zhen-zen/YogaSMC #119: Lenovo L340 Gaming not full support ?

HandaTakuya, 2021-02-14:
[image]
[image]

I use only YogaSMC 
I can unlock option of keyboard Backlight, DYTC, Function Key and Fan control ?

zhen-zen, 2021-02-14:
Available functions depends on specific models. Please attach ACPI and ioreg dump.

HandaTakuya, 2021-02-14:
[Jakkapan’s MacBook Pro.zip](https://github.com/zhen-zen/YogaSMC/files/5978450/Jakkapan.s.MacBook.Pro.zip)

[ACPI.zip](https://github.com/zhen-zen/YogaSMC/files/5978454/ACPI.zip)

I'm not sure to export ACPI.

zhen-zen, 2021-02-15:
According to your DSDT, always on USB and conventional keyboard backlight control are not supported.
```
Method (HALS, 0, NotSerialized)
{
    Local0 = Zero
    If (ECAV)
    {
        If ((Acquire (LFCM, 0xA000) == Zero))
        {
            Local0 |= 0x08
            Local0 |= 0x0200
            If (HKDB)
            {
                Local0 |= 0x0400
            }

            Local0 |= 0x0800
            If (((SPMO == One) && (SPMT == Zero)))
            {
                Local0 |= 0x2000
            }

            Release (LFCM)
        }
    }

    Return (Local0)
}
```
https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMC/YogaVPC/IdeaVPC.hpp#L47-L53

And DYTC method is not available on your laptop.

For function key support, you will need the NC app in the dmg in user-space as well.

Fan control on idea variant is not implemented due to lack of document.

HandaTakuya, 2021-02-15:
> According to your DSDT, always on USB and conventional keyboard backlight control are not supported.
> 
> ```
> Method (HALS, 0, NotSerialized)
> {
>     Local0 = Zero
>     If (ECAV)
>     {
>         If ((Acquire (LFCM, 0xA000) == Zero))
>         {
>             Local0 |= 0x08
>             Local0 |= 0x0200
>             If (HKDB)
>             {
>                 Local0 |= 0x0400
>             }
> 
>             Local0 |= 0x0800
>             If (((SPMO == One) && (SPMT == Zero)))
>             {
>                 Local0 |= 0x2000
>             }
> 
>             Release (LFCM)
>         }
>     }
> 
>     Return (Local0)
> }
> ```
> 
> https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMC/YogaVPC/IdeaVPC.hpp#L47-L53
> 
> And DYTC method is not available on your laptop.
> 
> For function key support, you will need the NC app in the dmg in user-space as well.
> 
> Fan control on idea variant is not implemented due to lack of document.
> 
> To read battery information, you will need to apply the same battery patch to `GBID` method.

Ohh! thank you, and my hackintosh can enable EC Access basic to RW ?

zhen-zen, 2021-02-16:
Just check SSDTSample/SSDT-ECRW.dsl in the release.

closed 2021-02-16 by zhen-zen