zhen-zen/YogaSMC #129: Conservation mode makes my Miix 520 shutdowns randomly

bangbaew, 2021-04-30:
I wonder if anyone encounters the same problem as me:
Normally my Miix 520 battery would go from 100% and shutdown when less than 10%.
But after turning on conservation mode in macOS (in Windows it works fine) plugging out my adapter and using it on battery for a while, my Miix 520 randomly shutdowns, sometimes at 40% / 60% / 80% (even after fully charged), my solution is to drain battery to 0% and recharge to 100%, don't turn on conservation and it will work fine as normal, but if I turn on conservation again, the problem persists.

zhen-zen, 2021-05-01:
Actually suffered from similar symptom when I have Miix 720 before. I suppose it's a firmware issue / current overload and it also got sudden shutdown in Windows.

The conservation mode implementation follows Linux one. Did you get the same issue if enable it in Linux? https://wiki.archlinux.org/index.php/Laptop/Lenovo#Battery_Conservation_Mode_on_IdeaPad_laptops

Or how about enabling it in Windows and reboot to macOS? If this way works, can you show the kernel log when boot up with and without conservation mode set?

zhen-zen, 2022-08-12:
Closed due to inactivity.

closed 2022-08-12 by zhen-zen