zhen-zen/YogaSMC #136: Flex 15IML won't boot after 1.5.0

Chris2fourlaw, 2021-05-19:
After updating to 1.5.0, I'm unable to boot without disabling the kext in the config.plist

Looks ACPI related

[image]

[EFI.zip](https://github.com/zhen-zen/YogaSMC/files/6510707/EFI.zip)

zhen-zen, 2021-05-20:
Hello, EFI file is not helpful since it's only a set of configuration and contains few model-specific information. Can you attach ACPI dump and a copy of ioreg with previous version?

Chris2fourlaw, 2021-05-20:
Attached
[IoregACPI.zip](https://github.com/zhen-zen/YogaSMC/files/6512470/IoregACPI.zip)

zhen-zen, 2021-05-20:
I guess something went wrong when populating WMI devices. Can you try the debug build here: https://github.com/zhen-zen/YogaSMC/actions/runs/844295415 ?

Ab2774, 2021-05-20:
Hi @zhen-zen, I did test the debug build and it works perfectly.
Thanks

Chris2fourlaw, 2021-05-20:
With the debug version 

[image]

zhen-zen, 2021-05-21:
> Hi @zhen-zen, I did test the debug build and it works perfectly.
> Thanks

Did you also get issues with latest release?

> With the debug version
> 
> [image]

This seem to be an issue related to WMI handling but I thought I have fixed that one. Can you try the latest debug version? I should have asked you to try that first.

Chris2fourlaw, 2021-05-21:
Testing more, it's pretty intermittent when it happens. Around half the time it boots normally with the debug version, but even on those I see some info errors for "Failed to evaluate BMF data". 

Here is a screenshot of when I can't get it to boot:
[image]

Chris2fourlaw, 2021-05-21:
The above is on the latest debug

zhen-zen, 2021-05-21:
I'm sorry for the inconvenience. It's a bit hard to find out where's going wrong.

So what I get is:

1. Boot without issue with 1.4.3
2. Since the CI build above, including the release one, there's possibility that kext will hang.

Please correct me if something's wrong.

Also, could you add https://github.com/acidanthera/DebugEnhancer and dump the log with `sudo dmesg | grep YSMC` after a successful boot with latest debug kext?

Chris2fourlaw, 2021-05-21:
Correct. dmesg output here: https://pastebin.com/imwWY2VU

Time for me to sleep, got a calculus midterm in the morning. I will try your next steps after that

Chris2fourlaw, 2021-05-21:
My thoughts:

Matching the following with my provided screenshots
```
[    2.724995]: YSMC - Debug: WMI2::YogaWMI Starting
[    2.724997]: YSMC - Info: WMI2::WMI2 Failed to evaluate BMF data
[    2.725279]: YSMC - Debug: EC0::IdeaSMC Starting
```

The last line seems to fail to print midline. My laptop is an IdeaPad Flex 15IML. Maybe the issue is with IdeaSMC?

zhen-zen, 2021-05-22:
If there's something wrong with YogaSMC, you can try YogaSMCAlter.kext . This version doesn't have SMC functionality. Not sure why it will cause intermittent kp.

zhen-zen, 2021-05-22:
Trying to refactor some SMC related code in #138 . I thought I already made that change but unable to find them.

Chris2fourlaw, 2021-05-22:
I'll test the latest built a bit and see if it fixes it

zhen-zen, 2021-05-22:
Can you try the latest release build in master branch? It should be available in https://github.com/zhen-zen/YogaSMC/actions/runs/867253917

Chris2fourlaw, 2021-05-22:
Looks like it's booting, I've attached the debug output again. The preference pane seems to think it's a ThinkPad maybe...
https://pastebin.com/FpRMQmjz

[image]

Chris2fourlaw, 2021-05-22:
Just rebooted and it failed again
[image]

zhen-zen, 2021-05-22:
> Looks like it's booting, I've attached the debug output again. The preference pane seems to think it's a ThinkPad maybe...

You need to switch to General or Idea tab. The debug prefpane has sub-panel for all variants enabled.

> Just rebooted and it failed again
> [image]

Will continue looking into this.

Chris2fourlaw, 2021-05-22:
> You need to switch to General or Idea tab. The debug prefpane has sub-panel for all variants enabled.

Ah makes sense. Let me know if you need anything else from my computer

zhen-zen, 2021-05-22:
> > You need to switch to General or Idea tab. The debug prefpane has sub-panel for all variants enabled.
> 
> Ah makes sense. Let me know if you need anything else from my computer

Sorry, issues mixed up and I forgot to mention the Alter version. Can you try `YogaSMCAlter` instead to help isolate the issue? Thanks!

Chris2fourlaw, 2021-05-22:
Alter seems to work fine
https://pastebin.com/chxcvbbP

zhen-zen, 2021-05-22:
It's weird. So IdeaSMC is about to start at that point, the init is also good.... Can you try this one?

[YogaSMC_log.kext.zip](https://github.com/zhen-zen/YogaSMC/files/6527324/YogaSMC_log.kext.zip)

Chris2fourlaw, 2021-05-23:
https://pastebin.com/sNPpLZB5
Here is a successful boot

Chris2fourlaw, 2021-05-23:
Unsuccessful:
[image]

Also there's this EC timeout thing that might be relevant
[image]

zhen-zen, 2021-05-23:
Thanks! I guess we are close to the point. Tried to make some changes, not sure if it could help that.
[YogaSMC_poller.kext.zip](https://github.com/zhen-zen/YogaSMC/files/6527855/YogaSMC_poller.kext.zip)

Chris2fourlaw, 2021-05-23:
Failed:
[image]

Successful:
https://pastebin.com/wu2qFLBX

zhen-zen, 2021-05-23:
Found a possible culprit that setPowerState could be executed before the poller is initialized. Can you try this one?

[YogaSMC_ready.kext.zip](https://github.com/zhen-zen/YogaSMC/files/6528671/YogaSMC_ready.kext.zip)

Chris2fourlaw, 2021-05-23:
Successful:
https://pastebin.com/kYnjts5C

I restarted like 6 times and can't get it to crash. I'll let you know if it starts doing it again

zhen-zen, 2021-05-23:
Cool, we got it! Thank you for being so patient!

Chris2fourlaw, 2021-05-23:
No problem :)

zhen-zen, 2021-05-23:
By the way, is the battery cycle and ID recognized correctly for your model? Do you mind upload a copy of current ioreg? Also it seems that the battery patch is not required?

Chris2fourlaw, 2021-05-23:
Yeah, my battery was recognized by VirtualSMC's SMCBatteryManager.kext automatically

[image]

IOReg:
[Chris’s MacBook Pro.zip](https://github.com/zhen-zen/YogaSMC/files/6528762/Chris.s.MacBook.Pro.zip)

zhen-zen, 2021-05-23:
[YogaSMC_info.kext.zip](https://github.com/zhen-zen/YogaSMC/files/6528845/YogaSMC_info.kext.zip)

Can you try if this fixed the two unknown field?

Chris2fourlaw, 2021-05-23:
[image]

Works!

zhen-zen, 2021-05-23:
Thanks! Could you attach another ioreg for this working one? I guess this issue is done then and feel free to join the gutter chat.

I'm planning to release 1.5.1 today at https://github.com/zhen-zen/YogaSMC/commit/be713b7aed14dcaf9cf171c135c88ccd42f3e94c. The release version there should be the same as local build I tested.

I was a bit rushed to test the release kext for 1.5.0 and that causes headaches. This shouldn't happen in future releases again.

zilchulant, 2021-05-23:
As YogaSMC 1.5 made my hackintosh (Ideapad 5-15IIL05 with i7-1065g7) also kernelpanic, I am happy to see that it should work now. I will try the new version tomorrow, should have pretty similar outcome ✊

Chris2fourlaw, 2021-05-23:
Here's the ioreg
[Chris’s MacBook Pro.zip](https://github.com/zhen-zen/YogaSMC/files/6528979/Chris.s.MacBook.Pro.zip)

closed 2021-06-11 by zhen-zen