zhen-zen/YogaSMC #141: Preferences Error - Thinkpad T460s Mac OS 11.3.1

meshackvictor, 2021-05-23:
I have installed YogaSMC v1.5.0 on Thinkpad t460s running Mac os Big Sur 11.3.1. I can successfully control the fan but when i try to open the prefpane from the setting i get the following error

**Preferences Error**
Could not load YogaSMCPane preference pane.

Is there something that i am not doing right ? Thanks.

zhen-zen, 2021-05-23:
That's a GateKeeper issue. Try `xattr -cr ~/Library/PreferencePanes/YogaSMCPane.prefPane`. Please reopen if it doesn't work.

closed 2021-05-23 by zhen-zen

meshackvictor, 2021-05-23:
> That's a GateKeeper issue. Try `xattr -cr ~/Library/PreferencePanes/YogaSMCPane.prefPane`. Please reopen if it doesn't work.

This also doesnt work

zhen-zen, 2021-05-24:
You can just reopen this one...

NotMifa, 2022-04-21:
facing the same issue here

NotMifa, 2022-04-21:
I tried the command but I get this:

Last login: Wed Apr 20 14:20:32 on console
yashxavier@Yashs-HackBook-Pro ~ % sudo xattr -cr ~/Library/PreferencePanes/YogaSMCPane.prefPane
Password:
xattr: No such file: /Users/yashxavier/Library/PreferencePanes/YogaSMCPane.prefPane
yashxavier@Yashs-HackBook-Pro ~ %

zhen-zen, 2022-04-21:
> I tried the command but I get this:
>
>
>
> Last login: Wed Apr 20 14:20:32 on console
>
> yashxavier@Yashs-HackBook-Pro ~ % sudo xattr -cr ~/Library/PreferencePanes/YogaSMCPane.prefPane
>
> Password:
>
> xattr: [...]

Have you checked the command in the linked issue?