zhen-zen/YogaSMC #146: Brightness Key

pierpaolodimarzo, 2021-07-08:
Hi zhen-zen,
I need your great help ... 🙏🏻
I can't get the lum keys to go on the Corebook Pro
I have tried with both SSDT and DSDT having EC Brightness Key enabled on the kext, but if I press on the keys, after numerous repetitions it only moves a notch.

Thanks a lot again for your availability.

I used this issue off topic as I had no other contact availability. Sorry.

zhen-zen, 2021-07-08:
I don't have experience on other vendors and I can't guarantee the result. Please attach ioreg & ACPI dump. Meanwhile, please check if it works in linux.

pierpaolodimarzo, 2021-07-08:
[SysReport.zip](https://github.com/zhen-zen/YogaSMC/files/6782989/SysReport.zip)
[MacBook Pro di Pierpaolo.ioreg.zip](https://github.com/zhen-zen/YogaSMC/files/6782993/MacBook.Pro.di.Pierpaolo.ioreg.zip)

these are the two files you asked me ... thanks for the support

zhen-zen, 2021-07-09:
`_Q09` and `_Q10` (why decimal here!) should be corresponding keys and work with BrightnessKeys.
1. Do they work in Linux?
2. Does brightness slider in preference work?
3. Does traditional way work? https://www.tonymacx86.com/threads/guide-patching-dsdt-ssdt-for-laptop-backlight-control.152659/
4. Please check debug log from BrightnessKeys with DebugEnhancer.

pierpaolodimarzo, 2021-07-09:
1. I don't work in Linux
2. The slider Brightness work fine
4. How use the DebugEnhancer?

for better communication, my telegram is @ pierpaolo2395
Would it be possible to hear us there? 🙏

zhen-zen, 2021-07-09:
Just grab any Linux LiveCD for testing in case a firmware issue. No need for installation.

DebugEnhancer just provides a OOBE experience for debugging on Big Sur+. The rest is the same approach for getting log from Lilu plugin.

pierpaolodimarzo, 2021-07-09:
ok, I can try a Linux Live CD, what exactly should I do?

zhen-zen, 2021-08-14:
Oops... Missed this issue. Just check if the brightness key works in Linux. Some vendors have broken implementation that only works in Windows.

Overc1ocker, 2021-08-23:
I also have a laptop that has non-working brightness keys. It is a HP notebook. Ive tried the BrightnessKeys kext and the Q11 to XQ11 and Q12 to XQ12 renames with notifying PS2k. None of these seem to work. It does work properly in linux and both keys seem to send the same ps2 codes to the computer.

Ive resorted to disabling F2 and F3 and remapping them to brightness for now

zhen-zen, 2021-08-23:
Try unload the hp-wmi module in Linux (modprob -r or blacklist at boot) https://github.com/torvalds/linux/blob/master/drivers/platform/x86/hp-wmi.c and try if brightness keys still work. If not, it's possible to achieve it through the WMI interface. I have implemented that part in DYVPC but haven't tested nor linked to the actual key.

Overc1ocker, 2021-08-23:
It does not appear as though removing HP-WMI does anything. EVTEST on linux behaves the same way as on mac when it comes to the brightness keys... Code ab is sent on both brightness up and brightness down. How does linux know which is which?

Update: I think I have something.
/dev/input/event5:      Video Bus

`Input device ID: bus 0x19 vendor 0x0 product 0x6 version 0x0
Input device name: "Video Bus"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 224 (KEY_BRIGHTNESSDOWN)
    Event code 225 (KEY_BRIGHTNESSUP)
    Event code 227 (KEY_SWITCHVIDEOMODE)
    Event code 241 (KEY_VIDEO_NEXT)
    Event code 242 (KEY_VIDEO_PREV)
    Event code 243 (KEY_BRIGHTNESS_CYCLE)
    Event code 244 (KEY_BRIGHTNESS_ZERO)
    Event code 245 (KEY_DISPLAY_OFF)
Properties:
Testing ... (interrupt to exit)
Event: time 1629696665.970658, type 1 (EV_KEY), code 225 (KEY_BRIGHTNESSUP), value 1
Event: time 1629696665.970658, -------------- SYN_REPORT ------------
Event: time 1629696665.970682, type 1 (EV_KEY), code 225 (KEY_BRIGHTNESSUP), value 0
Event: time 1629696665.970682, -------------- SYN_REPORT ------------
Event: time 1629696668.327183, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 1
Event: time 1629696668.327183, -------------- SYN_REPORT ------------
Event: time 1629696668.327206, type 1 (EV_KEY), code 224 (KEY_BRIGHTNESSDOWN), value 0
Event: time 1629696668.327206, -------------- SYN_REPORT ------------
`

Overc1ocker, 2021-08-23:
MacOS also fails to detect my laptop lid switch which EVTEST sees as /dev/input0

```
Input device ID: bus 0x19 vendor 0x0 product 0x5 version 0x0
Input device name: "Lid Switch"
Supported events:
  Event type 0 (EV_SYN)
  Event type 5 (EV_SW)
    Event code 0 (SW_LID) state 0
Properties:
Testing ... (interrupt to exit)
Event: time 1629696560.794591, type 5 (EV_SW), code 0 (SW_LID), value 1
Event: time 1629696560.794591, -------------- SYN_REPORT ------------
Event: time 1629696561.663948, type 5 (EV_SW), code 0 (SW_LID), value 0
Event: time 1629696561.663948, -------------- SYN_REPORT ------------
```
Any idea what causes this?

zhen-zen, 2021-08-25:
"Video Bus" should be an ACPI driver per https://github.com/torvalds/linux/blob/master/drivers/acpi/acpi_video.c, so do "Lid Switch": https://github.com/torvalds/linux/blob/master/drivers/acpi/button.c . Events are sent to some ACPI devices which didn't report it correctly. Can you try blacklist the `hp-wmi` driver so it will initialize nothing at boot?

Overc1ocker, 2021-08-25:
I blacklisted hp_wmi completely and tested it. All laptop functions still work. Here's all the buses evtest sees now.

```
Available devices:
/dev/input/event0:      Lid Switch  (Lid switch) 
/dev/input/event1:      Power Button
/dev/input/event2:      AT Translated Set 2 keyboard
/dev/input/event3:      HP Wireless hotkeys
/dev/input/event4:      PC Speaker
/dev/input/event5:      Video Bus (Contains my brightness keys)
/dev/input/event6:      SynPS/2 Synaptics TouchPad
/dev/input/event7:      G2Touch Multi-Touch by G2TSP 
/dev/input/event8:      HP TrueVision HD Camera: HP Tru
/dev/input/event9:      HDA Intel PCH Mic
/dev/input/event10:     HDA Intel PCH Headphone
/dev/input/event11:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event12:     HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event13:     HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event14:     HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event15:     HDA Intel PCH HDMI/DP,pcm=10
```
Note: before I blacklisted HP_WMI at boot, there was a /dev/input/event(x):  HP_WMI Hotkeys
I tested the bus. It never registered any keys.

zhen-zen, 2021-08-28:
In this case, the events seem to be reported according to the ACPI spec and should be received by BrightnessKeys. One possible cause is that EC is not initialized correctly, you will need to study the DSDT. Another one is that the event is sent to other devices. In this case, ACPI rename should work at least.