zhen-zen/YogaSMC #156: Can't boot with YogaSMC.kext on ThinkPad X1C8

HJebbour, 2022-01-28:
As said on the title, I cannot boot with YogaSMC.kext loaded on Lenovo ThinkPad X1 Carbon Gen 8 (20U9-0013FE).
My CPU: i7-10510U

zhen-zen, 2022-01-31:
The kext is not designed for a specific system. You might need to go over your config files or upload logging / early boot screenshot for debugging. Close until more details are provided.

HJebbour, 2022-01-31:
Thank you for your reply, please use this link to view the bootlogs (IMG) and OC (TXT) https://www.icloud.com/iclouddrive/08bbbhwfgA0w5ZaywjNZeSDZg#Logs

zhen-zen, 2022-01-31:
I can't access the iCloud Drive. Can you just drag the attachment to the editor below?

HJebbour, 2022-01-31:
[image]
[image]
[opencore-2022-01-31-185853.txt](https://github.com/zhen-zen/YogaSMC/files/7974301/opencore-2022-01-31-185853.txt)

Sorry for the inconvenience, here's again the logs.

HJebbour, 2022-02-02:
Any update or do you want me to provide more logs? Thank you.

zhen-zen, 2022-02-07:
To narrow down the issue, can you try removing either the `YogaHIDD` dict or the `ThinkVPC` dict in Info.plist ? Meanwhile, have you tried `YogaSMCAlter`?

HJebbour, 2022-02-07:
Unfortunately none of the above worked, I even tried YogaSMCAlter without YogaHIDD & ThinkVPC dict.
[image]

HJebbour, 2022-02-07:
Here's kernel panic log, I hope this can help:
[YogaSMC_kernel-panic.txt](https://github.com/zhen-zen/YogaSMC/files/8017970/YogaSMC_kernel-panic.txt)

HJebbour, 2022-02-12:
I managed to boot with YogaSMC.kext when I removed one dict only (YogaWMI), but Yoga app and prefPane isn't working properly and ECAccess is Basic instead of RW, SSDT-ECRW is already loaded. The weird thing is, when I patched plist file and removed YogaWMI, after first reboot it detected the laptop as "Think" in Class with Think tab, but after second reboot and so, it shows "YogaHIDD" in class with no Think tab in preferences.
Please see below screenshots.

**First reboot**
[image]
[image]
[image]

**Second reboot**
[image]

When I open the prefpane from system preferences for the second time it shows this
[image]

HJebbour, 2022-02-12:
I removed "YogaHIDD" dict from plist and now it is recognized as "Think". I replaced SSDT-ECRW with another one I found in a repo (I compared it with the sample file and IT'S THE SAME) and now ECAccess is RW, but I don't know how it worked even if it is the same file.
I don't know if I can add more features or enhance it (See below screenshots), but YogaSMC seems usable now with the patches I've done (of course with your help, and how you gave me the troubleshooting starting point with Info.plist)
[image]
[image]
[image]

zhen-zen, 2022-03-07:
I'm sorry for unable to response promptly. I have tested YogaHIDD coexistence earlier, but not sure why it was preferred on yours. Will look into that and find a fix.

Baio1977, 2022-09-21:
I have identical problem, can't boot with YogaSMC.kext loaded on Lenovo Thinkpad T14 I5 10310U.[(https://github.com/Baio1977/Lenovo-Thinkpad-T14-)](https://github.com/Baio1977/Lenovo-Thinkpad-T14-)

zhen-zen, 2022-10-23:
Closed as #173 is resolved. Please reopen if the issue persists.

closed 2022-10-23 by zhen-zen

HJebbour, 2022-11-19:
> Closed as #173 is resolved. Please reopen if the issue persists.

With the new kext macOS is booting now, but pref is showing unsupported (see screenshot) and app is not doing anything.

https://user-images.githubusercontent.com/44072897/200417271-f8360048-6875-4fea-9e28-b13f60c4cf9c.png

HJebbour, 2022-11-19:
> Closed as #173 is resolved. Please reopen if the issue persists.

With the new kext macOS is booting now, but pref is showing unsupported (see screenshot) and app is not doing anything.
[image]

Baio1977, 2022-11-23:
You need ACPI patches to have the desired functions, I've already written to you to see the repository of my T14 there is an SSDT that integrates everything you need.
https://github.com/zhen-zen/YogaSMC/tree/master/YogaSMC/SSDTSample