zhen-zen/YogaSMC #159: YogaSMC (1.5.1) System Preferences error when running it again (Big Sur)

mackonsti, 2022-03-23:
Hello @zhen-zen thank you for spending time on this extension for us.

I have an IdeaPad **S145-14iWL** so I do believe my laptop is compatible with your kext and work.

The kext is loaded fine, using latest BigSur 11.6.5 with OpenCore 07.9.

When I launch the preference pane, it switches to YogaSMC, then I return to main System Preferences icons via the main button (on top with the many black dots). When I enter again YogaSMC pane, I get this error:

[image]

If I press OK, and try again, it launches OK.

I noticed that if I start the System Preferences and run YogaSMC pane, then quit/exit System Preferences completely, and restart System Preferences and relaunch YogaSMC, I do _not_ get this error.

I suspect that YogaSMC session is **not** properly killed or exited, despite clicking on the dots-button (on top) ; otherwise, the relaunch of YogaSMC (without quitting System Preferences) doesn't detect an existing/open session, perhaps, hence the error?

Have you come across this or anyone else?

Thanks for your feedback on this.

zhen-zen, 2022-04-14:
I thought this issue is gone but not sure why it happened again. There's few documents regarding prefpane development and the debugging is difficult since it couldn't be launched individually. Don't have an idea right now.

experimentalperson, 2022-10-27:
I also had the same problem, I solved by opening first YogaSMCPane from Package Contents, give the permission to execute it then add again the prefPane and it will work

experimentalperson, 2022-10-27:
@mackonsti  Yes exactly, inside the package contents there is another executable that has to be given permission to execute, you can open it and you will see a terminal window, and you have to unlock from security that executable too

;)

zhen-zen, 2022-10-30:
The executables are not signed and are blocked by Gatekeeper. You will need to remove the quarantine manually or control-click the icon and click `open`.

mackonsti, 2022-11-26:
Hi @zhen-zen I have code-signed it locally but it still crashes when clicking on the icon again inside System Preferences window 😢 

```
% cd ~/Library/PreferencePanes/
% sudo codesign -f -s - YogaSMCPane.prefPane
YogaSMCPane.prefPane: replacing existing signature
```

So @0x090909 I do not clearly understand what you did.
Do you **not** have the Pane crashing if you relaunch it from inside System Preferences?
I was under the impression that if you'd `codesign` it, it would stop crashing...
Thanks to both.