zhen-zen/YogaSMC #161: read fan failed

0xMargo, 2022-04-07:
hi @zhen-zen, 
my laptop is ThinkPad L380, I assume it compatible with YogaSMC
I can set my keyboard backlight from YogaSMC PreferencePane but still can't read fan speed
can you take look what I missed?

[L380.zip](https://github.com/zhen-zen/YogaSMC/files/8439500/L380.zip)

zhen-zen, 2022-04-14:
Probably the classic interface for fan reading doesn't present in your laptop. Usually X and T series has the most feature available while E and L series only has a subset. If there's no way to obtain fan readings in Linux, then there's nothing this driver could do. (Arch linux wiki is a good source even for LiveCD of a different distro)

zhen-zen, 2022-04-14:
Close #161 for now. Feel free to reopen the issue if there's further update.

mackonsti, 2022-04-21:
Hi @0xMargo if it makes you feel better, I don't get any fan readings on my IdeaPad S145-14iWL either, not sure what's missing or what is the ACPI/BIOS not reporting to the OS... I don't see why they would hide such info.

Thanks @zhen-zen as well.

zhen-zen, 2022-05-08:
Actually it's dangerous to poke around EC registers. Sometimes the firmware is not programmed properly and it might lead to thermal runaway. Regarding fan speed and other temperature readings, actually there's a standard way to represent them in ACPI. But OEMs rarely follow the spec or reference code, trying to create their proprietary interface. For Lenovo case, it's more likely caused by legacy codebase.

mackonsti, 2022-05-08:
@zhen-zen I do not have any readings of fans speed, and I am trying to checking out your SSDT-THINK and see how it can fit with my own IdeaPad S145 ACPI/DSDT code, but I don't see matching things or devices.

Could I please bug you to expand more on what YogaSMC is expecting as device/readings/sensors etc. by asking you to tell us how these values are encountered/managed in your DSDT?
```
    External (_SB.PCI0.LPCB.EC.HKEY, DeviceObj)       // HKEY path
    External (_SB.PCI0.LPCB.EC.HFSP, FieldUnitObj)    // Fan control register
    External (_SB.PCI0.LPCB.EC.HFNI, FieldUnitObj)    // Fan control register
    External (_SB.PCI0.LPCB.EC.VRST, FieldUnitObj)    // Second fan switch register
```
None of the above EC keys are found in my DSDT 😞 but if we had a _sample_ of how your DSDT looks in structure for these registers, I could try to see if they have different names. Are they defined inside EC (renamed from EC0 for me) ? Are they found in some `OperationRegion` sections/definitions?

Thanks in advance for your kind help.

zhen-zen, 2022-08-12:
@0xMargo 
For reading and setting the fan speed, they worked  through a fixed interface that only presents on major ThinkPad models, which is missing on the ACPI dump provided. Both the method and the registers couldn't be found. It's highly possible that EC related features are not fully implemented in ACPI.

@mackonsti 
As mentioned above, there's no use studying the ACPI table for ThinkPad. The driver expects nothing from Idea models, since there's no standard interface. You can only find out which register could be sensor reading on your mono. It's usually a 8 bit register. You can try `ioio -s YogaVPC ReadECName NAME` with the 4 character name under EC to get the reading from dmesg. Then stress the system and see if the reading changes accordingly. After figuring them out, you can try to fit them in your kext plist. i.e.https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMC/Info.plist#L90

closed 2022-08-12 by zhen-zen