zhen-zen/YogaSMC #163: Basic ECAccess

Ziimmie, 2022-05-08:
Hello,

I just installed this plugin tonight, in hopes of saving some battery by enabling the Fan Stop feature. There is one problem though, and that is that the ECACcess is stuck at Basic. I have ECEnabler.Kext enabled. Not sure what to do with the SSDT-ECRW.dsl file though. I saved the file again under MaciASL as a AML file so OpenCore would recognize this, bu that didn't seem to do anything. Any help would be greatly appreciated.

Thanks!

zhen-zen, 2022-05-08:
OC will not discover aml automatically and you will need to add the filename along with other amls. Also fan won't consume much energy.

closed 2022-05-08 by zhen-zen

Ziimmie, 2022-05-08:
I'm sorry, but do you mean add the AML files into OC Configurator (or in my case OCAuxiliaryTools)?