zhen-zen/YogaSMC #165: Where can I get latest build 1.5.2?

cyberfunk, 2022-06-28:
Hackintool always gives current build 1 dot release above what is available at GitHub sites. Using https://dortania.github.io/builds/ I can get the latest builds of kexts for my system. YogaSMC is not available there. Can you tell me where to get the latest build 1.5.2 for YogaSMC?

zhen-zen, 2022-06-28:
Like all other kexts, version bump happens after each release is out. So literally you can only get a prebuilt "latest" kext from GitHub Actions, which is not a release build.

closed 2022-06-28 by zhen-zen

cyberfunk, 2022-06-28:
> Like all other kexts, version bump happens after each release is out. So literally you can only get a prebuilt "latest" kext from GitHub Actions, which is not a release build.

So you're saying the downloads available at the Dortania site are all GitHub Actions builds? How can I get YogaSMC 1.5.2 using GitrHub Actions? Is that a script I need to run? I've never used it so I'm unfamiliar with the term.

zhen-zen, 2022-06-28:
I don't know how their website works. The archives of GitHub Actions have expire dates so you are not likely to download builds from last year either.
Meanwhile, as I wrote earlier, the builds not shown in release are development build. And the project is also inactive for a while.
Just stick to the latest release build if you are not familiar with it.

cyberfunk, 2022-06-28:
> I don't know how their website works. The archives of GitHub Actions have expire dates so you are not likely to download builds from last year. Meanwhile, as I wrote earlier, the builds not shown in release are development builds. However the project is inactive for a while. Just stick to the latest release build if you are not familiar with it.

Ok thanks