zhen-zen/YogaSMC #166: Failed to connect to YogaSMC on Lenovo Yoga (ThinkPad) S1

Dobikot, 2022-07-19:
Hi! I tried to use YogaSMC and... i got this 
[image]
and i tried again and again, re looking to files and nothing...
If someone had a similar problem and solved it, then please help (here if someone needs my files in EFI)
[image]
[image]
[image]
thanks in advance to those who answer
(Sorry if my english is bad. I just from Ukraine)

zhen-zen, 2022-08-01:
I'm not familiar with S serie. All past developments are based on ThinkPad X, T, E series. Although I can't guarantee it, can you upload the ACPI dump?

Dobikot, 2022-08-09:
> 
Ok
here it is
[ACPI.zip](https://github.com/zhen-zen/YogaSMC/files/9289793/ACPI.zip)

zhen-zen, 2022-08-11:
Please upload the dump of original ACPI tables instead of hotpatches.

Dobikot, 2022-08-12:
what is ACPI dump? (yea I'm dumb)

zhen-zen, 2022-08-12:
It seems to be a configuration issue now. Get familiar with the dortania guides, such as:
https://dortania.github.io/OpenCore-Install-Guide/config.plist/#adding-your-ssdts-kexts-and-firmware-drivers
https://dortania.github.io/Getting-Started-With-ACPI/

closed 2022-08-12 by zhen-zen