zhen-zen/YogaSMC #167: HP da1023nia has different interface in various product line

Edwardwich, 2022-07-20:
APCI Dump :
https://github.com/Edwardwich/hp-DA1023nia-Hackintosh/tree/main/Opencore-Debug/SysReport/ACPI
ioreg:
[Ali’s MacBook Pro.zip](https://github.com/zhen-zen/YogaSMC/files/9150399/Ali.s.MacBook.Pro.zip)

[image]

[image]
WFDE:
[image]
WFTE:
[image]
WMI1
[image]
DYVPC:
[image]

Mifaxav, 2022-11-24:
are hp's supported with yogasmc???

Edwardwich, 2022-11-25:
> are hp's supported with yogasmc???

some HP supported

closed 2024-10-22 by Edwardwich