zhen-zen/YogaSMC #168: Feature request: Auto set backlight level based on activity

peddy22, 2022-07-23:
can you use ioreg to detect idle and then set an option to auto-dim the keyboard when it's idle for 5s or more and re-enable once the activity manages? thanks

peddy22, 2022-07-23:
if not, could you possibly post the applescript to set the level to low/med/high and i will contribute the code for 'auto'. the codebase is too large for me to understand it.

peddy22, 2022-07-27:
ideally we'd use apple's io system but i'm not afraid of writing a shell script, i just need some pointers on where to start.

the defaults output for org.zhen.YogaSMC doesn't change if I change the option via the system tray applet.

zhen-zen, 2022-08-01:
The backlight could be controlled by `ioio -s YogaVPC BacklightLevel level`. Due to the constraint in Idea models, the backlight level will only stay at low when set by the driver. ThinkPad variant should be fine.

Also, I suppose it would be easier to implement it in the driver level. Currently YogaSMC ignores a message from VoodooPS2 which is used for VoodooI2C palm rejection on each keypress.

peddy22, 2022-08-01:
Thanks, this should work nicely for my machine with a custom display board

zhen-zen, 2022-08-11:
@peddy22 Can you test the linked PR?

peddy22, 2022-08-11:
Sure

peddy22, 2022-08-12:
Awesome, it looks like it works (you can disregard my comments on the PR). Only improvement I could suggest is also making it activate based on mouse movement, not keyboard movement. Although that is a preferential issue I will leave to you. 

Thanks again! ^_^

zhen-zen, 2022-08-14:
Can you try the latest commit?

PR #171 (2022-10-09): YogaVPC: support setting timeout on keyboard backlight
Close #168
closed 2022-10-09 by zhen-zen