zhen-zen/YogaSMC #170: Can't set fan speed using ioio

iXPert12, 2022-08-05:
Hello. I'm trying to write a script to manually set fan speed, something like thinkfan on linux. So far i tried using ioio like this:

ioio -s ThinkVPC FanLevel 5

But the fan speed doesn't change. I'm using a Thinkpad T430. Could you please guide me which command to use?

iXPert12, 2022-08-05:
My issue is that the fan never goes full mode, even if the processor heats up to 95 C, and i have to set fan speed manually to full mode using YogaSMCNC

zhen-zen, 2022-08-09:
If it behaves the same without the kext, I suppose it's a firmware issue. There should be a few T430 users and I'm not aware of the issue.

For ioio, please use FanSpeed as the keyword. Acceptable values are 0-7, 0x47 and 0x84.

closed 2022-10-22 by zhen-zen