zhen-zen/YogaSMC #172: Not loading

Mifaxav, 2022-08-24:
Ive copied the kext to EFI/OC/Kexts and also added it in the config.plist area. On boot, ive set YogaSMCNC to start up on boot. YogaSMCNC now comes up with a 'Failed to connect to YogaSMC' error. Ive just changed my EFI from a previous one and it used to work! But its not working right now :(

any idea what might be the issue here?

I can also send the EFI file i am using and btw, i am on MacOS Monterey on a T480

zhen-zen, 2022-10-09:
There shouldn't be much difference after the OS upgrade. There are no debug files but please make sure if you can find `ThinkVPC` in ioreg.

Mifaxav, 2022-10-09:
Ill do that. Quite new to this community so any idea about the sub section it might be in?

Mifaxav, 2022-10-10:
i was able to search 'ThinkVPC' in ioreg but found no results. any idea?

zhen-zen, 2022-10-10:
So it seems that the kext is not loaded at all. Check the output of `kextstat | grep YogaSMC` and see if there's any matches.
If not, double check you plist and make sure it's enabled and similar to other entries.

Mifaxav, 2022-10-21:
downgraded to big sur because i already had other problems such as bluetooth etc etc. But, another request from my side. I really wanted to know the percentage of the 2 batteries that are there on my lenovo T480 while on macos. Any chance its possible to get this data from the laptop and display it somewhere in yogasmcnc or the pref pane???

zhen-zen, 2022-10-22:
It's off the scope for this driver as battery readings are handled separately under `BAT0`/`BAT1` device. There's no present interface under `HKEY` for such information nor did it fit current framework.

closed 2022-10-22 by zhen-zen

Mifaxav, 2022-10-22:
Alright! Good to know