zhen-zen/YogaSMC #173: Can't boot with YogaSMC.kext on ThinkPad T14 i5 10310U .....

Baio1977, 2022-09-22:
I have the same problem that the user has on Thinkpad X1C8 same Series CPU CometLake. https://github.com/zhen-zen/YogaSMC/issues/156.
Here the Thinkpad T14 reposity, includes Ioreg + files extracted with EFI Debug.
[](https://github.com/Baio1977/Lenovo-Thinkpad-T14-)

zhen-zen, 2022-10-09:
So it works after removing the YogaHIDD entry?

Baio1977, 2022-10-09:
Here is my Kext on the left and the original on the right.
As you can see I removed 2 items from the list.

[image]

zhen-zen, 2022-10-10:
I'm still trying to figure out a way to clean up the mess when starting the WMI stuff.
Also I wonder if you can help to test whether it works by just removing only one of the items in the list.

Baio1977, 2022-10-10:
TEST 1: removed from the plist only YogaHIDD the PC crashes in verbose.
TEST 2: removed from YogaWMI plist and PC boot regular, everything works.

YogaWMI is the problem.

THANK YOU.
For any evidence I am available.

zhen-zen, 2022-10-22:
Can you try if #176 fix the issue?

Baio1977, 2022-10-22:
One more block in verbose, the usual and right after a reboot

zhen-zen, 2022-10-22:
Please try the latest build on the master branch. The issue should be fixed now.

Baio1977, 2022-10-22:
Now it's ok I attach ioreg, I'll test this version and leave my comment here.
Thank you very much for your effort.
[YogaSMC 2.0 OK.zip](https://github.com/zhen-zen/YogaSMC/files/9844225/YogaSMC.2.0.OK.zip)

closed 2022-10-23 by Baio1977

HJebbour, 2022-11-07:
> Can you try if #176 could fix the issue? Also @HJebbour

With the new kext macOS is booting now, but pref is showing unsupported (see screenshot) and app is not doing anything.
[image]

Baio1977, 2022-11-08:
https://github.com/HJebbour/ThinkPad-X1C8-Hackintosh this the EFI use ??
If you are missing ALL the ACPI part about it.
Post ioreg of your Thinkpad.

HJebbour, 2022-11-08:
> https://github.com/HJebbour/ThinkPad-X1C8-Hackintosh this the EFI use ?? If you are missing ALL the ACPI part about it. Post ioreg of your Thinkpad.

Yes this is my EFI actually, but I don't understand your question?

Baio1977, 2022-11-09:
Specific SSDTs are needed to be able to monitor the functions of the Kext YogaSMC.
Can you leave the ACPI files extracted from your Bios here ???

If you check my reposity you see everything I talk about.
(https://github.com/Baio1977/Lenovo-Thinkpad-T14-/blob/main/0.8.5/EFI/OC/ACPI/SSDT-T14Gen1.dsl)