zhen-zen/YogaSMC #178: Don't read fan on thinkpad L560

mbarbierato, 2022-10-22:
Hi zen i have a problem with fan on lenovo L560, this is ioreg and sysreport
[MacBookPro.ioreg.zip](https://github.com/zhen-zen/YogaSMC/files/9844176/MacBookPro.ioreg.zip)
[SysReport.zip](https://github.com/zhen-zen/YogaSMC/files/9844177/SysReport.zip)

zhen-zen, 2022-10-22:
I didn't notice anything missing. Is the fan reading available in Windows or Linux?

mbarbierato, 2022-10-22:
Yes it's avaiable
[image]

zhen-zen, 2022-10-22:
It's a bit weird, only 1 fan presents and the EC layout is expected so there shouldn't be any difference with other models. If you have `ECEnabler` installed, can you try `ioio -s ThinkVPC ReadECName HSPD` and check kernel log output? Meanwhile, is fan speed reading available from SMC?

mbarbierato, 2022-10-24:
Hi Zhen,
result of offset 0x84 is "ioio: setting property 'ThinkVPC:ReadECOffset' as number to 132 (0x84)"
result of offset 0x85 is "ioio: setting property 'ThinkVPC:ReadECOffset' as number to 133 (0x85)"
the kernel output log is 
[    1.808205]: YSMC - Info: HKEY::ThinkVPC updateAdaptiveKBD 0 fffffffb
[    1.849034]: YSMC - Info: HKEY::ThinkVPC KeyboardLocale evaluation failed
[   35.870599]: 2 duplicate reports for Sandbox: icdd(449) allow mach-lookup com.apple.FSEventsSandbox: icdd(449) allow file-read-data /Library/Image Capture/Devices1 duplicate report for Sandbox: icdd(449) allow file-read-data /Library/Image Capture/DevicesSandbox: ctkd(470) deny(1) mach-lookup com.apple.CoreServices.coreservicesdsuccessfully parsed 1/4 entries for file MacBookPro.ioreg.zipSandbox: ctkd(470) deny(1) mach-lookup com.apple.DiskArbitration.diskarbitrationdSandbox apply: heard[439] <bytes>Sandbox apply: fileproviderd[421] <bytes>3 duplicate reports for Sandbox: ctkd(470) deny(1) mach-lookup com.apple.DiskArbitration.diskarbitrationdSandbox: icdd(449) allow network-outbound /private/var/run/mDNSResponder9 duplicate reports for Sandbox: icdd(449) allow network-outbound /private/var/run/mDNSResponderSandbox: icdd(449) allow mach-lookup com.apple.DiskArbitration.diskarbitrationd2 duplicate reports for Sandbox: icdd(449) allow mach-lookup com.apple.DiskArbitration.diskarbitrationdSandbox: icdd(449) allow mach-lookup com.apple.coreservices.launchservicesdYSMC - Info: ThinkVPC::YogaSMCUserClient registerNotificationPort subscribed

thanks in advance

mbarbierato, 2022-10-24:
This is the output of command 
[    0.667349]: ACPI: ACPI: RSDP 0x00000000B7FFE014 000024 (v02 LENOVO)RSDP 0x00000000B7FFE014 000024 (v02 LENOVO)
[    0.668817]: ACPI: ACPI: XSDT 0x00000000B7F8A000 00012C (v01 LENOVO TP-N1H   00000000 PTEC 00000002)XSDT 0x00000000B7F8A000 00012C (v01 LENOVO TP-N1H   00000000 PTEC 00000002)
[    0.671236]: ACPI: ACPI: FACP 0x00000000B7FF5000 0000F4 (v05 LENOVO TP-N1H   00001530 PTEC 00000002)FACP 0x00000000B7FF5000 0000F4 (v05 LENOVO TP-N1H   00001530 PTEC 00000002)
[    0.693466]: ACPI: ACPI: DSDT 0x00000000B7FCB000 024327 (v02 LENOVO SKL      00000000 INTL 20141107)DSDT 0x00000000B7FCB000 024327 (v02 LENOVO SKL      00000000 INTL 20141107)
[    0.695857]: ACPI: ACPI: FACS 0x00000000B7F93000 000040FACS 0x00000000B7F93000 000040calling mpo_policy_init for AMFI
[    1.241949]: ACPI: ACPI: UEFI 0x00000000B7FA9000 000042 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)UEFI 0x00000000B7FA9000 000042 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.255907]: ACPI: ACPI: SSDT 0x00000000B7FF7000 005366 (v02 LENOVO SaSsdt   00003000 INTL 20141107)SSDT 0x00000000B7FF7000 005366 (v02 LENOVO SaSsdt   00003000 INTL 20141107)
[    1.268613]: ACPI: ACPI: SSDT 0x00000000B7FF6000 0005C5 (v02 LENOVO PerfTune 00001000 INTL 20141107)SSDT 0x00000000B7FF6000 0005C5 (v02 LENOVO PerfTune 00001000 INTL 20141107)
[    1.281169]: ACPI: ACPI: HPET 0x00000000B7FF4000 000038 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)HPET 0x00000000B7FF4000 000038 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.293772]: ACPI: ACPI: LPIT 0x00000000B7FF3000 000094 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)LPIT 0x00000000B7FF3000 000094 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.306405]: ACPI: ACPI: APIC 0x00000000B7FF2000 0000BC (v03 LENOVO TP-N1H   00001530 PTEC 00000002)APIC 0x00000000B7FF2000 0000BC (v03 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.319022]: ACPI: ACPI: MCFG 0x00000000B7FF1000 00003C (v01 LENOVO TP-N1H   00001530 PTEC 00000002)MCFG 0x00000000B7FF1000 00003C (v01 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.331559]: ACPI: ACPI: WDAT 0x00000000B7FF0000 000134 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)WDAT 0x00000000B7FF0000 000134 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.344017]: ACPI: ACPI: SSDT 0x00000000B7FCA000 000248 (v02 LENOVO sensrhub 00000000 INTL 20141107)SSDT 0x00000000B7FCA000 000248 (v02 LENOVO sensrhub 00000000 INTL 20141107)
[    1.356504]: ACPI: ACPI: SSDT 0x00000000B7FC9000 0002D4 (v01 LENOVO EInkApp  00000000 INTL 20141107)SSDT 0x00000000B7FC9000 0002D4 (v01 LENOVO EInkApp  00000000 INTL 20141107)
[    1.368953]: ACPI: ACPI: SSDT 0x00000000B7FC8000 00019A (v02 LENOVO Sata0Ide 00001000 INTL 20141107)SSDT 0x00000000B7FC8000 00019A (v02 LENOVO Sata0Ide 00001000 INTL 20141107)
[    1.381453]: ACPI: ACPI: SSDT 0x00000000B7FC7000 000B49 (v02 LENOVO UsbCTabl 00001000 INTL 20141107)SSDT 0x00000000B7FC7000 000B49 (v02 LENOVO UsbCTabl 00001000 INTL 20141107)
[    1.393741]: ACPI: ACPI: SSDT 0x00000000B7FC6000 000120 (v02 LENOVO zpodd    00001000 INTL 20141107)SSDT 0x00000000B7FC6000 000120 (v02 LENOVO zpodd    00001000 INTL 20141107)
[    1.405878]: ACPI: ACPI: DBGP 0x00000000B7FC5000 000034 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)DBGP 0x00000000B7FC5000 000034 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.417920]: ACPI: ACPI: DBG2 0x00000000B7FC4000 000054 (v00 LENOVO TP-N1H   00001530 PTEC 00000002)DBG2 0x00000000B7FC4000 000054 (v00 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.429926]: ACPI: ACPI: SSDT 0x00000000B7FC3000 00075F (v02 LENOVO xh_rvp07 00000000 INTL 20141107)SSDT 0x00000000B7FC3000 00075F (v02 LENOVO xh_rvp07 00000000 INTL 20141107)
[    1.441531]: ACPI: ACPI: BOOT 0x00000000B7FC2000 000028 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)BOOT 0x00000000B7FC2000 000028 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.452879]: ACPI: ACPI: BATB 0x00000000B7FC1000 00004A (v02 LENOVO TP-N1H   00001530 PTEC 00000002)BATB 0x00000000B7FC1000 00004A (v02 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.464060]: ACPI: ACPI: SLIC 0x00000000B7FC0000 000176 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)SLIC 0x00000000B7FC0000 000176 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.475286]: ACPI: ACPI: SSDT 0x00000000B7FBF000 000E73 (v02 LENOVO CpuSsdt  00003000 INTL 20141107)SSDT 0x00000000B7FBF000 000E73 (v02 LENOVO CpuSsdt  00003000 INTL 20141107)
[    1.486307]: ACPI: ACPI: SSDT 0x00000000B7FFD000 000403 (v02 LENOVO CtdpB    00001000 INTL 20141107)SSDT 0x00000000B7FFD000 000403 (v02 LENOVO CtdpB    00001000 INTL 20141107)
[    1.497197]: ACPI: ACPI: MSDM 0x00000000B7FBD000 000055 (v03 LENOVO TP-N1H   00001530 PTEC 00000002)MSDM 0x00000000B7FBD000 000055 (v03 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.508077]: ACPI: ACPI: DMAR 0x00000000B7FBC000 0000A8 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)DMAR 0x00000000B7FBC000 0000A8 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.518612]: ACPI: ACPI: ASF! 0x00000000B7FBB000 0000A5 (v32 LENOVO TP-N1H   00001530 PTEC 00000002)ASF! 0x00000000B7FBB000 0000A5 (v32 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.528777]: ACPI: ACPI: FPDT 0x00000000B7FBA000 000044 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)FPDT 0x00000000B7FBA000 000044 (v01 LENOVO TP-N1H   00001530 PTEC 00000002)
[    1.538631]: ACPI: ACPI: UEFI 0x00000000B7F91000 00012A (v01 LENOVO TP-N1H   00001530 PTEC 00000002)UEFI 0x00000000B7F91000 00012A (v01 LENOVO TP-N1H   00001530 PTEC 00000002)
[... cut ...]
[10926.302927]: AppleACPIPlatformPower Wake reason: PWRB XDCI (User)
[11523.936151]: PMRD: kIOMessageSystemWillPowerOn to com.apple.driver.AppleACPIPlatform
[11523.942248]: PMRD: kIOMessageSystemCapabilityChange[3] to com.apple.driver.AppleACPIPlatform
[11523.958705]: PMRD: claimSystemWakeEvent(XHC, 0x1, AppleACPIPCI, 0x10000025f) aot 0 phase 0x27 add 1
[11526.006829]: PMRD: kIOMessageSystemCapabilityChange[3] to com.apple.driver.AppleACPIPlatform
[11526.013042]: PMRD: kIOMessageSystemHasPoweredOn to com.apple.driver.AppleACPIPlatform

zhen-zen, 2022-10-24:
Just realized I might forgot something fundamental. What do you mean by "don't read fan" and what's expected?

mbarbierato, 2022-10-24:
I mean the speed of the fans in the app is 0 and also in macfancontrol and also in other software
[image]

zhen-zen, 2022-10-30:
Does fan speed control work? The code is developed and confirmed working on X and T series, based on their EC layout. I'm not sure why L series broke when they have similar interface.

mbarbierato, 2022-11-01:
I confirm that's work also in e series i tried in e570 and e580