zhen-zen/YogaSMC #183: power modes

Mifaxav, 2022-11-02:
Hello @zhen-zen, few days ago i was wondering on how to get power saving mode for Big Sur on my hackintosh when i came across this. 
[image]
But it says DYTCRevision Unsupported. How do i fix this? btw, I am on a Lenovo T480 running Big Sur

zhen-zen, 2022-11-06:
You might need the extra in https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMC/SSDTSample/SSDT-THINK.dsl#L22. Please reopen if it didn't work.

Mifaxav, 2022-11-12:
[image]

Shows up as version 4.2 but not able to change the setting between my laptop being in quiet, performance or balanced mode.

Mifaxav, 2022-11-14:
also im not able to open this issue...

Mifaxav, 2022-11-17:
@zhen-zen does this work ok on your lenovo? Also, what model is it?

Mifaxav, 2022-11-26:
u sure there are no other patches needed??

Mifaxav, 2022-12-22:
Any update to this? I really do need a way to decrease power usage on my laptop as it only lasts for 3 hours

zhen-zen, 2022-12-22:
DYTC control should work for everybody after the interface is accessible. Can you use the debug version of DYTC and upload the `sudo dmesg | grep YSMC` log after opening the pane?

Mifaxav, 2022-12-23:
Alright, so i opened up the pref pane onto the general tab and ran 'sudo dmesg | grep YSMC' in the terminal, please correct me if i am doing something wrong. As with sudo, shell asked me for a password, i entered it and nothing... Any step i am missing?

zhen-zen, 2022-12-23:
Have you used the debug version of the kext?

Mifaxav, 2022-12-23:
nope. Will do that once i get home using an external drive

Mifaxav, 2023-01-03:
[   32.328397]: Waking up reference: 8Thread waiting on reference 8 woke upevaluation result: 8, allowed, cache,  1672747681Sandbox: iconservicesagen(429) deny(1) mach-lookup com.apple.tccd.systemSandbox: 1 duplicate report for iconservicesagen deny(1) mach-lookup com.apple.tccd.systemSandbox: iconservicesagen(429) deny(1) mach-lookup com.apple.windowserver.activeWaking up reference: 7Thread waiting on reference 7 woke upevaluation result: 7, allowed, cache,  1672748027Waking up reference: 9Thread waiting on reference 9 woke upevaluation result: 9, allowed, cache,  1672748027Sandbox: 1 duplicate report for iconservicesagen deny(1) mach-lookup com.apple.windowserver.activeSandbox: secd(362) deny(1) user-preference-write secdSandbox: AppSSOAgent(477) deny(1) file-issue-extension target:/System/Library/PrivateFrameworks/AppSSO.framework/Support/AppSSOAgent.app class:com.apple.app-sandbox.readSandbox: imagent(459) deny(1) mach-lookup com.apple.backupd.xpcSandbox: cloudd(391) deny(1) system-fsctl _IO('h', 47)Sandbox: imagent(459) deny(1) mach-lookup com.apple.backupd.xpcYSMC - Debug: HKEY::ThinkVPC BCTG 0x80000000
[   33.207726]: YSMC - Debug: HKEY::ThinkVPC BCSG 0x80000000
[   33.208658]: YSMC - Debug: HKEY::ThinkVPC BICG 0x80000000
[   33.209601]: YSMC - Debug: HKEY::ThinkVPC BDSG 0x80000000
[   33.210780]: YSMC - Debug: HKEY::ThinkVPC PSSG 0x70
[   33.210802]: YSMC - Debug: HKEY::ThinkVPC Battery 0x0
[   33.212163]: YSMC - Debug: HKEY::ThinkVPC updateMutestatus 0x2
[   33.213188]: YSMC - Debug: HKEY::ThinkVPC updateMuteLEDStatus 0x100
[   33.213319]: YSMC - Debug: HKEY::ThinkVPC updateMicMuteLEDStatus 0x101
[   33.214361]: YSMC - Debug: HKEY::ThinkVPC KeyboardLocale 0x80000000
[   33.214492]: YSMC - Debug: HKEY::ThinkVPC empty HKEY event
[   33.217401]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x2 result 0x0001f001
[   33.231123]: YSMC - Debug: HKEY::ThinkVPC BCTG 0x80000000
[   33.232037]: YSMC - Debug: HKEY::ThinkVPC BCSG 0x80000000
[   33.232952]: YSMC - Debug: HKEY::ThinkVPC BICG 0x80000000
[   33.233845]: YSMC - Debug: HKEY::ThinkVPC BDSG 0x80000000
[   33.234943]: YSMC - Debug: HKEY::ThinkVPC PSSG 0x70
[   33.234953]: YSMC - Debug: HKEY::ThinkVPC Battery 0x0
[   33.236590]: YSMC - Debug: HKEY::ThinkVPC BCTG 0x300
[   33.237746]: YSMC - Debug: HKEY::ThinkVPC BCSG 0x300
[   33.239209]: YSMC - Debug: HKEY::ThinkVPC BICG 0x70
[   33.240249]: YSMC - Debug: HKEY::ThinkVPC BDSG 0x700
[   33.241422]: YSMC - Debug: HKEY::ThinkVPC PSSG 0x70
[   33.241432]: YSMC - Debug: HKEY::ThinkVPC Battery 0x1
[   33.304366]: YSMC - Debug: HKEY::ThinkVPC BCTG 0x300
[   33.305589]: YSMC - Debug: HKEY::ThinkVPC BCSG 0x300
[   33.306703]: YSMC - Debug: HKEY::ThinkVPC BICG 0x70
[   33.307842]: YSMC - Debug: HKEY::ThinkVPC BDSG 0x700
[   33.308995]: YSMC - Debug: HKEY::ThinkVPC PSSG 0x70
[   33.309005]: YSMC - Debug: HKEY::ThinkVPC Battery 0x2
[   35.992193]: Sandbox: secd(362) deny(1) system-fsctl _IO('h', 47)Sandbox: ContainerMetadat(495) deny(1) user-preference-read kCFPreferencesAnyApplicationSandbox: 2 duplicate reports for ContainerMetadat deny(1) user-preference-read kCFPreferencesAnyApplicationSandbox: imagent(459) deny(1) mach-lookup com.apple.system.opendirectoryd.apiSandbox: com.apple.siri.C(498) deny(1) user-preference-read kCFPreferencesAnyApplicationSandbox: 2 duplicate reports for com.apple.siri.C deny(1) user-preference-read kCFPreferencesAnyApplicationSandbox: networkservicepr(369) deny(1) mach-lookup com.apple.securityd.xpcSandbox: networkservicepr(369) deny(1) file-write-unlink /private/var/folders/4t/d8dbtmmd5v7777kv1bv5ytth0000gn/C/mds/mdsObject.dbSandbox: 1 duplicate report for networkservicepr deny(1) file-write-unlink /private/var/folders/4t/d8dbtmmd5v7777kv1bv5ytth0000gn/C/mds/mdsObject.dbSandbox: routined(375) deny(1) mach-lookup com.apple.Maps.MapsSync.storeYSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x2 result 0x0001f001
[   37.894399]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x1ff result 0x0001f001
[   37.898672]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x12b001 result 0x08012b01
[   37.898677]: YSMC - Debug: HKEY::ThinkVPC Thermal Control Command set completed (DYTC)
[   37.898689]: YSMC - Debug: HKEY::ThinkVPC DYTC_FUNCTION_MMC on
[   37.898704]: YSMC - Debug: HKEY::ThinkVPC DYTCMode set to 0x2: see ioreg
[   39.131739]: Sandbox: com.apple.siri.C(498) deny(1) user-preference-read com.apple.internal.ckSandbox: com.apple.siri.C(498) deny(1) user-preference-read com.apple.assistant.supportYSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x2 result 0x08012b01
[   39.797689]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x1ff result 0x0001f001
[   39.797713]: YSMC - Debug: HKEY::ThinkVPC DYTCMode set to 0xf: see ioreg
[   40.231744]: Sandbox: studentd(435) deny(1) file-read-metadata /LibraryWaking up reference: 5Thread waiting on reference 5 woke upevaluation result: 5, allowed, cache,  1672748035Sandbox: 5 duplicate reports for studentd deny(1) file-read-metadata /LibrarySandbox: routined(375) deny(1) mach-lookup com.apple.Maps.MapsSync.storeWaking up reference: 12Thread waiting on reference 12 woke upevaluation result: 12, allowed, cache,  1672748035Waking up reference: 13Thread waiting on reference 13 woke upevaluation result: 13, allowed, cache,  1672748036macl_get_vnode: failed to resolve vnode: 2macl_get_vnode: failed to resolve vnode: 2macl_get_vnode: failed to resolve vnode: 2macl_get_vnode: failed to resolve vnode: 2macl_get_vnode: failed to resolve vnode: 2macl_get_vnod [... cut ...]
[   77.868713]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x2 result 0x0001f001
[   77.872027]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x1ff result 0x0001f001
[   77.872427]: YSMC - Info: HKEY::ThinkVPC DYTCMode command 0x13b001 result 0x0000000a (invalid argument)
[   77.872430]: YSMC - Info: HKEY::ThinkVPC DYTCMode toggle failed
[   77.873021]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x2 result 0x0001f001

Thats what i got

Also, im able to switch to performance mode with it showing DYTCFunction as Desk

Mifaxav, 2023-02-28:
also, whats the ioio command to get it to powersaving mode
and whats the level of confirmation on that?

Mifaxav, 2023-07-18:
Tried it again..

[59570.957857]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x2 result 0x0001f001
[59570.961809]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x1ff result 0x0001f001
[59570.962339]: YSMC - Info: HKEY::ThinkVPC DYTCMode command 0x13b001 result 0x0000000a (invalid argument)
[59570.962344]: YSMC - Info: HKEY::ThinkVPC DYTCMode toggle failed
[59570.963116]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x2 result 0x0001f001