zhen-zen/YogaSMC #187: Error mapping when connecting power and unplugging power

sunbos, 2022-12-22:
My laptop will incorrectly translate "Connect the power" and "Unplug the power" into backlightoff, what should I do

Mifaxav, 2022-12-22:
Which laptop do you have?

zhen-zen, 2022-12-22:
> My laptop will incorrectly translate "Connect the power" and "Unplug the power" into backlightoff, what should I do

This seems to be a common issue for ideapad_laptop since both events send the same code. I'm not sure if there's a better way to distinguish them.

sunbos, 2022-12-22:
YOGA 9 15MH5

sunbos, 2022-12-22:
My laptop:Yoga 9 15IMH5
I have a way to judge which one is plugged in and which one is unplugged from the DSDT
Through the test, I found that_ Q50 is the Connect the power ,_ Q51 is Unplug the power

sunbos, 2022-12-22:
This is my DSDT file
[System DSDT.dsl.zip](https://github.com/zhen-zen/YogaSMC/files/10288090/System.DSDT.dsl.zip)

sunbos, 2023-01-04:
Fn+Q(_Q78)
AC power connecting(_Q50)
AC power unplugging(_Q51)
power button (_Q5F)
laptop lid closed (_Q52)
laptop lid opened (_Q53)
Mixer output mute (F1)
Volume down or unmute(F2)
Volume up or unmute(F3)
Microphone Mute (F4, _Q3D)
Touchpad (F6,_Q22)
Network (F7, _Q28)
camera(F8,_Q2E)
Brightness up (F12, _Q1C)
Brightness down (F11, _Q1D)
IdeaVPC kbd backlight (Fn+Space, _Q1F)

Now F8 can map the camera key correctly, but the camera cannot be turned off
Fn+Q (_Q78) is always in Fn-Q cooling state
IdeaVPC kbd backlight (Fn+Space, _Q1F), the keyboard backlight cannot be mapped correctly
AC power connecting(_Q50)
AC power unplugging(_Q51)
Nor can it be mapped correctly