zhen-zen/YogaSMC #189: YogaSMCNC - SMC Pane

pierpaolodimarzo, 2023-01-08:
Hi Zen, I'm having a problem with the latest version of the YogaSMCNC app. 
After installing *.prefPane and the YogaSMCNC application, clicking on the application at the top of the status bar, and clicking on the "Preferences" item, the "Please install YogaSMCPane" error popup appears, while opening it directly from the "Preferences System" panel opens without problems.

[image]
[image]

I am attaching screenshots of the problem.

macOS Ventura 13.1 (22C65)

CobanRamo, 2023-01-08:
I have the same problem with Ventura, the system prefences can no longer be called up from the status bar.

[image]

Einstonoy, 2023-01-09:
It is interesting that my x1y3 appears opposite situation

dclive, 2023-01-22:
@CobanRamo : The version in your picture is an older 152 version; you might want to get the latest 153 version. 
That said, with Ventura / 13.1, using the 153 version, I have the same problem with a X1C6.

CobanRamo, 2023-01-23:
I don't think it's due to the version, it doesn't work under ventura.
Under Catalina and BigSur on the same computer with the same config is no problem.
[image]
[image]

https://user-images.githubusercontent.com/68192365/213952648-6abb91fc-2ccd-494d-8460-d4b06c86651e.mp4

bootlegseahorse, 2023-02-03:
The System Preference app was renamed to System Settings in Ventura, and I think that is exactly what caused the problem.

bibarub, 2023-02-07:
> The System Preference app was renamed to System Settings in Ventura, and I think that is exactly what caused the problem.

nope, the bundle id is still the same

one thing that might actually be related is broken permission management because of disabled AMFI. OCLP graphics patches need AMFI to be disabled on Ventura

5T33Z0, 2023-06-30:
Same for me. The behavior is the same on Sonoma beta 2, btw.

jozews321, 2024-02-18:
I made a PR to hopefully fix the launching issue. There is also an issue with the latest release prefpane in macOS 13+, for some reason it doesn't work after opening once and then going to another settings page, its resolved when compiling it again targeting 10.13 minimum.

Here are some binaries with the fix
 
[Builds.zip](https://github.com/zhen-zen/YogaSMC/files/14324664/Builds.zip)

CobanRamo, 2024-02-19:
We're riding a dead horse but I'll test it out tonight after work.
Thank you for your effort.

It's been years since @Zhen-Zen became active... :)

[image]

CobanRamo, 2024-02-19:
> I made a PR to hopefully fix the launching issue. 
> [Builds.zip](https://github.com/zhen-zen/YogaSMC/files/14324664/Builds.zip)

I tested it, thanks for your effort @
With this binary the call from “Prefences” works again...

Ergo, the problem of calling the “preferences” would be solved.
[image]

But now the “Start at Login” no longer works.
You always have to start it manually after a restart.
[image]

Otherwise it looks good.

5T33Z0, 2024-02-19:
Hmm, ok. For me, the "Start at Login" feature is of higher priority than the prefpane opening at first click.

jozews321, 2024-02-19:
> > I made a PR to hopefully fix the launching issue.
> > [Builds.zip](https://github.com/zhen-zen/YogaSMC/files/14324664/Builds.zip)
> 
> I tested it, thanks for your effort @jozews321 With this binary the call from “Prefences” works again...
> 
> Ergo, the problem of calling the “preferences” would be solved. ![211217181-b17739ca-2293-4ca4-9082-f6797895312f](https://private-user-images.githubusercontent.com/68192365/306037055-495b961a-c9eb-410e-b714-2d263f6ac25d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgzNjQ5NTYsIm5iZiI6MTcwODM2NDY1NiwicGF0aCI6Ii82ODE5MjM2NS8zMDYwMzcwNTUtNDk1Yjk2MWEtYzllYi00MTBlLWI3MTQtMmQyNjNmNmFjMjVkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE5VDE3NDQxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVjZDA2N2YxNzQwNzJlZmY2ZjRiOWIxYWU1MDhkYjA1MDhiNTBkYWY5Y2NkZmI4Zjc4YWIyM2RiNTBmMDZkMDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.NYDfmm4l5CCGwA5FGB48AoZzikEwhXYXmwAQJUtrEYk)
> 
> But now the “Start at Login” no longer works. You always have to start it manually after a restart. ![Bildschirmfoto 2024-02-19 um 18 23 22](https://private-user-images.githubusercontent.com/68192365/306037222-c3fe2c34-6a7a-44bf-a540-6646b1961239.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgzNjQ5NTYsIm5iZiI6MTcwODM2NDY1NiwicGF0aCI6Ii82ODE5MjM2NS8zMDYwMzcyMjItYzNmZTJjMzQtNmE3YS00NGJmLWE1NDAtNjY0NmIxOTYxMjM5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE5VDE3NDQxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFiNjljZWFjMjg5NjYyZjMwZjNlNTllODg0NWY1YWI1ZjllNzU2NzQ3YTRiNzE1NDYxYTFhMTRhYzY5N2YyYWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.FUYENCbyPk9DdtxKxIw4aYbcy9jOdhDtdRtYGU9K4us)
> 
> Otherwise it looks good.

You need to disable it first in the release one, and then enable it in the new one for it to work

ManOfDiamond, 2024-10-01:
> Working fine after replacing both the prefpane and the app.
> 
> @jozews321: I would say, just fork the repo and upload your own release, if the PR won't be merged.

i guess he should xD

tjjjia, 2024-11-10:
> You need to disable it first in the release one, and then enable it in the new one for it to work, alternatively you can add the app to login items in settings to set it to open automatically

Somehow this is not working for me. YogaSMCNC.app is in the "Allow in the Background" list, but I cannot add it to "Open at Login". See attached screen shot. I am using the version (Builds.zip) you shared on 18/02/2024, on Ventura.

BTW, thanks for making it work, really appreciate it. It just doesn't start automatically yet.

[image]