zhen-zen/YogaSMC #190: Hi There, Zhen-zen, the DYTC and charge ctrl seem not work at my Thinkpad x1y3, wonder if you could give some suggestions?

Einstonoy, 2023-01-10:
I used the debug version of 1.5.3 yogaSMC as follow pic shows.
[image]
[image]
SSDT-THINK and SSDT-YVPC were load in OC, the SSDT-YVPC has add my _HID of 'LEN0268'.
1st. For now the DYTC seems good after any reboot, however, the panel flaws as below pic shows when I try to adjust power mode, the situation remain same when the psc was enabled. 
[image]
2nd. The charge control seems not work either when I try to set the threshold level. The '80' and '95' in the screenshots are adjusted under windows 11 system.
It seems the EC can be read but cannot be write through the panel.
3rd. Concerning about the 'WNTF' in SSDT-THINK, how should I get right 'WNTF' number in my DSDT. Related screenshots as follow:
[image]
[image]
4th. Another one is about the 'SSDT-ECRW', you note that I should check the conflicts if I patched battery, which actually I did patch in my ACPI. Wonder it would be a cause for my flaws?
It would be so appreciative if you could give some suggestions for these problems.

The Yoga-debug log as follow:
[ 2516.705349]: YSMC - Debug: ThinkVPC::YogaSMCUserClient getTargetAndMethodForIndex (index=2)
[ 2516.706052]: YSMC - Debug: ThinkVPC::YogaSMCUserClient readEC 0x02 @ 0x84
[ 2518.530060]: YSMC - Debug: ThinkVPC::YogaSMCUserClient getTargetAndMethodForIndex (index=2)
[ 2518.530698]: YSMC - Debug: ThinkVPC::YogaSMCUserClient readEC 0x02 @ 0x84
[ 2518.990220]: YSMC - Debug: HKEY::ThinkVPC BCTG 0x80000000
[ 2518.990802]: YSMC - Debug: HKEY::ThinkVPC BCSG 0x80000000
[ 2518.991383]: YSMC - Debug: HKEY::ThinkVPC BICG 0x80000000
[ 2518.991939]: YSMC - Debug: HKEY::ThinkVPC BDSG 0x80000000
[ 2518.992732]: YSMC - Debug: HKEY::ThinkVPC PSSG 0x70
[ 2518.992737]: YSMC - Debug: HKEY::ThinkVPC Battery 0x2
[ 2518.993646]: YSMC - Debug: HKEY::ThinkVPC updateMutestatus 0x2
[ 2518.994226]: YSMC - Debug: HKEY::ThinkVPC updateMuteLEDStatus 0x100
[ 2518.994292]: YSMC - Debug: HKEY::ThinkVPC updateMicMuteLEDStatus 0x101
[ 2518.994853]: YSMC - Debug: HKEY::ThinkVPC KeyboardLocale 0x80000000
[ 2518.994930]: YSMC - Debug: HKEY::ThinkVPC empty HKEY event
[ 2518.995170]: YSMC - Debug: HKEY::ThinkVPC tablet mode: 40001, raw: 1
[ 2518.995182]: YSMC - Debug: HKEY::ThinkVPC TopCase Input enabled
[ 2518.997176]: YSMC - Debug: HKEY::ThinkVPC DYTCMode command 0x2 result 0x0001f001
[ 2519.004554]: YSMC - Debug: HKEY::ThinkVPC BCTG 0x80000000
[ 2519.005131]: YSMC - Debug: HKEY::ThinkVPC BCSG 0x80000000
[ 2519.005691]: YSMC - Debug: HKEY::ThinkVPC BICG 0x80000000
[ 2519.006249]: YSMC - Debug: HKEY::ThinkVPC BDSG 0x80000000
[ 2519.007177]: YSMC - Debug: HKEY::ThinkVPC PSSG 0x70
[ 2519.007185]: YSMC - Debug: HKEY::ThinkVPC Battery 0x0
[ 2519.009059]: YSMC - Debug: HKEY::ThinkVPC BCTG 0x350
[ 2519.009911]: YSMC - Debug: HKEY::ThinkVPC BCSG 0x35f
[ 2519.010823]: YSMC - Debug: HKEY::ThinkVPC BICG 0x70
[ 2519.011757]: YSMC - Debug: HKEY::ThinkVPC BDSG 0x700
[ 2519.012567]: YSMC - Debug: HKEY::ThinkVPC PSSG 0x70
[ 2519.012573]: YSMC - Debug: HKEY::ThinkVPC Battery 0x1
[ 2519.013239]: YSMC - Debug: HKEY::ThinkVPC BCTG 0x80000000
[ 2519.013697]: YSMC - Debug: HKEY::ThinkVPC BCSG 0x80000000
[ 2519.014150]: YSMC - Debug: HKEY::ThinkVPC BICG 0x80000000
[ 2519.014626]: YSMC - Debug: HKEY::ThinkVPC BDSG 0x80000000
[ 2519.015521]: YSMC - Debug: HKEY::ThinkVPC PSSG 0x70
[ 2519.015528]: YSMC - Debug: HKEY::ThinkVPC Battery 0x2
[ 2580.529421]: YSMC - Debug: HKEY::ThinkVPC Hotkey(MHKP) key presses event: 0x1010
[ 2580.725064]: YSMC - Debug: HKEY::ThinkVPC Hotkey(MHKP) key presses event: 0x1010
[ 2580.948261]: YSMC - Debug: HKEY::ThinkVPC Hotkey(MHKP) key presses event: 0x1010
[ 2581.489357]: YSMC - Debug: HKEY::ThinkVPC Hotkey(MHKP) key presses event: 0x1011

Looking forward to your reply and Thx a lot for you contribution to such a outstanding work for Hackintosh!

Einstonoy, 2023-01-10:
By the way, how should add patch for the _LID parameter to enable 'Clamshell Mode'?

Einstonoy, 2023-01-10:
The ASL seems difficult for me

Baio1977, 2023-02-11:
I answer about it since I also try to understand something by studying the ACPI origin.
[image]
So it is activated , Zhen can always prove me wrong .
I follow the conversation curiously.

Baio1977, 2023-02-11:
Thinkpad don't want this SSDT-YVPC

Baio1977, 2023-03-12:
I'm trying to understand how YogaSMC works at the ACPI level.
I've had these Thinkpads:
X280 Kabylake -r , X1 Tablet Kabylake -r , T480 Kabylake and T14 CometLake.Where I can get the best keyboard functionality is T14 I am attaching photos and file DSDT

[DSDT_T14.aml.zip](https://github.com/Baio1977/Lenovo-ThinkPad-X280/files/10950619/DSDT_T14.aml.zip)
[image]
[image]

On T480 X280 and X1Tablet 3Gen instead the keyboard does not respond as it should and many FN keys have no function

[DSDT_X1 Tablet.aml.zip](https://github.com/Baio1977/Lenovo-ThinkPad-X280/files/10950624/DSDT_X1.Tablet.aml.zip)
[image]
[image]

Maybe this could be the solution, use the Variable as written in DSDT.
It should be tested but now I only have T14 and I can't test.[image]

Baio1977, 2023-03-24:
I made some discoveries about Lenovo DSDT, here is what I learned:
Origin DSDT 
[image]

Error ACPI DSDT Origin
[image]

DSDT Fix Error ACPI FIX " \WNTF"
[image]

I attach ACPI Origin
[ACPI Origin.zip](https://github.com/zhen-zen/YogaSMC/files/11061730/ACPI.Origin.zip)

Einstonoy, 2023-03-24:
And will reply here once I finished.

Einstonoy, 2023-03-27:
Hey man, I add the '\' however the DYTC flaw remains, the keyboards setup works perfectly as before. Anyway, Thanks a lot for you advices.

Einstonoy, 2023-03-27:
''means '\'

Einstonoy, 2023-03-27:
means the symbol slash

m1k0lay, 2024-04-27:
[image]
Hi. A have the same problem as described. Can't switch DYTC mode after some time after reboot, switcher isn't active. T490 20N3-S3J800. How can i fix it?