zhen-zen/YogaSMC #198: Lacking Documentation for Prefpane features

5T33Z0, 2023-06-30:
Your Readme for YogaSMC doesn't contain any information about all the features available in the Preferance Pane of YogaSMC. Can you please provide some info about the following features/settings?

- General TAB
  - What is AutoSleep?
  - What is YogaMode
  - What is DYTC?
  - What does the Slider do that says Quiet –– Balance –– Performance
  - What is  PSC Support and why does the Slider change once you enable it? 
- Think Tab
  -  How does Charge Control work (what do the abbreviations mean) and how to configure it?

Thank you

gelomon, 2023-07-28:
I have dualboot windows 10 and ventura, YogaSMCPane picked up my vantage settings for battery limits.

Charge Control - Limits the maximum battery charge
PRI - Primary Battery
SEC - Secondary Battery

First Box - Charge Below
Second Box - Maximum Charge

For the General Tab, I don't get those things also

Baio1977, 2023-09-09:
General TAB
What is YogaMode = It activates when I disconnect the USB keyboard of my Thinkpad X1 Tablet 3rd Gen.
What is DYTC = This function activates the functionality of the YogaSMC kext, I deduce it is an ACPI method that controls various functions.
Keyboard mapping, LED control etc etc