zhen-zen/YogaSMC #209: Thinkpad T14 Gen 1 AMD

Collin8000, 2024-03-25:
When using the YogaSMC kext on my Thinkpad T14 Gen 1 AMD, I encounter slowdowns and it becomes quite unusable. When im change the DYTC Function, it changes to lap mode, I do not know why this happens but it is very frustrating. Currently, I am using this on my desk and am not moving it, the computer is also nowhere in the thermal throttling range, temps are between 70-75. It has also changed when the temps were 50 degrees Celcius. I have also attempted to use a cooling pad to no avail. believe that this is the fault of YogaSMC, yet something to do with ACPI inside of it. Because I have no issues like this just on my normal hackintosh without this kext. If there is anything I can do to edit this and to help figure out this issue, I would love to here it.

Sam-SSTU, 2024-10-04:
I met the same issue with intel chip. Have u solved this issue?

Collin8000, 2024-10-04:
No, I have not fixed it, it just fluctuates from time to time, its probably just an issue with how the ACPI functions with YogaSMC, I haven't found anything that fixes it though.