zhen-zen/YogaSMC #211: Thinkpad P14s gen 1 - Intel DYTC not support

ngophuong, 2024-07-31:
Hello, on my hackintosh, i cannot get the DYTC to work dispite install both SSDT-Think and SSDT-YVPC (already change the HID to LEN0268 as guide). 

## Model: Thinkpad P14s gen 1 Intel
## BIOS version: 1.31 (N2XET41W)

[image]

SSDT-YVPC update

[image]

Any help is appreciated...

Baio1977, 2024-08-01:
Post ioreg + DSDT origin

ngophuong, 2024-08-02:
Hello @Baio1977,

Please help to check attached file IOREG and ACPI Origin (ACPI dump from SSDT Time)

[APIC.aml.zip](https://github.com/user-attachments/files/16464361/APIC.aml.zip)
[MacBook Pro.ioreg.zip](https://github.com/user-attachments/files/16464362/MacBook.Pro.ioreg.zip)

Sam-SSTU, 2024-10-04:
I got the same issue, have u solved it at last?

Baio1977, 2024-10-05:
You need the ACPI extracted from Bios for an accurate analysis, all the ACPI files, you need EFI Debug to extract the files, the previous user attached ACPI.aml and I DON'T do anything with that file and that's it.