zhen-zen/YogaSMC #216: No HUD on Tahoe

qt2409xde, 2025-06-15:
Afaik, Tahoe changed the HUD to the new one but YogaSMCNC still utilizes the old HUD, so the HUD doesn't want to show up

jozews321, 2025-12-28:
@qt2409xde 
I made a PR to implement a custom OSD using SwiftUI for Tahoe, that replicates the behavior of the old one.

[image]

[image]

[image]

Here is the binary for you to test 

[YogaSMCNC 2025-12-28 02-37-17.zip](https://github.com/user-attachments/files/24358805/YogaSMCNC.2025-12-28.02-37-17.zip)

payoyo5150, 2026-09-26:
> Here is the binary for you to test
> 
> [YogaSMCNC 2025-12-28 02-37-17.zip](https://github.com/user-attachments/files/24358805/YogaSMCNC.2025-12-28.02-37-17.zip)

@jozews321 you should really post this as a release, even if it doesn't sync properly with the old prefpane, this is game changer for PSC support (It seems to work fine on Sonoma for me, been using it for the past week or so). Most users won't dig enough to find the binary here and use the old broken version instead.